espressif/arduino-esp32 #11389: HTTP/HTTPS POST timeout setting ignored

bfeldman-threadb, 2025-05-21:
### Board

ESP32 S2 SOLO 2 module

### Device Description

ESP32 S2 SOLO 2 module in a custom environment connected to a variety of other hardware, including I2C (DS3231 RTC, ADS7828) and SPI (Adafruit ILI9341 TFT, PN532 RFID board, NAU7802 ADC).

### Hardware Configuration

ESP32 S2 SOLO 2 module in a custom environment connected to a variety of other hardware, including I2C (DS3231 RTC, ADS7828) and SPI (Adafruit ILI9341 TFT, PN532 RFID board, NAU7802 ADC).

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE 2.3.6

### Operating System

Windows 11

### Flash frequency

240Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have code that runs fine on 3.2.0. This issue is occurring when I update the ESP32 board definition to 3.3.0-alpha1, and the problem disappears when board definition is reverted to 3.2.0.

In the posting routine the timeouts for https and http are set to 5000 (from the default 120000).

In 3.2.0 the line:
int postResult = https.POST(encodedString, ilength);
continues if no response is received after 5000 ms

In 3.3.0 the line:
int postResult = https.POST(encodedString, ilength);
will hang for the full 120000 ms (120 seconds) if no response is received even though a 5000 ms timeout is specified earlier in the routine

### Sketch

```cpp
HTTPClient1 https;
    HTTPClient1 http;
    https.setReuse(false);
    http.setReuse(false);
    https.setTimeout(5000);
    http.setTimeout(5000);

esp_task_wdt_reconfigure(&twdt_config2);

if (https.begin(server, rootCert))
        {
            protectedPrint("connected: ");
            protectedPrintln(https.connected());
            https.addHeader("Content-Type", "application/x-www-form-urlencoded");

            protectedPrint("POST: ");
            int postResult = https.POST(encodedString, ilength);
            protectedPrintln(postResult);
            protectedPrint("Response: ");
            https.getString().toCharArray(scc, 490);
            protectedPrintln(scc);
            https.end();
}
```

### Debug Message

```plain
No specific debug message is provided as this is not causing a failure, although

https.POST(encodedString, ilength);

returns -1 when no response is received
returns 200 when a successful response is received

That behavior is as expected. In board definition 3.2.0 a failure would return -1 in 5000 ms. In board definition 3.3.0-alpha1 -1 would not be returned for a failure for 120000 ms.
```

### Other Steps to Reproduce

This issue never occurred on board definition 3.2.0 and only started occurring when the ESP32 board definition was updated to 3.3.0-alpha1, and the problem disappeared when board definition was reverted to 3.2.0.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-10-02:
please try latest Arduino release core 3.3.1

Parsaabasi, 2025-10-29:
Hi @bfeldman-threadb

Does the issue still persist after using the latest Arduino release?

bfeldman-threadb, 2025-10-29:
I will test with the latest release shortly

bfeldman-threadb, 2026-01-08:
I believe this issue may be being caused by low internal heap memory caused by the change from core 3.2.0 to core 3.3.x. In the current state if I do not use SSL/TLS or certificates the HTTP posting works fine. When using SSL/TLS with verbose debugging enabled on core 3.3.5 the processor would hang until timing out on "ssl_starttls_handshake(): Performing the SSL/TLS handshake." Testing was done with another program with significantly more free heap memory and this issue was not occurring. The problem was noticed with free heap memory below about 55000. When the same program was compiled on 3.2.0 the free heap memory was approximately 65000.