espressif/arduino-esp32 #11395: Preferences doesnt work on esp32-c6

josef2600, 2025-05-23:
### Board

esp32-c6

### Device Description

esp32-c6 dev

### Hardware Configuration

[image]
or any other partition setting with spiffs. with or without zigbee.

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

win 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

256000

### Description

the Preferences doesn't save or load anything. i tested on many partition scheme, including zigbee. all are with spiffs. no matter what partition is selected or any zigbee is enabled or not (in both zigbee config).
as you can see in the code, i even save something in it inside setup, but nothing is saved, no matter how many times i restart it! and i do erase flash in programming. 

### Sketch

```cpp
#include <Arduino.h>
#include <string.h>
#include <algorithm>
#include <cstring>
#include <esp_err.h>
#include <esp_log.h>
#include <esp_mac.h>
#include <esp_system.h>
#include <freertos/FreeRTOS.h>
#include <freertos/event_groups.h>
#include <freertos/queue.h>
#include <freertos/task.h>
#include <nvs.h>
#include <nvs_flash.h>
#include <driver/uart.h>
#include <esp_attr.h>
#include <esp_check.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <esp_zigbee_core.h>
#include <ha/esp_zigbee_ha_standard.h>
#include <hal/ieee802154_ll.h>
#include <Preferences.h>  // For storing settings in flash

void setup()
{
  Serial.begin(115200);
  vTaskDelay(pdMS_TO_TICKS(500));
  //nvs_flash_init();
  //Ieee802154 my_ieee802154({.channel = my_frq_channel, .pan_id = my_pan_id}, onMessageCallback);
  vTaskDelay(pdMS_TO_TICKS(200));
  
  esp_err_t ret = nvs_flash_init();
  if (ret != ESP_OK)
  {
    nvs_flash_erase();
    nvs_flash_init();
    ESP_LOGI(DEVICE_NAME, "setup: nvs flash didnt intialized", 0);
  }  
  
  
  my_task_discovery_delay = 10;
  
  // Read saved configuration from flash using Preferences
  
  delay(10);
  my_preferences.begin(my_zigbee_set_per, false);
  
  schLennn = my_preferences.getBytesLength("me_pan_id");
  if (schLennn > 0)
  {
    my_pan_id = my_preferences.getUShort("me_pan_id", 0);
    if (my_pan_id == 0 || my_pan_id == 0xffff)
    {
      //my_preferences.end();
      ESP_LOGI(DEVICE_NAME, "setup: Bad PAN ID loaded: 0x%04X .trying to load the original", my_pan_id);
      my_pan_id = my_default_pan_id;
      delay(10);
      //my_preferences.begin(my_zigbee_set_per, false);
      my_preferences.putUShort("me_pan_id", my_pan_id);
      //my_preferences.end();
      delay(10);
      //my_preferences.begin(my_zigbee_set_per, true);
    }
    ESP_LOGI(DEVICE_NAME, "setup: New PAN ID loaded: 0x%04X", my_pan_id);
  }
  else
  {
    my_preferences.putUShort("me_pan_id", my_default_pan_id);
  }
  my_preferences.putUChar("me_frq_channel", my_frq_channel);
  schLennn = my_preferences.getBytesLength("me_frq_channel");
  if (schLennn > 0)
  {
    my_frq_channel = my_preferences.getUChar("me_frq_channel", 0);
    if (my_frq_channel == 0 || my_frq_channel > 26)
    {
      //my_preferences.end();
      ESP_LOGI(DEVICE_NAME, "setup: Wrong Frequency Channel loaded: %u .trying to load the original", my_frq_channel);
      my_frq_channel = my_default_frq_channel;
      delay(10);
      //my_preferences.begin(my_zigbee_set_per, false);
      my_preferences.putUChar("me_frq_channel", my_frq_channel);
      //my_preferences.end();
      delay(10);
      //my_preferences.begin(my_zigbee_set_per, true);
    }
    ESP_LOGI(DEVICE_NAME, "setup: New Frequency Channel loaded: %u", my_frq_channel);
  }
  else
  {
    my_preferences.putUChar("me_frq_channel", my_frq_channel);
  }
  
  delay(10);
  my_preferences.end();
  delay(10);
  
}
```

### Debug Message

```plain
[   718][E][Preferences.cpp:526] getBytesLength(): nvs_get_blob len fail: me_pan_id NOT_FOUND
[   726][E][Preferences.cpp:526] getBytesLength(): nvs_get_blob len fail: me_frq_channel NOT_FOUND
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Xylopyrographer, 2025-05-23:
Did you declare a Preferences object first?

josef2600, 2025-05-23:
yes, if i don't, it wont compile!
Preferences my_preferences;

P-R-O-C-H-Y, 2025-05-23:
I can tell that the preferences are working fine on C6 with Zigbee enabled.
You can check this example which is in PRs now: https://github.com/espressif/arduino-esp32/pull/11316/files#diff-a07d10b882a453a162de53818c6d675838412371ed2dd2f0adfdd6eb63f39924

SuGlider, 2025-05-23:
@josef2600 

``` cpp
  esp_err_t ret = nvs_flash_init();   // <<<<<==== 
  if (ret != ESP_OK)
  {
    nvs_flash_erase();
    nvs_flash_init();
    ESP_LOGI(DEVICE_NAME, "setup: nvs flash didnt intialized", 0);
  }  
```

Why executing this code in setup?
This is already done by Arduino `esp32-hal-misc.c` -> `void initArduino()` in the Core initialization
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L291-L307

This will most probably create problems with NVS. 
NVS is the Flash Partition used to store Preferences data.
https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/preferences.html#introduction

SuGlider, 2025-05-23:
@josef2600 - I have found the issue with the error message.

Instead of executing 
``` cpp
  // this line generate the error message because "me_pan_id" is not a valid Key <<<<<<=========
  schLennn = my_preferences.getBytesLength("me_pan_id");
  if (schLennn > 0)
  {
    my_pan_id = my_preferences.getUShort("me_pan_id", 0);
....
```

Try doing this:
``` cpp
  // if NVS is empty of there is no my_zigbee_set_per PREF Domain
  // my_pan_id will be initialized with 0, otherwise it will read the stored value from "me_pan_id" key
  my_pan_id= my_preferences.getBytesLength("me_pan_id", 0); 
  if (my_pan_id == 0 || my_pan_id == 0xffff)
  {
      ESP_LOGI(DEVICE_NAME, "setup: Bad PAN ID loaded: 0x%04X .trying to load the original", my_pan_id);
....
```

josef2600, 2025-05-23:
@SuGlider  thank you, but:
1. the point of `if (ret != ESP_OK)` is to know if it get error or not, and do something. so what if Arduino do it or not?
2. why everybody stuck on `getBytesLength`? do you know what and how it does it? whether its a byte or multi byte, it return the size. it return 0 if they are null or don't exist. its global so i don't miss match the instructions. 
3. after that, the `getBytesLength` is saying the key doesn't exist, even when i am doing exactly that in the setup itself. do you understand the problem? 
4. when you do `my_pan_id= my_preferences.getBytesLength("me_pan_id", 0); ` when it doesn't find the key, it returns the second register "0", so the return value will be zero if it doesn't find the key. if you put 10 instead of 0, it will return 10 if it doesn't find the key.OK?

@P-R-O-C-H-Y !?
funny! the `getBytes` works, but the others, `getUShort` or `putUShort` or `getUChar `or `putUChar` do not work! this is a big problem!
i am attaching the full test code. you can just send "ZigBee:my_pan_id01" and enter. 

```

#include <Arduino.h>
#include <string.h>
#include <algorithm>
#include <cstring>
#include <esp_err.h>
#include <esp_log.h>
#include <esp_mac.h>
#include <esp_system.h>
#include <freertos/FreeRTOS.h>
#include <freertos/event_groups.h>
#include <freertos/queue.h>
#include <freertos/task.h>
#include <nvs.h>
#include <nvs_flash.h>
#include <driver/uart.h>
#include <esp_attr.h>
#include <esp_check.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <esp_zigbee_core.h>
#include <ha/esp_zigbee_ha_standard.h>
#include <hal/ieee802154_ll.h>
#include <Preferences.h>  // For storing settings in flash

// -------------------------------------------------------------------
// Global parameters and registry
// -------------------------------------------------------------------
#define my_maximum_retry_send  6
#define the_maximum_message_lengh  72
#define the_size_of_my_full_discMsg  250
#define my_minimum_linkQuality  33
#define my_UART_PORT  UART_NUM_0
#define my_default_frq_channel   14
#define my_default_pan_id   0x6193

uint8_t my_frq_channel = my_default_frq_channel;
uint16_t my_pan_id = my_default_pan_id;

const char* DEVICE_NAME = "Hanta_IEEE:";
const char* hanta_starting_message = "Hanta_MSG_:";

// Command strings for updating configuration
const char* my_zigbee_pan_id_command    = "ZigBee:my_pan_id";  // these should begin with "ZigBee"
const char* my_zigbee_frq_channel_command = "ZigBee:my_frq_channel";  // these should begin with "ZigBee"
const char* my_zigbee_factory_reset_command = "ZigBee: factory Reset";  // these should begin with "ZigBee"
const char* my_zigbee_set_per = "meZigBee";

char my_full_discMsg[the_size_of_my_full_discMsg+1];

char my_global_payload[the_size_of_my_full_discMsg];
char my_serial_message[the_size_of_my_full_discMsg] = {0};
char my_message_backup[the_size_of_my_full_discMsg] = {0};
uint16_t t_serial_total_len = 0;
char my_serial_message2[the_size_of_my_full_discMsg] = {0};

bool serial_should_we_send = 0;
uint8_t my_current_count = 1;
[... 212 lines cut ...]
        ESP_LOGI(DEVICE_NAME, "Serial: we have reset config: %u", serial_should_we_reset);
        my_preferences.putUShort("me_pan_id", my_pan_id);
        my_preferences.putUChar("me_frq_channel", my_frq_channel);
        delay(10);
        my_preferences.end();
        delay(100);
        serial_we_have_change_config = 1;
        serial_should_we_reset = 1;
      }
      // --- End configuration commands ---      
      
      Serial.flush();
      t__len = 0;
      t_serial_total_len = 0;
    }
  }
  if (serial_should_we_reset)
  {
    ESP_LOGI(DEVICE_NAME, "Serial: we have reset config: %u", serial_should_we_reset);
    vTaskDelay(pdMS_TO_TICKS(5000));
    ESP.restart();
  }
}

```

Xylopyrographer, 2025-05-23:
A review of the API doc might help.

https://docs.espressif.com/projects/arduino-esp32/en/latest/api/preferences.html

'getBytesLength' is only valid for key of type Bytes in the currently open namespace.

josef2600, 2025-05-24:
yes, it works now. thank you, the` if (my_preferences.isKey("me_pan_id")) `works.
but what i dont understand is why `schLennn = my_preferences.getBytesLength("me_pan_id"); ` doesn't work? base on its code, it should work.
`public: size_t getBytesLength(const char *key)`
`uint16_t getUShort(const char *key, uint16_t defaultValue = 0);`

closed 2025-05-24 by josef2600