espressif/arduino-esp32 #11396: Trojan on riscv32-esp-elf-c++filt.exe

gorillale, 2025-05-23:
### Board

esp boards

### Device Description

Norton 360 say:
Trojan on riscv32-esp-elf-c++filt.exe
regards

### Hardware Configuration

Testing all installed files 

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE (newest)

### Operating System

win11 pro german

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Norton 360 say:
Trojan on riscv32-esp-elf-c++filt.exe
regards

### Sketch

```cpp
no sketch
```

### Debug Message

```plain
Norton 360 say:
Trojan on riscv32-esp-elf-c++filt.exe
regards
```

### Other Steps to Reproduce

Norton 360 say:
Trojan on riscv32-esp-elf-c++filt.exe
regards

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-05-23:
@gorillale - it seems to some mistake from Norton 360.
When I test `User\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2411\bin\riscv32-esp-elf-c++filt.exe` in `https://https://www.virustotal.com/`, it can't find any virus, trojan or issue with the file.

Report can be seen here: 
https://www.virustotal.com/gui/file/59a24c43c83e8302f725ec2a10f1d59bc2c66a74f9919fd1b10e5d56daa2c79c/detection

P-R-O-C-H-Y, 2025-05-23:
Also result for ARM64: https://www.virustotal.com/gui/file/708b1c0ec521f62f9ed8ce5de88ee3fcf052bf35d9cb9320a277dbc8a696bb0f/detection

Closing this as there is no virus.

Nexusnui, 2025-06-05:
I updated to 3.2.0 today and got this result:
https://www.virustotal.com/gui/file/36e5fc5db7555cc2952b1ac3cb18f357b8f7e3fff3b9a0dba6d875a362fc4f95

SuGlider, 2025-06-06:
> I updated to 3.2.0 today and got this result: https://www.virustotal.com/gui/file/36e5fc5db7555cc2952b1ac3cb18f357b8f7e3fff3b9a0dba6d875a362fc4f95

[image]

Based on https://www.virustotal.com/gui/file/59a24c43c83e8302f725ec2a10f1d59bc2c66a74f9919fd1b10e5d56daa2c79c
Those are not the same file...

[image]

those 2 files have different sizes.

gorillale, 2025-06-06:
i have uninstalled esp32, not problem anymore.