espressif/arduino-esp32 #11398: Hardware SPI in arduino-esp32 not work on ESP32-P4

linghuye, 2025-05-24:
### Board

DNE ESP32-P4

### Device Description

DNESP32P4 mini-board with esp32 p4 chip

### Hardware Configuration

no

### Version

v3.2.0

### IDE Name

VS Code platformIO

### Operating System

Windows 11

### Flash frequency

40MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi, I tried to use a ST7789 SPI 320*240 lcd screen on ESP32-P4 with Arduino TFT_eSPI and Adafruit ST7735 and ST7789 Library, 
and found they both don't work on ESP32-P4.
After some testing, I found the software SPI mode in Adafruit T7789 Library works good on ESP32-P4 but with very slow speed,
and it also works good with ESP32-IDF framwork's esp_lcd component with hardware SPI.
It just doesn't work with the Arduino's hardware SPI on ESP32-P4.

The Arduino core that I used is lastest 3.2.0. 

Finally I found that, the same simple code below lights up the ST7789 LCD with messy colors on ESP32-S3/ESP32-C3, but not work on ESP32-P4 with only a black backlighted screen.

```
// Works on ESP32 S3/C3 with hardware SPI
    pinMode(TFT_CS, OUTPUT);
    pinMode(TFT_DC, OUTPUT);
    pinMode(TFT_RST, OUTPUT);
    digitalWrite(TFT_CS, HIGH); // Chip select high (Inactive)
    digitalWrite(TFT_DC, HIGH); // Data/Command high = data mode
    digitalWrite(TFT_RST, HIGH);// NO Reset

    SPI.begin(TFT_SCLK, TFT_MISO, TFT_MOSI, -1);

    digitalWrite(TFT_CS, HIGH); // Chip select high (inactive)
    digitalWrite(TFT_DC, HIGH); // Data/Command high = data mode
    digitalWrite(TFT_RST, HIGH);
    delay(200);
    digitalWrite(TFT_RST, LOW);
    delay(20);
    digitalWrite(TFT_RST, HIGH);
    delay(200);
    
    // Sleep out
    GPIO.out_w1tc.val = (1 << TFT_CS);
[... 479 lines cut ...]
    delay(500);

    testTriangles();
    delay(500);

    testMediaButtons();
    delay(500);

    Serial.println("done");
    delay(1000);
}

void loop()
{
    tft.invertDisplay(true);
    delay(500);
    tft.invertDisplay(false);
    delay(500);
}

```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-05-24:
The lib `TFT_eSP` is outdated, not maintained and not working with core 3.x at all.

CreoKinetics, 2025-06-02:
Any update on this? I am also having the same sort of problem, <Adafruit_GFX.h> <Adafruit_SSD1351.h> on ESP32-P4, worked on Platformio Adruino framework 53.03.11, after updating to 54.03.20 (3.2.0) I cannot get the SPI bus to work with the display....

lbernstone, 2025-06-02:
I've had TFT_eSPI working properly on ESP32-C3 & C6, I don't have a P4 to test on.  Make sure to force it to HSPI (-DUSE_HSPI_PORT)

me-no-dev, 2025-06-03:
I've found that IOs might work slightly different depending on how their domains are powered. Maybe you can try with different IOs?

CreoKinetics, 2025-06-03:
I've tried several IO combinations, it seems like HSPI is the only bus exposed, but I have initialised as HSPI in my testing, not using a build flag though, I will try this. Need to try a loop back test and scope the pins too, which I will do when I get time later this week. For reference my testing code:

https://github.com/CreoKinetics/Olimex-ESP32-P4-Test

cniedzi, 2025-06-26:
What are the default SPI pins used for the hardware SPI mode in ESP32-P4?
According to ESP32-P4 Arduino core definition they should be:
```
MOSI (DIN): 32
MISO: 33
SCK (CLK): 36
CS/SS: 26
```

cniedzi, 2025-07-01:
Hi, any chance for fixing this issue?

Thanks!

schreibfaul1, 2025-07-01:
This is what my programme looks like:
````c++
#include "Arduino.h"
#include "SPI.h"

#define SPI_FREQUENCY      100000
#define SPI_MOSI          14 // 1
#define SPI_MISO          13 // 2
#define SPI_SCK           12 // 3

SPIClass spiBus(HSPI);

void setup() {
    spiBus.begin(SPI_SCK, SPI_MISO, SPI_MOSI, -1);
}

void loop() {
    vTaskDelay(1);
    spiBus.beginTransaction(SPISettings(SPI_FREQUENCY, MSBFIRST, SPI_MODE0));
    log_w("transf %i",     spiBus.transfer16(0x1234));
    spiBus.endTransaction();
}
````
and this is the result:
ESP32 works

[image]

ESP32-P4 does not work:

[image]

schreibfaul1, 2025-07-01:
I'm not sure which pins should be favoured.

[image]

I have tested various combinations.
`log_w("transf %i",     spiBus.transfer16(0x1234));` should not be 0? But it's always 0 for me.

[image]

P-R-O-C-H-Y, 2025-07-02:
@me-no-dev I can confirm the issue. Using same pins for SPI (tested on SD card - external connection) works on 3.1.3 and fails on master.

PR #11547 (2025-07-03): fix(spi): Fix bus clock for ESP32-P4 + remove S2 leftover
## Description of Change
This pull request updates the SPI handling for the ESP32P4 target in the `cores/esp32/esp32-hal-spi.c` file. It introduces new logic for enabling and resetting SPI bus clocks specific to ESP32P4 and removes legacy handling for unsupported configurations.

## Tests scenarios
Tested using ESP32-P4 and SD card example. Tested both FSPI and HSPI.

## Related links
Closes #11398