espressif/arduino-esp32 #11399: DHCP option 114 for captive portal

eyaler, 2025-05-24:
### Board

esp32s3

### Device Description

lilygo t-energy s3

### Hardware Configuration

n/a

### Version

latest master (checkout manually)

### IDE Name

arduino ide

### Operating System

windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

dhcp option 114 has been added to esp-idf 5.4 [here](https://github.com/espressif/esp-idf/commit/3035ce294d94cec9d801ab72866211cb534dac43), which has recently been [merged](https://github.com/espressif/arduino-esp32/pull/11212) into arduino-esp32 master. how can i use it in captive portal?

### Sketch

```cpp
i am using the captiveportal example
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-05-26:
It looks like there is a [CONFIG_ option](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/captive_portal/main/main.c#L82), so this would need to be enabled in the sdkconfig.

SuGlider, 2025-05-27:
> It looks like there is a [CONFIG_ option](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/captive_portal/main/main.c#L82), so this would need to be enabled in the sdkconfig.

@eyaler - As pointed by @lbernstone, in such case, the only solution would be building the Arduino Project as IDF Component in order to change settings with `idf.py menuconfig` and then enabling `CONFIG_ESP_ENABLE_DHCP_CAPTIVEPORTAL`, among other necessary IDF features.

Given that this is not an Arduino matter, I'll close this issue.

me-no-dev, 2025-05-27:
We could try and implement this, since it's optional and will require additional calls to DHCP. I've kept the issue in my backlog

me-no-dev, 2026-01-19:
@MoegerleStephan nice find! Fix coming: https://github.com/espressif/arduino-esp32/pull/12263