espressif/arduino-esp32 #11402: Provided example code "AnalogReadContinuous.ino" gives errors

PaulChristopher, 2025-05-25:
### Board

ESP32 Dev Module

### Device Description

ESP32 Development board (ESP WROOM 32)
Installed board support Arduino IDE: esp32 by Espressif Systems version 3.3.0-alpha1

### Hardware Configuration

Yes, a variable resistor to pin 34

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE

### Operating System

Win10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Running the example https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogReadContinuous/AnalogReadContinuous.ino works, but gives the following errors:

```
E (12579) adc_cali: adc_cali_raw_to_voltage(41): invalid argument: null pointer
E (12579) adc_cali: adc_cali_raw_to_voltage(41): invalid argument: null pointer
E (12582) adc_cali: adc_cali_raw_to_voltage(41): invalid argument: null pointer
E (12589) adc_cali: adc_cali_raw_to_voltage(41): invalid argument: null pointer
```

The errors are a pain, since they mess up the serial monitor with tons of error messages when reading results in continous mode using analogContinuousRead. This basically renders the serial monitor useless for other debugging purposes.

Since the errors cannot be supressed by the chosen debug level in the IDE, it is a blocker for me.

### Sketch

```cpp
See your own code: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogReadContinuous/AnalogReadContinuous.ino
```

### Debug Message

```plain
E (12579) adc_cali: adc_cali_raw_to_voltage(41): invalid argument: null pointer
E (12579) adc_cali: adc_cali_raw_to_voltage(41): invalid argument: null pointer
E (12582) adc_cali: adc_cali_raw_to_voltage(41): invalid argument: null pointer
E (12589) adc_cali: adc_cali_raw_to_voltage(41): invalid argument: null pointer
```

### Other Steps to Reproduce

Even commenting out "analogContinuousStop(); ... Serial.println ... analogContinousStart()" in loop() does not make a difference: The error seems to be caused by analogContinuousRead solely..

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

PaulChristopher, 2025-05-25:
Workaround: In C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0-alpha1\cores\esp32\esp32-hal-adc.c, function AnalogContinousRead comment out the call of adc_cali_raw_to_voltage (line 629) and work with raw values only.

lbernstone, 2025-05-25:
I'm not using the packaged 3.3.0, but current master is not giving me this error.  Did you try changing the adc_pins[] line to just have pin 34 in the list?

P-R-O-C-H-Y, 2025-05-27:
@PaulChristopher Will test and reach back.

P-R-O-C-H-Y, 2025-06-10:
Hi @PaulChristopher  I did test on 3.2.0 and also 3.3.0-alpha1 and both versions works for me. tested on ESP32-WROOM

closed 2025-06-10 by PaulChristopher