espressif/arduino-esp32 #11404: ESP32P4/C6 WiFi init doesn't work SDIO fail

smily77, 2025-05-27:
### Board

ESP32P4/C6

### Device Description

M5Stack Tab5

### Hardware Configuration

ESP32-P4 | G11            | G10          | G9             | G8             | G13              | G12

ESP32-C6 | SDIO2_D0 | SDIO2_D1 | SDIO2_D2 | SDIO2_D3 | SDIO2_CMD | SDIO2_CK

### Version

v3.2.0

### IDE Name

Arduino IDE 2.3.6

### Operating System

Windows11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

When the board should initialize the WiFi it crashes

### Sketch

```cpp
#include <M5Unified.h>
#include <WiFi.h>

const char* ssid     = "xxx";
const char* password = "yyy";

void setup() {
  Serial.begin(115200);
  auto cfg = M5.config();
  M5.begin(cfg);          

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(200);
    Serial.print(".");
  }
 }

void loop() {
  // ...
}
```

### Debug Message

```plain
ESP-ROM:esp32p4-eco2-20240710
sdio_mempool_create free:32254544 min-free:32254544 lfb-def:31457268 lfb-8bit:31457268

E (1736) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (1736) sdio_wrapper: sdmmc_card_init failed
E (4386) H_SDIO_DRV: sdio card init failed

FreeRTOS: FreeRTOS Task "sdio_read" should not return, Aborting now!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-05-27:
We can't really troubleshoot the M5 stack here, but ESP32-P4 does not have WiFi.

me-no-dev, 2025-05-28:
When we get the Tab5, we will have a look and add support. Currently only specific pin combination is supported on the Espressif P4 board https://github.com/espressif/arduino-esp32/blob/master/variants/esp32p4/pins_arduino.h#L75-L83

nikthefix, 2025-05-30:
@smily77 Did you try the new M5Stack boards package 3.2.0  
(Additional Boards URL: https://static-cdn.m5stack.com/resource/arduino/package_m5stack_index.json)
There is now a Tab5 board option which should remap the SDIO -> C6 pins which are not the same as the Espressif dev kit / Waveshare defaults.

smily77, 2025-05-30:
@nikthefix yes i did. Unfortunatly same result

nikthefix, 2025-06-20:
@cheops No I've not found a simple solution so far.
I think recompiling the relevant Arduino libs using the Espressif Library Builder is an interim solution.

Jason2866, 2025-06-23:
The SDIO GPIOs for Hosted communication are set in sdkconfig. So the are hard coded and can't be changed in Arduino. Currently there is only the solution to compile Arduino as an component of IDF or use pioarduino HybridCompile where sdkconfig settings can be changed.
The Arduino libs are compiled with the changed settings and provided to Arduino. HybridCompile is a pioarduino only feature.

me-no-dev, 2025-06-23:
This is almost true. In 3.2.x only SDIO pins can be defined (not the RST pin), in 3.3.x all pins will be definable, even at runtime

me-no-dev, 2025-06-24:
Fix is coming: https://github.com/espressif/arduino-esp32/pull/11513
```cpp
WiFi.setPins(clk, cmd, d0, d1, d2, d3, rst);
WiFi.begin(ssid, pass);
```

me-no-dev, 2025-06-24:
> Looks promising, but will this work with SDIO2 as well? As the M5Stack Tab 5 uses SDIO2.

It is set to use SDIO2 by default

nikthefix, 2025-06-25:
This is great! Plenty to experiment with now thanks to me-no-dev.

nikthefix, 2025-06-26:
@Jason2866 I've never understood the rationale behind hard coding elements of sdkconfig and / or the pre-compiled libs for the ESP_Arduino Core.
Sure, things compile faster and are error limited to the set of all things familiar in the Arduino environment.
But this is holding us back. I get the idea of damage limitation but am constantly frustrated that it's sometimes difficult or impossible to duplicate my IDF projects and ideas in an accessible Arduino form. Perhaps there should be an option in the Arduino core to compile from source, respecting sdkconfig and with all responsibility on the part of the coder to resolve the inevitable c vs c++ conflicts which occur.

Jason2866, 2025-06-26:
@nikthefix  Use Arduino as an component of IDF and you can (and have to) configure everything. Arduino is simplified by design. Don't understand what you mean with HybridCompile only works local.

nikthefix, 2025-06-27:
@Jason2866 Thanks for the info. Yes please ignore my comment about HybridCompile - I misunderstood.

Westcott1, 2025-06-28:
The Guition JC8012P4A1 panel works without any extra configuration.

closed 2025-06-30 by me-no-dev

nikthefix, 2025-06-30:
@Westcott1 Yes I believe the Guition board follows the C6 pin config of the Espressif P4 dev board so I would expect it to work with Arduino wifi out of the box.

The Tab5 has a different pinout which I find odd given it's an early kit which could otherwise have worked with the existing preliminary Arduino P4 examples.

They've crammed a lot into their product so perhaps IO choice was constrained.

@me-no-dev I didn't get far integrating the changes to WiFiGeneric.cpp/h to V3.2.0 (I didn't expect to really).
The reference in the constructor of WiFi.setPins() to rst was the problem. 
I modified the structures in the wifiHosted code to accommodate it as an extra member but things were soon beyond my pay grade. 

Any ETA of 3.3.x in which these issues will be resolved?

felmue, 2025-07-05:
Hello guys

I can confirm that using `3.2.1` and setting the correct GPIOs works for me.
```
#define SDIO2_CLK GPIO_NUM_12
#define SDIO2_CMD GPIO_NUM_13
#define SDIO2_D0  GPIO_NUM_11
#define SDIO2_D1  GPIO_NUM_10
#define SDIO2_D2  GPIO_NUM_9
#define SDIO2_D3  GPIO_NUM_8
#define SDIO2_RST GPIO_NUM_15

WiFi.setPins(SDIO2_CLK, SDIO2_CMD, SDIO2_D0, SDIO2_D1, SDIO2_D2, SDIO2_D3, SDIO2_RST);
```
Note: verified with VSC and pioarduino.

@me-no-dev : nicely done!

Thanks
Felix

nikthefix, 2025-07-06:
I also confirm that it's working with Tab5 if board = ESP32P4 Dev Module.

Clearly the M5tab5 board selection will not work until M5stack update their boards package to integrate arduino_esp32 3.2.1

This is no problem as everything else I've tried works fine using 'ESP32P4 Dev Module'.