espressif/arduino-esp32 #11405: DoubleException crash when using `std::regex_search`

av4625, 2025-05-27:
### Board

ESP32 Dev Module 38pin

### Device Description

DevKitC

### Hardware Configuration

Nothing is attached, just plugged into MacBook Pro with a USB cable.

### Version

v3.2.0

### IDE Name

VSCode

### Operating System

MacOS 12.7.6

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Getting a DoubleException crash when using `std::regex_search`.

If you change `rmc_fix` to a shorter string like `"$GPRMC"` it does not crash.

I'm unable to decode the exception because I run into this issue:
https://github.com/me-no-dev/EspExceptionDecoder/issues/34

### Sketch

```cpp
#include <regex>
#include <string>

void setup()
{
   static const std::regex rmc(
        "\\$(G[ABLPN]RMC,(?:([0-9]{2})([0-9]{2})([0-9]{2})\\.?([0-9]{0,3}))?,(A|V),"
        "(?:([0-9]{1,2})([0-9]{2}\\.[0-9]{0,6}))?,(N|S)?,"
        "(?:([0-9]{1,3})([0-9]{2}\\.[0-9]{0,6}))?,(E|W)?,"
        "([0-9]+\\.?[0-9]*)?,(-?[0-9]+\\.?[0-9]*)?,(?:([0-9]{2})([0-9]{2})([0-9]{2}))?,"
        "(-?[0-9]+\\.?[0-9]*)?,(E|W)?,(A|D|E|N)?)\\*([0-9A-Fa-f]{2})\\r");

    Serial.begin(115200);

    Serial.println("1");
    const std::string rmc_fix("$GPRMC,111111.45,A,1111.29088,N,00554.79795,W,3.308,110.50,270525,,,A*7D\r");
    Serial.println("2");
    std::smatch matches;
    Serial.println("3");
    std::regex_search(rmc_fix, matches, rmc);
    Serial.println("4");
}

void loop()
{
}
```

### Debug Message

```plain
Calculated checksum='d8ca7b41'
Image checksum='ffffffff'
1
2
3
Guru Meditation Error: Core  1 panic'ed (Double exception). 

Core  1 register dump:
PC      : 0x40090b96  PS      : 0x00040d36  A0      : 0x800d50ee  A1      : 0x3ffc6d70  
A2      : 0x3ffc8c1c  A3      : 0x00000001  A4      : 0x000000dd  A5      : 0x00000000  
A6      : 0x3ffca6a4  A7      : 0x3ffcd174  A8      : 0x40080080  A9      : 0x3ffc6ea0  
A10     : 0x00060f36  A11     : 0x00040026  A12     : 0x000000d7  A13     : 0x00000000  
A14     : 0x3ffcb1dc  A15     : 0x3ffcd1bc  SAR     : 0x0000000b  EXCCAUSE: 0x00000002  
EXCVADDR: 0xffffffe0  LBEG    : 0x400d373c  LEND    : 0x400d374d  LCOUNT  : 0x00000000  

Backtrace: 0x40090b93:0x3ffc6d70 0x400d50eb:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 0x40090b93:0x3ffc6e40 0x40090b93:0x3ffc6e70 0x40090b93:0x3ffc6ea0 0x4008007d:0x3ffc6d70 0x400d51f1:0x3ffc6da0 0x40090b93:0x3ffc6dc0 0x40090b93:0x3ffc6df0 0x40090b93:0x3ffc6e20 |<-CONTINUES

ELF file SHA256: 13af93d6d
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-05-28:
@av4625 - I have tested your example and tried to print all `std:string` matches that were found. Apparently `std::regex_search(rmc_fix, matches, rmc);` still needs stack space to process the 22 found regex matches.

Therefore, the only way to solve it is, as @lbernstone mentioned, resizing the main `setup/loop` task Stack Size by using the clause `SET_LOOP_TASK_STACK_SIZE(16 * 1024);  // 16KB`

``` cpp
SET_LOOP_TASK_STACK_SIZE(16 * 1024);  // 16KB
#include <iostream>  // for std::cout
#include <regex>
#include <string>

void setup()
{
   static const std::regex rmc(
        "\\$(G[ABLPN]RMC,(?:([0-9]{2})([0-9]{2})([0-9]{2})\\.?([0-9]{0,3}))?,(A|V),"
        "(?:([0-9]{1,2})([0-9]{2}\\.[0-9]{0,6}))?,(N|S)?,"
        "(?:([0-9]{1,3})([0-9]{2}\\.[0-9]{0,6}))?,(E|W)?,"
        "([0-9]+\\.?[0-9]*)?,(-?[0-9]+\\.?[0-9]*)?,(?:([0-9]{2})([0-9]{2})([0-9]{2}))?,"
        "(-?[0-9]+\\.?[0-9]*)?,(E|W)?,(A|D|E|N)?)\\*([0-9A-Fa-f]{2})\\r");

    Serial.begin(115200);

    Serial.println("1");
    const std::string rmc_fix("$GPRMC,111111.45,A,1111.29088,N,00554.79795,W,3.308,110.50,270525,,,A*7D\r");
    Serial.println("2");
    std::smatch matches;
    Serial.println("3");
    std::regex_search(rmc_fix, matches, rmc);
    Serial.println("4");
    if (std::regex_search(rmc_fix, matches, rmc)) {
        int numMatches = matches.size();
        Serial.printf("Found %d matches.\r\n", numMatches);
        Serial.printf("Global Match found: %s\r\n", matches.str(0).c_str()); 
        for (int i = 1;  i < numMatches; i++) {
            std::cout << "group: " << i << " :: " << matches.str(i) << std::endl;
        }
        Serial.println("No More Group Matches...");
    } else {
        Serial.println("No Match!");
    }
    Serial.println("5");
}

void loop()
{
}
```

closed 2025-05-28 by SuGlider

av4625, 2025-05-28:
@lbernstone  @SuGlider Thanks guys, annoyed with myself for not thinking of this! The error threw me. Seems I "just" crept over the default too.