espressif/arduino-esp32 #11414: [Arduino] ESP32-C6 Constantly reboots when joining network

ScaredyCat, 2025-05-30:
### Board

XIAO ESP32-C6 

### Device Description

Seeed Studio XIAO ESP32C6 

### Hardware Configuration

No additional hardware

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE

### Operating System

MacOS / HAOS

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Using the example Zigbee_on_off_light example in the arduino environment (2.3.6). Erase flash before upload, Zigbee 4MB with spiffs and Zigbee end device set. 

Using Zigbe2MQTT device repeately reboots when trying to join the network and tries to rejoin again.

`
mode:DIO, clock div:2
load:0x40875720,len:0x1260
load:0x4086c110,len:0xdc4
load:0x4086e610,len:0x3018
entry 0x4086c110
Adding ZigbeeLight endpoint to Zigbee Core
[    66][I][ZigbeeCore.cpp:160] zigbeeInit(): List of registered Zigbee EPs:
[    73][I][ZigbeeCore.cpp:162] zigbeeInit(): Device type: On/Off Light Device, Endpoint: 10, Device ID: 0x0100
[    86][I][ZigbeeCore.cpp:241] esp_zb_app_signal_handler(): Zigbee stack initialized
[    95][I][ZigbeeCore.cpp:248] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[   104][I][ZigbeeCore.cpp:255] esp_zb_app_signal_handler(): Start network steering
Connecting to network
.............................ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1e (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1260
load:0x4086c110,len:0xdc4
load:0x4086e610,len:0x3018
entry 0x4086c110
Adding ZigbeeLight endpoint to Zigbee Core
[    65][I][ZigbeeCore.cpp:160] zigbeeInit(): List of registered Zigbee EPs:
[    72][I][ZigbeeCore.cpp:162] zigbeeInit(): Device type: On/Off Light Device, Endpoint: 10, Device ID: 0x0100
[    85][I][ZigbeeCore.cpp:241] esp_zb_app_signal_handler(): Zigbee stack initialized
[    94][I][ZigbeeCore.cpp:248] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[   103][I][ZigbeeCore.cpp:255] esp_zb_app_signal_handler(): Start network steering
Connecting to network
.............................

`

### Sketch

```cpp
// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates simple Zigbee light bulb.
 *
 * The example demonstrates how to use Zigbee library to create a end device light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 */

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

/* Zigbee light bulb configuration */
#define ZIGBEE_LIGHT_ENDPOINT 10
uint8_t led = LED_BUILTIN;
uint8_t button = BOOT_PIN;

ZigbeeLight zbLight = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT);

/********************* RGB LED functions **************************/
void setLED(bool value) {
  digitalWrite(led, value);
}

/********************* Arduino functions **************************/
void setup() {
  Serial.begin(115200);

  // Init LED and turn it OFF (if LED_PIN == RGB_BUILTIN, the rgbLedWrite() will be used under the hood)
  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);

  // Init button for factory reset
  pinMode(button, INPUT_PULLUP);

  //Optional: set Zigbee device name and model
  zbLight.setManufacturerAndModel("Espressif", "ZBLightBulb");

  // Set callback function for light change
  zbLight.onLightChange(setLED);

  //Add endpoint to Zigbee Core
  Serial.println("Adding ZigbeeLight endpoint to Zigbee Core");
  Zigbee.addEndpoint(&zbLight);

  // When all EPs are registered, start Zigbee. By default acts as ZIGBEE_END_DEVICE
  if (!Zigbee.begin()) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println();
}

void loop() {
  // Checking button for factory reset
  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(button) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.println("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
    // Toggle light by pressing the button
    zbLight.setLight(!zbLight.getLightState());
  }
  delay(100);
}
```

### Debug Message

```plain
[   123][D][ZigbeeCore.cpp:97] addEndpoint(): Endpoint: 10, Device ID: 0x0100
[   124][D][ZigbeeCore.cpp:145] zigbeeInit(): Initialize Zigbee stack
[   181][D][ZigbeeCore.cpp:152] zigbeeInit(): Register all Zigbee EPs in list
[   182][I][ZigbeeCore.cpp:160] zigbeeInit(): List of registered Zigbee EPs:
[   182][I][ZigbeeCore.cpp:162] zigbeeInit(): Device type: On/Off Light Device, Endpoint: 10, Device ID: 0x0100
[   187][I][ZigbeeCore.cpp:241] esp_zb_app_signal_handler(): Zigbee stack initialized
[   187][D][ZigbeeCore.cpp:242] esp_zb_app_signal_handler(): Zigbee channel mask: 0x07fff800
[   189][I][ZigbeeCore.cpp:248] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[   190][I][ZigbeeCore.cpp:255] esp_zb_app_signal_handler(): Start network steering
Connecting to network
..........................=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   478028 B ( 466.8 KB)
  Free Bytes        :   446684 B ( 436.2 KB)
  Allocated Bytes   :    24920 B (  24.3 KB)
  Minimum Fre  441456 B ( 431.1 KB)
  Largest Free Block:   409588 B ( 400.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     3 : GPIO
    12 : USB_DM
    13 : USB_DP
    14 : GPIO
    15 : GPIO
    16 : UART_TX[0]
    17 : UART_RX[0]
============ After Setup End =============
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ScaredyCat, 2025-05-31:
Just tried again - this is the logging from Zigbee2MQTT. 

`2025-05-31 07:47:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb43a45fffe8a2c28","ieee_address":"0xb43a45fffe8a2c28"},"type":"device_joined"}'
[2025-05-31 07:47:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb43a45fffe8a2c28","ieee_address":"0xb43a45fffe8a2c28","status":"started"},"type":"device_interview"}'
[2025-05-31 07:47:08] debug: 	zh:deconz:driver: DATA_REQUEST - destAddr: 0xfe64 EP:0 SeqNr. 122 request id: 41
[2025-05-31 07:47:08] debug: 	zh:deconz:driver: query aps data indication
[2025-05-31 07:47:08] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 123
[2025-05-31 07:47:08] debug: 	zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 41 status: 0
[2025-05-31 07:47:08] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-05-31 07:47:08] debug: 	zh:deconz:driver: query aps data indication
[2025-05-31 07:47:08] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 124
[2025-05-31 07:47:08] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 123 srcAddr: 0xb43a45fffe8a2c28 destAddr: 0x0 profile id: 0x0 cluster id: 0x2 lqi: 220
[2025-05-31 07:47:08] debug: 	zh:deconz:frameparser: response payload: 020000
[2025-05-31 07:47:08] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-05-31 07:47:08] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - Error: Unsupported response reading for cluster ID '2'.
[2025-05-31 07:47:09] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 100110
[2025-05-31 07:47:09] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-05-31 07:47:09] debug: 	zh:deconz:driver: query aps data confirm
[2025-05-31 07:47:09] debug: 	zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 125
[2025-05-31 07:47:09] debug: 	zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xfe64 request id: 41 confirm status: 0
[2025-05-31 07:47:09] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-05-31 07:47:11] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 101010
[2025-05-31 07:47:11] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-05-31 07:47:11] debug: 	zh:deconz:driver: query aps data indication
[2025-05-31 07:47:11] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 126
[2025-05-31 07:47:11] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 126 srcAddr: 0xb43a45fffe8a2c28 destAddr: 0xffff profile id: 0x0 cluster id: 0x13 lqi: 221
[2025-05-31 07:47:11] debug: 	zh:deconz:frameparser: response payload: 01a3b5282c8afeff453ab48c
[2025-05-31 07:47:11] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-05-31 07:47:11] debug: 	zh:deconz: Try to find network address of b43a45fffe8a2c28
[2025-05-31 07:47:11] debug: 	zh:controller: Device '0xb43a45fffe8a2c28' joined
[2025-05-31 07:47:11] debug: 	zh:controller: Received ZDO response: clusterId=END_DEVICE_ANNOUNCE, status=SUCCESS, payload={"nwkAddress":46499,"eui64":"0xb43a45fffe8a2c28","capabilities":{"alternatePANCoordinator":0,"deviceType":0,"powerSource":1,"rxOnWhenIdle":1,"reserved1":0,"reserved2":0,"securityCapability":0,"allocateAddress":1}}
[2025-05-31 07:47:11] debug: 	zh:controller: Device announce from '0xb43a45fffe8a2c28:46499'
[2025-05-31 07:47:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xb43a45fffe8a2c28', payload '{"last_seen":"2025-05-31T06:47:11.318Z"}'
[2025-05-31 07:47:11] debug: 	zh:controller: Device '0xb43a45fffe8a2c28' got new networkAddress '46499'
[2025-05-31 07:47:11] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-05-31 07:47:11] debug: 	z2m: Device '0xb43a45fffe8a2c28' changed network address
[2025-05-31 07:47:11] debug: 	z2m: Device '0xb43a45fffe8a2c28' announced itself
[2025-05-31 07:47:11] debug: 	zh:controller: Device '0xb43a45fffe8a2c28' accepted by handler
[2025-05-31 07:47:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xb43a45fffe8a2c28', payload '{"last_seen":"2025-05-31T06:47:11.339Z"}'
[2025-05-31 07:47:11] debug: 	zh:controller: Not interviewing '0xb43a45fffe8a2c28', interviewState=IN_PROGRESS'
[2025-05-31 07:47:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb43a45fffe8a2c28","ieee_address":"0xb43a45fffe8a2c28"},"type":"device_announce"}'
[2025-05-31 07:47:11] debug: 	zh:deconz:driver: query aps data indication
[2025-05-31 07:47:11] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 127
[2025-05-31 07:47:11] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 127 srcAddr: 0xb43a45fffe8a2c28 destAddr: 0x0 profile id: 0x0 cluster id: 0x2 lqi: 221
[2025-05-31 07:47:11] debug: 	zh:deconz:frameparser: response payload: 020000
[2025-05-31 07:47:11] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...]
[2025-05-31 07:47:26] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xb43a45fffe8a2c28', payload '{"last_seen":"2025-05-31T06:47:26.971Z"}'
[2025-05-31 07:47:26] debug: 	zh:controller: Device '0xb43a45fffe8a2c28' got new networkAddress '47602'
[2025-05-31 07:47:26] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-05-31 07:47:27] debug: 	z2m: Device '0xb43a45fffe8a2c28' changed network address
[2025-05-31 07:47:27] debug: 	z2m: Device '0xb43a45fffe8a2c28' announced itself`

P-R-O-C-H-Y, 2025-06-02:
Hi @ScaredyCat, can you please use Verbose debug logging and also in Zigbee mode select your mode with the DEBUG option.
I done think there is anything broken in Zigbee library. The main issue I see is the `Seeed Studio XIAO ESP32C6`, you need to have a proper antenna setup. Do you use external one and do you set it up in your app correctly?

P-R-O-C-H-Y, 2025-06-02:
Please refer to this comment: https://github.com/espressif/arduino-esp32/issues/10951#issuecomment-2673795818

closed 2025-06-02 by P-R-O-C-H-Y