espressif/arduino-esp32 #11419: Division error when mixing (int32_t) with (uint32_t)

Rob58329, 2025-05-31:
### Board

Waveshare ESP32-S3 zero

### Device Description

Plain board only.

### Hardware Configuration

Plain board only.

### Version

latest stable Release (if not listed below)

### IDE Name

Adruino IDE v1.8.19

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

The following sketch/calculation gives a big (int32_t) number when it should give a small negative (int32_t) number.

A similar error occurs if doing a (uint32_t) divided by an (int32_t).

This error can be got around by limiting the (uint32_t) number to 0x7FFFFFFF and then casting it to an (int32_t).

However, I think it would be preferable if this was not necessary!?!

### Sketch

```cpp
void setup() {
  Serial.begin(115200);
  delay(3000);
  int32_t  x=-2000;
  uint32_t y= 1000;
  int32_t  z=x/y; // an (int32_t) divided by a (uint32_t)
  Serial.printf("Error: x=%i, y=%u, x/y=z=%i\n",x,y,z); // Error: x=-2000, y=1000, x/y=z=4294965
  x=abs(x); z=x/y;
  Serial.printf("But:   x=%i, y=%u, x/y=z=%i\n",x,y,z); // But:   x=2000, y=1000, x/y=z=2
}

void loop() {}
```

### Debug Message

```plain
Error: x=-2000, y=1000, x/y=z=4294965
But:   x=2000, y=1000, x/y=z=2
```

### Other Steps to Reproduce

None

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-06-01:
Uh, take that up with the gcc folks.  We are at the mercy of the compiler in this project.

closed 2025-06-01 by Rob58329

oqibidipo, 2025-06-02:
> It is a pity that there is no compiler-warning (even with “Compiler Warnings=All” set) which warns when this “mixed (int32_t) numbers and (uint32_t) numbers in division” is found!..

There is option `-Wsign-conversion` (not included in `-Wall -Wextra`)

```
>>  g++ -fsyntax-only -Wall -Wextra -Wsign-conversion foo.cpp
foo.cpp: In function ‘void foo(int32_t, uint32_t)’:
foo.cpp:8:26: warning: conversion to ‘uint32_t’ {aka ‘unsigned int’} from ‘int32_t’ {aka ‘int’} may change the sign of the result [-Wsign-conversion]
    8 |             << "x*y=" << x*y << '\t'
      |                          ^
foo.cpp:9:26: warning: conversion to ‘uint32_t’ {aka ‘unsigned int’} from ‘int32_t’ {aka ‘int’} may change the sign of the result [-Wsign-conversion]
    9 |             << "x/y=" << x/y << std::endl;
      |                          ^
```