espressif/arduino-esp32 #11422: Support Arduino on latest esp-idf toolchain (IDFGH-15408)

apalazzi, 2025-06-03:
### Is your feature request related to a problem?

Arduino is not supported on esp-id >5.1.6, but I'd like to use some features that are only available in the latest toolchains. 

### Describe the solution you'd like.

Add support for Arduino on the latest toolchain.

### Additional context.

The error is:

  ERROR: Because espressif/arduino-esp32 (3.0.7) depends on idf
  (>=5.1.0,<5.2.0) which doesn't match any versions, espressif/arduino-esp32
  is forbidden.

igrr, 2025-06-03:
@apalazzi could you show what your idf_component.yml file adding arduino-esp32 looks like? There are newer versions than 3.0.7 available (which are compatible with newer IDF versions), any chance you are pinning an old version?

me-no-dev, 2025-06-03:
- Arduino 3.0.x supports ESP-IDF 5.1
- Arduino 3.1.x supports ESP-IDF 5.3
- Arduino 3.2.x supports ESP-IDF 5.4
- Arduino 3.3.x supports ESP-IDF 5.5 (only alpha1 released, but latest is available in the `release/v3.3.x` branch from git)

Jason2866, 2025-06-03:
Recently compiled successfully Arduino 3.0 with IDF 5.1 by using the branch https://github.com/espressif/arduino-esp32/tree/release/v3.0.x

Jason2866, 2025-06-03:
@apalazzi  You have to change the YML file entry. You did before, there was no such specific version "nailing"

apalazzi, 2025-06-03:
@Jason2866 I don't remember manually editing that file, in fact I think what I did was somehow follow the instructions here https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html - and note that the instructions make you install the 3.0.2 version.
On that note, I really wish the system would be more polished and with better documentation, that would avoid a lot of struggle and frustration to people like me that just want to program for esp32 and are not interested in wasting hours in learning yet another build system just to make things work.