espressif/arduino-esp32 #11455: ledcWrite() returns successful execution but does not take effect.

zhengyangliu, 2025-06-10:
### Board

ESP32 DEV  and a Unknown esp32 board

### Device Description

Board only.

### Hardware Configuration

Nothing. Only oscilloscope.

### Version

v3.1.3

### IDE Name

Arduino IDE

### Operating System

Windows10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

This is a strange problem. When I use v3.1.3, after executing the following code, io5 on the oscilloscope cannot detect any waveform.

```c
#include <Arduino.h>

bool success1 = 0;
bool success2 = 0;

void setup() {
  Serial.begin(115200);
  ledcAttachChannel(4, 490, 8, 0);
//  ledcAttachChannel(5, 50, 10, 2);
  ledcAttachChannel(5, 50, 10, 2);
 // delay(21);
  success1 = ledcWrite(4, 128);
  success2 = ledcWrite(5, 75);
  Serial.println("success1:" + String(success1) + "  success2:" + String(success2));
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Once I change the timer and resolution of io5, io5 cannot output waveforms. But as long as I enable the delay function greater than 20ms in the middle, the code works fine again.

With the same code structure, this problem does not occur when I use version v1.0.6.

```c
#include <Arduino.h>

void setup() {
  ledcSetup(0, 490, 8);  
  ledcAttachPin(4, 0);
  
  ledcSetup(2, 490, 10);  
  ledcAttachPin(5, 2);
  
  ledcWrite(0, 128);
  ledcWrite(2, 75);
}

void loop() {
}
```

### Sketch

```cpp
#include <Arduino.h>

bool success1 = 0;
bool success2 = 0;

void setup() {
  Serial.begin(115200);
  ledcAttachChannel(4, 490, 8, 0);
//  ledcAttachChannel(5, 50, 10, 2);
  ledcAttachChannel(5, 50, 10, 2);
  delay(21);
  success1 = ledcWrite(4, 128);
  success2 = ledcWrite(5, 75);
  Serial.println("success1:" + String(success1) + "  success2:" + String(success2));
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16492
load:0x40080400,len:4
load:0x40080404,len:3524
entry 0x400805b8
[     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d40f4
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d40c4
[    26][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d4094
[    39][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d4064
[    53][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d40f4
[    66][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d40c4
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d4094
[    93][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d4064
[   107][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d40f4
[   120][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d40c4
[   133][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d4094
[   147][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d4064
[   163][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   176][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb60
[   187][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb60
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q6
  Revision          : 1.00
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   379844 B ( 370.9 KB)
  Free Bytes        :   340444 B ( 332.5 KB)
  Allocated Bytes   :    32212 B (  31.5 KB)
  Minimum Free Bytes:   334904 B ( 327.1 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jun 10 2025 20:29:37
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.2-584-g489d7a2b3a-dirty
  Arduino Version   : 3.1.3
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=verbose,EraseFlash=none,ZigbeeMode=default
============ Before Setup End ============
[   697][V][esp32-hal-uart.c:421] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   706][V][esp32-hal-uart.c:510] uartBegin(): UART0 not installed. Starting installation
[   716][V][esp32-hal-uart.c:575] uartBegin(): UART0 initialization done.
[   731][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type LEDC (12) successfully set to 0x400d1964
[   743][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 4 successfully set to type LEDC (12) with bus 0x3ffb8df0
[   753][I][esp32-hal-ledc.c:166] ledcAttachChannel(): LEDC attached to pin 4 (channel 0, resolution 8)
[   762][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type LEDC (12) successfully set to 0x400d1964
[   773][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 5 successfully set to type LEDC (12) with bus 0x3ffb8e10
[   784][I][esp32-hal-ledc.c:166] ledcAttachChannel(): LEDC attached to pin 5 (channel 2, resolution 10)
success1:1  success2:1
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   379844 B ( 370.9 KB)
  Free Bytes        :   337836 B ( 329.9 KB)
  Allocated Bytes   :    34468 B (  33.7 KB)
  Minimum Free Bytes:   332160 B ( 324.4 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
     4 : LEDC[0][0]
     5 : LEDC[0][2]
============ After Setup End =============
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-06-10:
Thanks for re-testing. I will give it a try tomorrow and keep you updated about the issue.

P-R-O-C-H-Y, 2025-06-11:
I was able to reproduce the issue using ESP-IDF directly. I will report it to them.

zhengyangliu, 2025-06-11:
Thank you very much for your additional testing work. This should make it easier to locate this bug. It looks very much like a hardware bug of the chip, but I can be sure that I cannot trigger this problem using the 1.x core. Because I am not very familiar with the register configuration of esp32, I can't find the problem code by myself. I really hope this bug can be fixed soon.

zhengyangliu, 2025-06-11:
@P-R-O-C-H-Y Thank you very much for your additional testing work. This should make it easier to locate this bug. It looks very much like a hardware bug of the chip, but I can be sure that I cannot trigger this problem using the 1.x core. 
Because I am not very familiar with the register configuration of esp32, I can't find the problem code by myself. I really hope this bug can be fixed soon.

P-R-O-C-H-Y, 2025-06-12:
Btw. this behavior only occurs also if the frequency of the 2nd channel is 52 of less Hz. All above is working fine without issues.

brightproject, 2025-07-15:
Here I configure `freeRTOS` and `PWM`😀

> https://github.com/pioarduino/platform-espressif32/issues/228