espressif/arduino-esp32 #11496: I2c error when I update esp32 board from 3.1.3 to 3.2.0

pvdw0310, 2025-06-21:
### Board

esp32 wroom 30pin

### Device Description

using an esp32 wroom with led ws2812 strip and touch sensors and an I2c lcddisplay 16*2 for a debug display

### Hardware Configuration

gpio21 and gpio22 is used as sda and scl
gpio26 as data for ws2812
gpio14 as input for a switch to detect high or low
gpio 32 and gpio33 as touch contact

### Version

v3.2.0

### IDE Name

arduino ide

### Operating System

windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

It is compiling and working perfectly when I compile it with arduino ide 1.8.19 and esp32 board version 3.1.3.
So I connect the lcd display and reset the sketch and i see all the debug messages.
If i disconnect the lcd display and reset the esp32 board, the sketch is working and there are no extra messages on the serial monitor.

If I upgrade the esp32 board to version 3.2.0 and recompile the sketch, then it is working with the display connected.
If I disconnect the display and after reset, then I get non stop messages on my serial monitor.

like:

15:05:34.739 -> E (2056) i2c.master: I2C hardware NACK detected
15:05:34.739 -> E (2056) i2c.master: I2C transaction unexpected nack detected
15:05:34.739 -> E (2056) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
15:05:34.786 -> E (2062) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
15:05:35.773 -> E (3069) i2c.master: I2C hardware NACK detected
15:05:35.773 -> E (3069) i2c.master: I2C transaction unexpected nack detected
15:05:35.773 -> E (3069) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
15:05:35.773 -> E (3074) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
15:05:35.773 -> E (3082) i2c.master: I2C hardware NACK detected
15:05:35.773 -> E (3086) i2c.master: I2C transaction unexpected nack detected
15:05:35.773 -> E (3091) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
15:05:35.820 -> E (3098) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed

and I did'nt had that with esp board version 3.1.3

Is it possible to check that issue and change the board manager so that we don't have that problem.
Many thanks.

Peter

### Sketch

```cpp
using:

#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcdsetup(0x27,16,2);
void setup(){
  Serial.begin(115200);
  delay(2000);                                        
  Serial.println("");
  Serial.println(__FILE__ "  " __DATE__);

  lcdsetup.init();           // initialize the lcdsetup
  lcdsetup.backlight();
  lcdsetup.clear();
  lcdsetup.setCursor(0, 0);
  lcdsetup.print("Opstarten");
}
```

### Debug Message

```plain
message on the serial monitor:

15:05:34.739 -> E (2056) i2c.master: I2C hardware NACK detected
15:05:34.739 -> E (2056) i2c.master: I2C transaction unexpected nack detected
15:05:34.739 -> E (2056) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
15:05:34.786 -> E (2062) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
15:05:35.773 -> E (3069) i2c.master: I2C hardware NACK detected
15:05:35.773 -> E (3069) i2c.master: I2C transaction unexpected nack detected
15:05:35.773 -> E (3069) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
15:05:35.773 -> E (3074) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
15:05:35.773 -> E (3082) i2c.master: I2C hardware NACK detected
15:05:35.773 -> E (3086) i2c.master: I2C transaction unexpected nack detected
15:05:35.773 -> E (3091) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
15:05:35.820 -> E (3098) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Discorez, 2025-06-25:
With version 3.2.0 I had these errors -

E (4270) i2c.master: I2C transaction unexpected nack detected E (4270) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failedE (4272) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failedE (11138) i2c.master: I2C transaction unexpected nack detectedE (11138) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failedE (11140) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failedE (15356) i2c.master: I2C transaction unexpected nack detectedE (15356) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failedE (15358) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failedE (19624) i2c.master: I2C transaction unexpected nack detectedE (19624) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failedE (19626) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed

After returning to version 3.1.3 all errors disappeared and everything works fine.

schneeer, 2025-07-08:
Ist zum kotzen und rücksichtslos, wie rigeros Espressif seine Software pflegt.
Plötzlich sind alle Arduino Sketche, welche seit 10 jahren funktionieren, nicht mehr zu gebrauchen.
Danke Espressif.

pvdw0310, 2025-10-02:
Hi, 
Yes, because of the many Serial reports, my program isn't working like it should be. ( to slow in some cases )
I'am using 1 template program as basic program and it is made to connect 2 lcd displays if necessary. So I don't had to change the programm if I use now 1 or 2 lcd's. It was always working. Now I have to check how many lcd boards are connected in the setup. I'am using it already a long time with no problem. Now I have to change all my programs because of the stricter control when the device is not connected. What is the advance of giving an error if it is not connected, you will see it also because there is nothing to see....