espressif/arduino-esp32 #11497: functional callbacks

jonnytest1, 2025-06-21:
### Related area

Wire

### Hardware specification

-

### Is your feature request related to a problem?

 
class MyClass:
{
 void setup()
 {

     i2cinput.onRequest([this]() {
       // dothings
     });
  }
};

 void (*user_onRequest)(void);
  void (*user_onReceive)(int);

  void onReceive(void (*)(int)) override;
  void onRequest(void (*)(void)) override;

### Describe the solution you'd like

this instead 🤔
    std::function<void()> user_onRequest;
    std::function<void(int)> user_onReceive;
    void onReceive(std::function<void(int)> function);
    void onRequest(std::function<void()> function);

### Describe alternatives you've considered

manually changing the lib locally

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2025-07-11:
@jonnytest1 - If possible, please test the PR #11582

jonnytest1, 2025-07-11:
if there arent any real changes between const ... & (which i wasnt familiar with before) and the one i proposed then i had tested it simply because i copied a modified version into my code ^^  so im pretty sure it still works 🤔 you probably also know more about backwards compaitibility concerns with this version if there are any 

i dont get any compiler errors with this version at least , not sure when i have time to do more in depth testing tho

SuGlider, 2025-07-12:
It works fine.

`std::function<>` passes by value -- copy takes ~32bytes with double copy cost and more stack usage (slower)
`const std::function<>&`passes by const reference -- no copy (~8 bytes reference) with a single copy cost and lower stack usage (faster)

PR #11582 (2025-07-12): feat(wire):  std::functional Wire slave callback functions
## Description of Change
Modify Wire.h and Wire.cpp to allow the Wire Slave callback functions to be declared as std::functional.
Adds example and changes the documentation.

## Tests scenarios
CI + ESP32/ESP32-C3/ESP32-C6/ESP32-S3  using new WireSlaveFunctionalCallback.ino example

``` cpp
#include "Wire.h"

#define I2C_DEV_ADDR 0x55

uint32_t i = 0;

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Wire.onRequest([]() {
    Wire.print(i++);
    Wire.print(" Packets.");
    Serial.println("onRequest");
  });
   [... cut]
closed 2025-07-12 by SuGlider