espressif/arduino-esp32 #11509: Esp32 core v3 compiles and uploads non-functioning binary

vnp85, 2025-06-23:
### Board

ESP32 dev modules, most every board I have

### Device Description

Physical board: many kinds of ESP32s, some modern, some older. It worked with a more recent camera module dev board, but failed with the  Heltec oled wifi boards / analogous. These board have the CP2012 implementation, USB-to-UART programming method. I didn't try with boards that feature native USB.

### Hardware Configuration

Nothing custom, all boards made and available in bulk, that have worked previously right out of the box.

### Version

v3.1.3

### IDE Name

Arduino IDE

### Operating System

win 11

### Flash frequency

tried with both 40 and 80

### PSRAM enabled

no

### Upload speed

115200 default

### Description

Nutshell version: source code, both complex custom code, then a simple custom hello world serial printf code and _**even the provided blinky**_ example, and physical boards that used to work, all of a sudden fail silently on a fresh arduino install, and I traced the issue to having esp32 core v3 instead of core v2 that's been on another machine I used to successfully program the same boards with the same source code, blinky, hello world.

Setup: new computer, fresh Arduino install, esp32 core (v3) grabbed from the board manager. 
Board selection from many: against many types, in particular Devkit v1, WROOM etc -- boards that I remembered were working.

Bug: compile anything, my own code, the provided blinky from the examples: no issue. Upload: no issue. Run? Nothing beyond the bootloader, no blink, no serial port, no guru crash core dump, just total silence. Sometimes I did see some garbage in the serial, sometimes it even looked like the wrong baudrate with the right timing, but overall I couldn't consistently reproduce the behavior, nor see anything meaningful with the logic analyzer, it just looked like garbage, or, more often, nothing at all. 

Switching back, manually, to ESP32 core v2: everything works, like it used to.

Somewhat similar to https://github.com/espressif/arduino-esp32/issues/9802 and what it links to, though still confused about the behavior of silent failure, of not inspecting the bootloader's output for potential incompatibilites etc. This is the reason I come to open the issue: the failure's nature. 

PS: last time I came, it was for a muted compiler warning that was perhaps turned into an error, for the code that it produced did crash the firmware on the spot (non-void function has no return value), which happened, I guess, due to changing the calling convention or something, unrelated.

### Sketch

```cpp
- stock blinky
- three liner hello world over serial
- complex custom firmware
```

### Debug Message

```plain
:: emptystring
```

### Other Steps to Reproduce

hello world over serial: 

have core v3 to fail,
have core v2 to succeed

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-06-24:
I am going to guess this is on devices with a 26MHz crystal, which is no longer supported in the default configuration. You can check the crystal rate with `esptool chip_id` (you can also see this on upload if you set that to verbose in preferences). I can't remember exactly which incantations you need to do to get 26MHz, but read through https://github.com/espressif/arduino-esp32/issues/9837

lbernstone, 2025-06-24:
The [heltec_wifi_kit_32](https://github.com/espressif/arduino-esp32/blob/master/variants/heltec_wifi_kit_32/pins_arduino.h#L10) and [heltec_wifi_lora_32](https://github.com/espressif/arduino-esp32/blob/master/variants/heltec_wifi_kit_32/pins_arduino.h#L10) have F_XTAL_MHZ set.

vnp85, 2025-06-25:
You are right there. As I mentioned, my goto board selection is/was "_Devkit v1, WROOM etc_", **perhaps my bad**, as these were the settings that worked with the boards I have, WROOM, WROVER dev boards and capsule modules without the breakout board, Heltec variants, etc. And still work with core v2.

https://github.com/espressif/arduino-esp32/blob/master/variants/doitESP32devkitV1/pins_arduino.h