espressif/arduino-esp32 #11551: How to read ESP32 SPI output timing half clock period earlier

sanjulanipun, 2025-07-03:
### Board

ESP32-WROOM-32

### Device Description

ESP32-WROOM-32

### Hardware Configuration

i have try VSPI & HSPI CS PIN 5 & 15

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

i have a few dirrent sd cards. some models are working with half clock period earlier. 

**### Current Working Normal SD Card Timing Output**
[image]

**### Not Working SD Card half clock period earlier**
[image]

> The response timing for ACMD41 , which is used to detect the completion status of initialization, value 0x00 is for completion. 
> From the timing capture by logic analyzer, it is seen that response data comes half clock period Working SD Card, so it is highly suggested that sampling response data 
> Not Working with half clock period earlier . otherwise, it is possible to get a wrong 
> data of 0x01 instead of 0x00 . 
> This response output is on the DO pin in this case.

<br>

**### Current Working Normal SD Card Response Timing**
[image]

**### Not Working Normal SD Card Response Timing**
[image]

> These example are all the response timing for CMD8 , which is used to send SD card interface condition , value 0x08 is the correct response value . 
> From the timing capture by logic analyzer, it is seen that response data comes half clock period 
> earlier for Notworking SD Card, so it is highly suggested that sampling response data 
> for Notworking SD Card with half clock period earlier . otherwise, it is possible to get a wrong 
> data of 0x10 instead of 0x08 . 
> This response output is on the CMD pin in this case

How I can fix with Not Working SD card SPI read write half clock period earlier ? can anyone to help with this ?

### Sketch

```cpp
#include "FS.h"
#include "SD.h"
#include "SPI.h"

void setup() {
  Serial.begin(115200);
  if(!SD.begin()){
    Serial.println("Card Mount Failed");
    return;
  }

  uint8_t cardType = SD.cardType();
  if(cardType == CARD_NONE){
    Serial.println("No SD card attached");
    return;
  }

  Serial.print("SD Card Type: ");
  if(cardType == CARD_MMC){
    Serial.println("MMC");
  } else if(cardType == CARD_SD){
    Serial.println("SDSC");
  } else if(cardType == CARD_SDHC){
    Serial.println("SDHC");
  } else {
    Serial.println("UNKNOWN");
  }

  uint64_t cardSize = SD.cardSize() / (1024 * 1024);
  Serial.printf("SD Card Size: %lluMB\n", cardSize);

  // Write to file
  File file = SD.open("/hello.txt", FILE_WRITE);
  if(!file){
    Serial.println("Failed to open file for writing");
    return;
  }
  file.println("Hello from ESP32");
  file.close();
  Serial.println("File written");

  // Read file
  file = SD.open("/hello.txt");
  if(!file){
    Serial.println("Failed to open file for reading");
    return;
  }
  Serial.println("Reading file:");
  while(file.available()){
    Serial.write(file.read());
  }
  file.close();
}

void loop() {
  // Nothing to do here
}
```

### Debug Message

```plain
Card Mount Failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-07-03:
it looks like they are working in a different SPI mode all together... @P-R-O-C-H-Y PTAL if you have ideas

heidnerd, 2025-08-05:
I think I am seeing the same issue.   I do not have screen captures.   But I've been using Seeed Xiao SAMD21 (first XIAO),  ESP32-C3, ESP32-C6,  ESP32-S3+ and RP2350 versions of the XIAOs.  Pull one module out - drop onto board and recompile.   The ESP32-S3 version can't see LCD screen when the newer SPI libraries are used - but OLD SPI libraries that were cross platform and work on the other chips -- allow that LCD screen to be seen.

But big issue is SD card won't mount.  And I've slowly been working through the board, software libraries and analyzer traces and it looks like timing issue related to the MUCH faster cores.

closed 2026-05-11 by github-actions[bot]