espressif/arduino-esp32 #11554: Zigee Error - Crash

def1149, 2025-07-04:
### Board

Adafruit Feather ESP32-C6 

### Device Description

Occasionally, my Deep Sleep Zigbee apps crash during the Zigbee.begin() call. I finally caught the error in a log

I'm using library 3.2.1, but the error also occurred when using 3.2.0 .

This may occur anytime from 30 minutes to a few days. The app wakes up every 180 seconds and checks the temperature and attempts a Zigbee connection every 3 to 15 minutes, depending on the temperature value. I'm using the analog endpoint.
'''
2025-07-04 10:38:27 [  3779][V][ESP32_C6H2_DeepSleep_Temperature_Sensor_V5.ino:365] initializeZigbee(): 
2025-07-04 10:38:27 Model: Temp Sensor V5 Jul  3 2025
2025-07-04 10:38:27 
2025-07-04 10:38:27 [  3790][V][ESP32_C6H2_DeepSleep_Temperature_Sensor_V5.ino:381] initializeZigbee(): Add Endpoints
2025-07-04 10:38:27 
2025-07-04 10:38:27 [  3801][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 1, Device ID: 0x000c
2025-07-04 10:38:27 [  3808][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 2, Device ID: 0x000c
2025-07-04 10:38:27 [  3815][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 3, Device ID: 0x000c
2025-07-04 10:38:27 [  3822][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 4, Device ID: 0x000c
2025-07-04 10:38:27 [  3829][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 5, Device ID: 0x000c
2025-07-04 10:38:27 [  3836][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 6, Device ID: 0x000c
2025-07-04 10:38:27 [  3844][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 8, Device ID: 0x000c
2025-07-04 10:38:27 [  3851][V][ESP32_C6H2_DeepSleep_Temperature_Sensor_V5.ino:438] initializeZigbee(): Begin Zigbee
2025-07-04 10:38:27 
2025-07-04 10:38:27 [  3864][D][ZigbeeCore.cpp:148] zigbeeInit(): Initialize Zigbee stack
2025-07-04 10:38:27 [  3887][D][ZigbeeCore.cpp:155] zigbeeInit(): Register all Zigbee EPs in list
2025-07-04 10:38:27 [  3901][I][ZigbeeCore.cpp:163] zigbeeInit(): List of registered Zigbee EPs:
2025-07-04 10:38:27 [  3908][I][ZigbeeCore.cpp:165] zigbeeInit(): Device type: Simple Sensor device, Endpoint: 1, Device ID: 0x000c
2025-07-04 10:38:27 [  3918][I][ZigbeeCore.cpp:165] zigbeeInit(): Device type: Simple Sensor device, Endpoint: 2, Device ID: 0x000c
2025-07-04 10:38:27 [  3928][I][ZigbeeCore.cpp:165] zigbeeInit(): Device type: Simple Sensor device, Endpoint: 3, Device ID: 0x000c
2025-07-04 10:38:27 [  3938][I][ZigbeeCore.cpp:165] zigbeeInit(): Device type: Simple Sensor device, Endpoint: 4, Device ID: 0x000c
2025-07-04 10:38:28 [  3948][I][ZigbeeCore.cpp:165] zigbeeInit(): Device type: Simple Sensor device, Endpoint: 5, Device ID: 0x000c
2025-07-04 10:38:28 [  3958][I][ZigbeeCore.cpp:165] zigbeeInit(): Device type: Simple Sensor device, Endpoint: 6, Device ID: 0x000c
2025-07-04 10:38:28 [  3968][I][ZigbeeCore.cpp:165] zigbeeInit(): Device type: Simple Sensor device, Endpoint: 8, Device ID: 0x000c
2025-07-04 10:38:28 [  3997][V][ZigbeeCore.cpp:444] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
2025-07-04 10:38:28 [  4007][I][ZigbeeCore.cpp:257] esp_zb_app_signal_handler(): Zigbee stack initialized
2025-07-04 10:38:28 [  4015][D][ZigbeeCore.cpp:258] esp_zb_app_signal_handler(): Zigbee channel mask: 0x02000000
2025-07-04 10:38:28 [  4326][V][ZigbeeCore.cpp:444] esp_zb_app_signal_handler(): ZDO signal: NLME Status Indication (0x32), status: ESP_OK
2025-07-04 10:38:28 [  4337][W][ZigbeeCore.cpp:292] esp_zb_app_signal_handler(): Commissioning failed, trying again...
2025-07-04 10:38:28 [  4598][W][ZigbeeCore.cpp:292] esp_zb_app_signal_handler(): Commissioning failed, trying again...
2025-07-04 10:38:29 ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42001772
file: "C:\Users\occam\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.1\libraries\Zigbee\src\ZigbeeCore.cpp" line 240
func: void bdb_start_top_level_commissioning_cb(uint8_t)
expression: esp_zb_bdb_start_top_level_commissioning(mode_mask)
ESP-ROM:esp32c6-20220919
2025-07-04 10:38:29 Build:Sep 19 2022
2025-07-04 10:38:29 rst:0xc (SW_CPU),boot:0x8 (SPI_FAST_FLASH_BOOT)
2025-07-04 10:38:29 Saved PC:0x4001975a
2025-07-04 10:38:29 SPIWP:0xee
2025-07-04 10:38:29 mode:DIO, clock div:2
2025-07-04 10:38:29 load:0x40875720,len:0x1260
2025-07-04 10:38:29 load:0x4086c110,len:0xdd0
2025-07-04 10:38:29 load:0x4086e610,len:0x3030
2025-07-04 10:38:29 entry 0x4086c110
2025-07-04 10:38:27 [  3779][V][ESP32_C6H2_DeepSleep_Temperature_Sensor_V5.ino:365] initializeZigbee(): 
2025-07-04 10:38:27 Model: Temp Sensor V5 Jul  3 2025
2025-07-04 10:38:27 
2025-07-04 10:38:27 [  3790][V][ESP32_C6H2_DeepSleep_Temperature_Sensor_V5.ino:381] initializeZigbee(): Add Endpoints
2025-07-04 10:38:27 
2025-07-04 10:38:27 [  3801][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 1, Device ID: 0x000c
2025-07-04 10:38:27 [  3808][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 2, Device ID: 0x000c
2025-07-04 10:38:27 [  3815][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 3, Device ID: 0x000c
2025-07-04 10:38:27 [  3822][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 4, Device ID: 0x000c
2025-07-04 10:38:27 [  3829][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 5, Device ID: 0x000c
2025-07-04 10:38:27 [  3836][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 6, Device ID: 0x000c
2025-07-04 10:38:27 [  3844][D][ZigbeeCore.cpp:100] addEndpoint(): Endpoint: 8, Device ID: 0x000c
2025-07-04 10:38:27 [  3851][V][ESP32_C6H2_DeepSleep_Temperature_Sensor_V5.ino:438] initializeZigbee(): Begin Zigbee
2025-07-04 10:38:27 
[... 155 lines cut ...]
  logLocation(9);
  #endif
  if (!Zigbee.begin(&zigbeeConfig, false)) {
  #if defined(LOG_LOCATION)
  logLocation(10);
  #endif

#if defined(SERIAL_LOG)
    log_w("Zigbee failed to start @ cycle: %d\r\n",cycleCount);
#endif

#if defined(LOG)
     myLog.print("Failed to start @ cycle: " ); myLog.println(cycleCount);
#endif
```

### Debug Message

```plain
see above
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-07-07:
Hi @def1149, Thank you for the report. I will take a look

def1149, 2025-07-07:
Hi @P-R-O-C-H-Y  Update. I reduced the debug level to debug [D] from Verbose [V] to see if that may have been a contributing factor, but the errors still occur. It may take hours before these occur, so you may need to run your tests for a long time. I have replicated the problem on multiple Feather C6 boards.

NB. I'm repeating this on an Olimex H2 board

def1149, 2025-07-07:
Hi @P-R-O-C-H-Y I can reproduce the problem on the Olimex H2 board

huyisb, 2025-07-08:
I ran into the same problem on ESP32C6 Supermini using the example sketch pair: Zigbee_Temperature_Sensor and Zigbee_Thermostat.

def1149, 2025-07-25:
This problem appears to be fixed. After running a stress test for 24 hours, which sleeps for 30 seconds between waking up and sending results, not a single crash occurred.

PR #11614 (2025-07-21): fix(zigbee): Replace assert with error log to solve immediate crash
## Description of Change
Replace `ESP_ERROR_CHECK` with proper error handling to prevent system crashes when `esp_zb_bdb_start_top_level_commissioning()` returns `ESP_FAIL` due to invalid commissioning state (can be in progress). This allows the existing retry mechanism to continue working while providing better error logging instead of assertion failures.

Same way its now done in esp-zigbee-sdk examples:
```cpp
static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_ [... cut]