espressif/arduino-esp32 #11555: Zigbee_On-Off_MultiSwitch example problems

dbn-b4e, 2025-07-05:
### Board

ESP32C6

### Device Description

Button and RGB led onboard

### Hardware Configuration

ESP32C6Zero (Waveshare)

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Mac OSX 15.1

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921000

### Description

Hello,

I have tested to create ESP32C6 zigbee device with light, temperature and input switch at the same time. Basically it works. Temperature and Light are fully ok. While I tested many things to get the switch to work without any success.

I want it to work with Z2M and not as local coordinator.

First problem from the example is zbSwitch.setManualBinding(true); that does not exist as method. Strange. Looked at all repos and no clue.

Second problem is it reports as IAS with alarm1, alarm2 what is in the code. How to create a simple (or multiple) wall input buttons with simple, double or whatever function? I do not see it in any example. Seems to me a basic function but I cannot find anything working in Arduino IDE with the wrappers.

Third problem is the ROUTER mode, if I try the MultiSwitch example with  Coordinator in ESP stack config and start with Zigbee.begin(ZIGBEE_ROUTER) it crashes at startup.

If anybody has any idea... mmany thanks. Library ESP is latest 3.3.0 alpha 1.

### Sketch

```cpp
Here is the global sketch:

// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.
[... 219 lines cut ...]
  // Send temperature each 10 seconds
  static uint32_t lastPrint = 0;
  if (millis() - lastPrint > 10000) {
    lastPrint = millis();
    zbTempSensor.reportTemperature();
  }
}

[Zigbee_Sensor.txt](https://github.com/user-attachments/files/21077061/Zigbee_Sensor.txt)
```

### Debug Message

```plain
[  1336][E][ZigbeeContactSwitch.cpp:82] report(): Failed to send IAS Zone status changed notification: 0x2: ERROR
Switch true
```

### Other Steps to Reproduce

No additional steps

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dbn-b4e, 2025-07-07:
Hello,

Many thanks for your fast response! You rock!

In order to be clear about my goal:
1. I originaly wanted to use PVTO firmware but...
2. It is closed firmware and the developer did not wanted to add a function to driver one (or many) I/O expander like PCA9554
3. I am a hardware developer and had like to bring a lot my 1000+ hardware under Altium to the open source community
4. We are using 20.000+ ESP32-S3 for our commercial product (OCPP EVSE charger with 4G)
5. So basically I will open a discussion (I don't know where nor how... if you have an idea) in order to decide with interested people to define the "best" playground for ESP32-C6 and Zigbee at hardware level. I mean for example:
 * External antenna (the waveshare ESP32-C6-Zero antenna is a real drama)
 * 8 input buttons (hence the "Moes" like questions to have home push buttons inputs)
 * 8 relay outputs
 * ATM90E26 or other energy meter
 * 1-wire bus
 * RS485 modbus
 * PCF RTC I2C
 * ... other things people would like
 * In a 24 or 230V DIN rail box

If I build this, would you need one to be able to test your code, would be happy to send you one.

Now this is explained, regrading your answers:

1. I don't have Coordinator/router in Arduino IDE, on which version are you? I am on 2.3.6 with ESP version 3.3.0 alpha. ANy idea?
2. If I do compile with coordinator and ROUTER in setup it enter a boot loop, is it because I have 4MB only? Strange since it compiles.
3. I am using Z2M with dozens of device without any problem. I can setup raspi with new HA and use ZHA in order to do a quick test.
4. With the "Multiswicth" and Z2M unable to have any trigger. 
5. For the "smart button" I really don't understand why such a thing common to hundred of devices is not in the spec... Do you have a plan to do such thing, a lot of people are looking for it on forums and nobody manages to do so, be it under vscode or Arduino...
6. For the ROUTER thing it is setup as all project (working) I am trying out.
7. I tested the code from one of the "issues" from https://github.com/espressif/arduino-esp32/discussions/11112 and it is ok (still the [mvermand22] open question in the issue). But it is a on/off (tested with toggle) and does not reports triggers.

Again, many thanks for you time and developments.

David

dbn-b4e, 2025-07-08:
> P-R-O-C-H-Y
>  left a comment
> (espressif/arduino-esp32#11555)
> @dbn-b4e <https://github.com/dbn-b4e> You can join our discord server <https://discord.gg/8xY6e9crwv> and you can ask the community there about the ideas or features they would like to have on the board :) Also you can ping me there directly so we can set a 1:1 private communication (I don't see your email :D)

Jan,

My email is in the issue: ***@***.*** ***@***.***>

Ir. David Baldwin
B4E SRL - Administrateur
40, Rue de l'Etang
B-1410 Waterloo
Belgium
Mob:  +32 477 56 29 78 <tel://+32 477 56 29 78>
Mail:   ***@***.*** ***@***.***>
VAT :  BE0635.630.310

Please consider our environment before printing this e-mail
 
Disclaimer:
 
This message and any attachments thereto are confidential. They may also be privileged or otherwise protected by work product immunity or other legal rules. If you have received it by mistake, please let us know by e-mail reply and delete it from your system; you may not copy this message or disclose its contents to anyone. E-mail transmission cannot be guaranteed to be secure or error free as information could be intercepted, corrupted, lost, destroyed, arrive late or incomplete, or contain viruses. The sender therefore is in no way liable for any errors or omissions in the content of this message, which may arise as a result of e-mail transmission. If verification is required, please request a hard copy.
Cet email, ainsi que ses annexes, est confidentiel. Il contient des informations particulières destinées à l'usage exclusif de celui à qui il est adressé. Si ce message est reçu par quelqu'un d'autre que son destinataire, merci de le retourner simplement en y répondant et ensuite de le supprimer de votre ordinateur. Si vous n'êtes la personne à qui ce message est destiné, vous ne pouvez en faire de copie, le faire circuler ou en divulguer l'existence à qui que ce soit. Si le présent message est incomplet ou incorrect, merci de le retourner à son expéditeur.

P-R-O-C-H-Y, 2025-07-08:
[image]

closed 2025-07-19 by P-R-O-C-H-Y