espressif/arduino-esp32 #11557: ULP ADC read fails

ayavilevich, 2025-07-06:
### Board

ESP32 generic dev module

### Device Description

ESP32 generic dev module. Nothing additional connected.

### Hardware Configuration

Nothing additional connected.

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE 2.3.6 and Platform.IO on VSC

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Since Arduino-ESP32 2.0.14, using an ADC read operation in ESP32 ULP FSM co processor will cause the ULP to get "stuck".

I am including minimal code to reproduce.

Expected: wake due to an ULP wake after 5 seconds.  
Actual: wakes due to (a fallback) timer wake after 20 seconds.

If using this code with ESP-IDF directly then it works as expected.  
If using this code in a version before 2.0.14, then it works as expected.  
If disabling the ADC read operation in the ULP code, then it works as expected.  

More details at https://github.com/ayavilevich/ulp_adc_issue_ard_min/blob/main/README.md

### Sketch

```cpp
https://github.com/ayavilevich/ulp_adc_issue_ard_min/blob/main/src/main.cpp

Above can can be used with Arduino IDE, PIO or ESP-IDF.
```

### Debug Message

```plain
Output, Arduino ESP32 3.2.0

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
ESP-IDF: 5.4.1, ESP Arduino: 3.2.0
Start
Chip revision: 301
Not ULP/timer wakeup
TEST: sleeping for 20 sec
Entering deep sleep
ets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
ESP-IDF: 5.4.1, ESP Arduino: 3.2.0
Start
Chip revision: 301
Timer wakeup
Deep sleep wakeup
ULP do adc 1
ULP start counter: 2
ULP wake counter: 2
TEST: sleeping for 20 sec
Entering deep sleep

Output, Arduino ESP32 3.2.0 with ADC read disabled

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
ESP-IDF: 5.4.1, ESP Arduino: 3.2.0
Start
Chip revision: 301
Not ULP/timer wakeup
TEST: sleeping for 20 sec
Entering deep sleep
ets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
ESP-IDF: 5.4.1, ESP Arduino: 3.2.0
Start
Chip revision: 301
ULP wakeup
Deep sleep wakeup
ULP do adc 0
ULP start counter: 51
ULP wake counter: 50
TEST: sleeping for 20 sec
Entering deep sleep

Output, ESP-IDF 4.4.6

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5484
ho 0 tail 12 room 4
load:0x40078000,len:14824
ho 0 tail 12 room 4
load:0x40080400,len:3460
0x40080400: _init at ??:?

entry 0x40080638
ESP-IDF: 4.4.6
Start
Chip revision: 3, full: 301
Not ULP/timer wakeup
TEST: sleeping for 20 sec
Entering deep sleep
ets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5484
ho 0 tail 12 room 4
load:0x40078000,len:14824
ho 0 tail 12 room 4
load:0x40080400,len:3460
0x40080400: _init at ??:?

entry 0x40080638
ESP-IDF: 4.4.6
Start
Chip revision: 3, full: 301
ULP wakeup
Deep sleep wakeup
ULP do adc 1
ULP start counter: 51
ULP wake counter: 50
TEST: sleeping for 20 sec
Entering deep sleep
```

### Other Steps to Reproduce

also tried on another ESP32 board. it made no difference.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-07-06:
I don't see the settings `ulp_low_thr = 1500;  ulp_high_thr = 2000` in your code.

ayavilevich, 2025-07-06:
> I don't see the settings `ulp_low_thr = 1500; ulp_high_thr = 2000` in your code.

There are no such settings in this project. You are probably confusing this with the "ulp_adc" example in ESP-IDF SDK.
examples/system/ulp/ulp_fsm/ulp_adc/main/ulp_adc_example_main.c

This POC is somewhat based on that code but the ULP program is different. It is ULP MACROs instead of ULP assembly, so there are no "c variables". You have 4 settings in RTC_SLOW_MEM. See:

```
#define U_MEM_START_COUNTER 0 // use as index to RTC_SLOW_MEM
#define U_MEM_WAKEUP_COUNTER 1
#define U_MEM_WAKEUP_COUNTER_GOAL 2
#define U_MEM_DO_ADC 3
```

I couldn't find a way to run the ESP-IDF ulp_adc example as-is on Arduino.

ayavilevich, 2025-07-06:
Those thresholds are used in another project where the ULP code compares the ADC reading with the thresholds. In the POC of this issue the value of the ADC is not used nor compared with anything.

This reproduce code is a different ulp program than the one in ulp_adc.

SuGlider, 2025-07-15:
@ayavilevich - This is not an Arduino issue. It is about ULP ASM and how to implement a specific project. I'll move it to discussion.

I think that the ULP has never completed running the ASM code.
It seems to miss `esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);` in order to define how the power domain for the ULP will behave when the ESP32 goes into deep sleep. By default, when the ESP32 deep sleeps, the ULP is not powered.

But I didn't try out the project, just took a quick look.
ULP will run at about 8MHz. It is slow. The ULP is started and right after that the ESP32 goes into deep sleep, which may stop the ULP automatically. 

You can check this example from Arduino to see how the ESP32 ULP is used (maybe already done it?)
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/DeepSleep/SmoothBlink_ULP_Code/SmoothBlink_ULP_Code.ino