espressif/arduino-esp32 #11573: How to connect to more than 3 devices

Grandpa-G, 2025-07-08:
### Board

CoreS2

### Device Description

Just the CoreS2

### Hardware Configuration

Just the CoreS2

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE 1.8.x

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Using a CoreS2 (ESP32) with Arduino IDE. I cannot make more than 3 connections simultaneously. I need 4. Is there a way to make this work.

### Sketch

```cpp
`String wheel[] = {"35:c6:fe:1a:1d:47",
                    "bc:5c:43:78:a1:33",
                    "65:fb:b5:43:7f:d1",
                    "d5:57:2c:60:ef:d3"
                   };

  for (int i = 0; i < 4; i++)
  {
    Serial.print(wheel[i]);

    BLEClient* thisClient = BLEDevice::createClient();
    Serial.println(" - Created client ");

    bool cRet = thisClient->connect(wheel[i]);
    Serial.print("RET:"); Serial.println(cRet);
    if (!cRet) {
      Serial.println("Connection failed 1");
     else
      Serial.println(" - Connected to server");
  }`
```

### Debug Message

```plain
None other than the 4th connection fails and prints Connection failed 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Grandpa-G, 2025-07-08:
Am I correct in understanding doing all this moves me from using the Arduino IDE?

lbernstone, 2025-07-08:
Yes, if you wish to continue using Arduino IDE, you can use [lib-builder](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html), as mentioned in the first link I posted above.  There is no trivial method for doing this, but lib-builder is not too taxing.

Grandpa-G, 2025-07-08:
Isn't there any simplier methods to change that one BLE constant from 3 to 4?

SuGlider, 2025-07-09:
>The lib-builder seems to only to apply to Linux type enviornments, I am using IDE in windows.

This is correct, but with Windows it may be possible to use WSL Ubuntu.

SuGlider, 2025-07-09:
> Isn't there any simplier methods to change that one BLE constant from 3 to 4?

I'm afraid that there is no simple way to make it.
Only 2 options: Lib-Builder or Arduino as IDF Component.

Grandpa-G, 2025-07-09:
In my opinion, neither option is very good. This would mean completely changing an existing application development project.

Why was 3 connections choosen? Why not 2 or 4. 3 seems very arbitrary.

For this project, an new strategy will have to be developed.

SuGlider, 2025-07-09:
The answer for the number of connections is that it is the IDF default that has not been changed by Arduino Core Project.
Therefore, nothing was choosen, just built with default parameters.