espressif/arduino-esp32 #11579: partitions.csv ignored in Arduino IDE v1 and v2

MikeyMoMo, 2025-07-10:
### Board

ESP32 S3

### Device Description

The target board is a T-Display S3 with 16MB.  The goal is a 3M program partition (app0) and a 12M LittleFS/spriffs partition.  No app1, no factory.  Maximum storage to the LittleFS/spiffs partition.

### Hardware Configuration

n/a

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE v1 and v2 (both fail the same)

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am not the only person who has reported this.

partitions.csv in sketch folder:

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 3M,
spiffs,   data, spiffs,         , 12M,
coredump, data, coredump,       , 0x10000,

```
This does not quite fill all 16M of memory.  There is another but when the memory is filled, exactly.  Will submit that, later.  

Output from \gen_esp32part.py on the created .bin compile of the partitions.csv looks good:
```
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,20K,
otadata,data,ota,0xe000,8K,
app0,app,ota_0,0x10000,3M,
spiffs,data,spiffs,0x310000,12M,
coredump,data,coredump,0xf10000,64K,
```

I have received useless guidance that there is some magic character in here, somewhere, that, if missing, causes this file to be ignored.  I am not sure what that even means and it does not apply.  When I use this SAME file (not copy and paste but move the file and rename it) to the partitions area, and select it by name, it works without any change to the contents.  To get around this problem, I renamed the custom file to one of the standard names and I select it to get the needed result.  Of course, this is a temporary fix since any updating will destroy it.

Partition definition selected in Tools Menu>Partition Scheme (should not matter):
`Default 4MB with ffat (1.2MB APP/1.5MB FATCS)
`
This was chosen to show that it is being selected instead of the partitions.csv in the sketch folder.

The copied partitions.csv text file is correct in the temp build area as well as the compiled partitions.bin file. (C:\Users\bangu\AppData\Local\Temp\arduino_build_466306)

Short result:

> Sketch uses 1014314 bytes (77%) of program storage space. Maximum is 1310720 bytes.

This is not correct.  It should be a 3M app0.  Also, if this is loaded, it destroys the 12M LittleFS/spiffs partition that is currently loaded.

### Sketch

```cpp
Does not matter.  It will not run.  The error happens during compile time.  The important bit is the mismatch between the required partitions size set forth in the partitions.csv and the actual created app0 size.
```

### Debug Message

```plain
It is not runnable so no Guru error.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-07-10:
> Sketch uses 1014314 bytes (77%) of program storage space. Maximum is 1310720 bytes.

This message depends on the partition scheme selected in the board menu. We have no way to tell the IDE and Arduino builder what is the app size of the partition in your sketch folder, so select a partition scheme that matches or is bigger than your custom one

On another note I did the following test yesterday:
- Copied `huge_app.csv` from the core to `partitions.csv` in the sketch folder
- Selected the default partition in the board menu
- Uploaded the firmware (blink sketch) with Core Debug: Verbose
- Sketch was uploaded and the debug indicated that it was running the `huge_app.csv` partition scheme
```
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  3072.0 KB, type:  APP, subtype: OTA_0
             spiffs : addr: 0x00310000, size:   896.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
```

I can not confirm your issue. Everything is working as expected

me-no-dev, 2025-07-10:
And please fill the issue form diligently! All current versions of the core are listed and you did not select the one you are running