espressif/arduino-esp32 #11585: Compilation error: exit status 101

HomeMade-Channel, 2025-07-11:
### Board

ESP32

### Device Description

Any ESP32 devices

### Hardware Configuration

Any ESP32 devices

### Version

v3.2.1

### IDE Name

Arduino IDE 2.3.6

### Operating System

Win 10 x64

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Good day.
On all kernel versions 3.1.0 - 3.2.1, when compiling any sletch, an error occurs:

```
thread 'main' panicked at 'assertion failed: (left != right)
  left: 0,
 right: 0: Failed to get path name. Error code: 3', main.rs:65:9
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
exit status 101

Compilation error: exit status 101
```
On earlier kernel versions (3.0.7 and below) there is no error.
The error does not depend on the Arduino IDE version.

### Sketch

```cpp
void setup() {
  pinMode(4, OUTPUT);
}

void loop() {
  digitalWrite(4, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(4, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
```

### Debug Message

```plain
thread 'main' panicked at 'assertion failed: (left != right)
  left: 0,
 right: 0: Failed to get path name. Error code: 3', main.rs:65:9
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
exit status 101

Compilation error: exit status 101
```

### Other Steps to Reproduce

I have tried different versions of Arduino IDE, also tried different versions of ESP32 core.
On earlier kernel versions (3.0.7 and below) there is no error.
The error does not depend on the Arduino IDE version. Any sketch, even from examples, gives an error.
It is worth noting that I use Cyrillic in the user path:
_C:\Users\**Игорь**\Documents\Arduino\_
But this did not cause any problems before.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-07-11:
There is no rust anywhere in arduino-esp32 and AFAIK in Arduino IDE.  That must be coming from something you have added.  Set compile debug output enabled in preferences and post a larger section of the log which shows what command throws this error.

Jason2866, 2025-07-13:
It was a suggestion to track down the problem. Maybe this is not the reason. Idk. If you want a solution you have to try.

lucasssvaz, 2025-07-14:
@HomeMade-Channel Which boards did you use to test ?

lucasssvaz, 2026-09-25:
4.0.0-RC1 have the new toolchains and should have this issue fixed. Please reopen if you encounter it in 4.0.x

closed 2026-09-25 by lucasssvaz