espressif/arduino-esp32 #11596: Can't make Arduino-esp32 as a global components

xixixihaha, 2025-07-13:
### Board

ESP32C3

### Device Description

I used generic ESP32-C3 board. But it should be a common issue not related to hardware

### Hardware Configuration

No need for hardware config

### Version

latest stable Release (if not listed below)

### IDE Name

Latest VScode, ESP-IDF 

### Operating System

Ubuntu desktop 24.04

### Flash frequency

No change

### PSRAM enabled

no

### Upload speed

460800

### Description

I've followed the manual to add Arduino-esp32 as global component. During the compilation, the process ended with the following error.

/home/simon/esp/v5.4.2/esp-idf/components/mbedtls/port/include/aes/esp_aes.h:27:10: fatal error: hal/aes_types.h: No such file or directory
   27 | #include "hal/aes_types.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.

I moved the Arduino-esp32 to the project components folder, the error disappeared and the project can be compiled successfully and the code worked like a charm on the board.

### Sketch

```cpp
#include "Arduino.h"
#define BLUE_LED     4
#define GREEN_LED    5
#define RED_LED      7

//file: main.cpp
extern "C" void app_main()
{
  initArduino();

  // Arduino-like setup()
  Serial.begin(115200);
  while(!Serial){
    ; // wait for serial port to connect
  }
  pinMode(BLUE_LED, OUTPUT);
  pinMode(GREEN_LED, OUTPUT);
  pinMode(RED_LED, OUTPUT);
  digitalWrite(BLUE_LED, HIGH);
  digitalWrite(GREEN_LED, HIGH);
  digitalWrite(RED_LED, HIGH);
  // Arduino-like loop()
  while(true){
    digitalWrite(BLUE_LED, HIGH);
    delay(500);
    digitalWrite(BLUE_LED, LOW);
    delay(500);
  }
  // WARNING: if program reaches end of function app_main() the MCU will restart.
}

//file: main.cpp
// void setup(){
//   Serial.begin(115200);
//   while(!Serial){
//     ; // wait for serial port to connect
//   }
//   pinMode(BLUE_LED, OUTPUT);
//   pinMode(GREEN_LED, OUTPUT);
//   pinMode(RED_LED, OUTPUT);
//   digitalWrite(BLUE_LED, HIGH);
//   digitalWrite(GREEN_LED, HIGH);
//   digitalWrite(RED_LED, HIGH);
// }

// void loop(){
//   digitalWrite(GREEN_LED, HIGH);
//   delay(500);
//   digitalWrite(GREEN_LED, LOW);
//   delay(500);
// }
```

### Debug Message

```plain
/home/simon/esp/v5.4.2/esp-idf/components/mbedtls/port/include/aes/esp_aes.h:27:10: fatal error: hal/aes_types.h: No such file or directory
   27 | #include "hal/aes_types.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-07-13:
What manual did you follow? Did you [create a project from a template](https://github.com/espressif/arduino-esp32/tree/master/idf_component_examples/hello_world#how-to-use-example)?

SuGlider, 2025-07-15:
Given that the issue was solved by the user, I'll close it.