espressif/arduino-esp32 #11597: Zigbee - Data missing in Home Assistant

def1149, 2025-07-13:
### Board

Adafruit ESP36-C6

### Device Description

Adafruit ESP32-C6 - Nothing attached 

### Hardware Configuration

No

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino 2.3.5

### Operating System

Windows 11 

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have been occasionally missing expected data in Home Assistant using ZHA integration, but it's been difficult to see consistently. I created the following test program that sends 8 analog values every 3 minutes, tand deep sleeps between cycles.
The data is created so that each value increments by 1 for each cycle, and all values are offset from each other by 2. This is to ensure no duplicates are created and all values are monotonically increasing. This data is charted, and the missing data is obvious
 

### Sketch

```cpp
/// Test Zigbee sleopy sensor with 8 Analog values
/// ---------------------------------------------------
/// !3-JUL-2025
/// Demonstrate missing values in Home Assistant

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"
#include <rom/rtc.h>
#include <esp_task_wdt.h>
#include <Preferences.h>

// Hardware pins
#define BOOT_PIN 9 // Boot button for C6/H2

// Zigbee endpoint configuration
#define ANALOG_DEVICE_ENDPOINT_NUMBER 10

// Zigbee objects 
static ZigbeeAnalog zbA1(ANALOG_DEVICE_ENDPOINT_NUMBER);
static ZigbeeAnalog zbA2(ANALOG_DEVICE_ENDPOINT_NUMBER + 1);
static ZigbeeAnalog zbA3(ANALOG_DEVICE_ENDPOINT_NUMBER + 2);
static ZigbeeAnalog zbA4(ANALOG_DEVICE_ENDPOINT_NUMBER + 3);
[... 415 lines cut ...]
11:11:14.666 -> [  1301][D][ZigbeeAnalog.cpp:186] setAnalogInput(): Setting analog input to 8.0
11:11:14.666 -> [  1308][D][ZigbeeAnalog.cpp:186] setAnalogInput(): Setting analog input to 10.0
11:11:14.666 -> [  1311][D][ZigbeeAnalog.cpp:186] setAnalogInput(): Setting analog input to 12.0
11:11:14.666 -> [  1317][D][ZigbeeAnalog.cpp:186] setAnalogInput(): Setting analog input to 14.0
11:11:14.666 -> [  1324][D][ZigbeeAnalog.cpp:186] setAnalogInput(): Setting analog input to 16.0
11:11:14.666 -> [  1328][D][ZigbeeAnalog.cpp:186] setAnalogInput(): Setting analog input to 18.0
11:11:15.211 -> [  1830][D][Test_multiple.ino:78] sleep(): Sleep: Timed, 180 seconds
11:11:15.211 ->
```

### Other Steps to Reproduce

Run the program and chart Float 1 through Float 8

This chart highlights the missing data ... the missing steps in th chart

The attached ZHA log is related to  the missing data highlighted by the yellow arrow 

[zha_log 13-JUL-2025.txt](https://github.com/user-attachments/files/21205787/zha_log.13-JUL-2025.txt)

[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-07-16:
This will be hopefully solved by implementing #11038.

def1149, 2025-07-25:
Fixed by #11038