espressif/arduino-esp32 #11612: [Zigbee] Multiple contact switches one ESP32 H2

M4rkoV2, 2025-07-19:
### Board

ESP32-H2 Super mini board

### Device Description

https://www.espboards.dev/esp32/esp32-h2-super-mini/

### Hardware Configuration

sensor_pin = 4
sensor2_pin = 5

### Version

v3.2.1

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

64Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I would like to use one ESP-H2 board and multiple contact switches, connect it over Zigbee to my Smartthings hub. Is that possible?
Example code works fine, but when I try to add more it does not compile. What am I doing wrong?

Code compiles fine and i can upload it. But ESP keep rebooting.

### Sketch

```cpp
// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates Zigbee contact switch (IAS Zone).
 *
 * The example demonstrates how to use Zigbee library to create a end device contact switch.
 * The contact switch is a Zigbee end device, which is reporting data to the Zigbee network.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 */

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

/* Zigbee contact sensor configuration */
#define CONTACT_SWITCH_ENDPOINT_NUMBER 10
#define CONTACT_SWITCH_ENDPOINT_NUMBER2 11
uint8_t button = BOOT_PIN;
uint8_t sensor_pin = 4;
uint8_t sensor2_pin = 5;

ZigbeeContactSwitch zbContactSwitch = ZigbeeContactSwitch(CONTACT_SWITCH_ENDPOINT_NUMBER);
ZigbeeContactSwitch zbContactSwitch2 = ZigbeeContactSwitch(CONTACT_SWITCH_ENDPOINT_NUMBER2);

void setup() {
  Serial.begin(115200);

  // Init button + switch
  pinMode(button, INPUT_PULLUP);
  pinMode(sensor_pin, INPUT_PULLUP);
  pinMode(sensor2_pin, INPUT_PULLUP);

  // Optional: set Zigbee device name and model
  zbContactSwitch.setManufacturerAndModel("Espressif", "ZigbeeContactSwitch");
  zbContactSwitch2.setManufacturerAndModel("Espressif", "ZigbeeContactSwitch2");

  // Add endpoint to Zigbee Core
  Zigbee.addEndpoint(&zbContactSwitch);
  Zigbee.addEndpoint(&zbContactSwitch2);

  Serial.println("Starting Zigbee...");
  // When all EPs are registered, start Zigbee in End Device mode
  if (!Zigbee.begin()) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  } else {
    Serial.println("Zigbee started successfully!");
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println();
}

void loop() {
  // Checking pin for contact change
  static bool contact = false;
  if (digitalRead(sensor_pin) == HIGH && !contact) {
    // Update contact sensor value
    zbContactSwitch.setOpen();
    contact = true;
  } else if (digitalRead(sensor_pin) == LOW && contact) {
    zbContactSwitch.setClosed();
    contact = false;
  }

  static bool contact2 = false;
  if (digitalRead(sensor2_pin) == HIGH && !contact2) {
    // Update contact sensor value
    zbContactSwitch2.setOpen();
    contact2 = true;
  } else if (digitalRead(sensor2_pin) == LOW && contact2) {
    zbContactSwitch2.setClosed();
    contact2 = false;
  }

  // Checking button for factory reset
  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(button) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.println("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
  }
  delay(100);
}
```

### Debug Message

```plain
16:14:57.851 -> ELF file SHA256: 9725cd921
16:14:57.851 -> 
16:14:57.851 -> Rebooting...
16:14:57.851 -> ESP-ROM:esp32h2-20221101
16:14:57.851 -> Build:Nov  1 2022
16:14:57.896 -> rst:0xc (SW_CPU),boot:0x8 (SPI_FAST_FLASH_BOOT)
16:14:57.896 -> Saved PC:0x400031b6
16:14:57.896 -> SPIWP:0xee
16:14:57.896 -> mode:DIO, clock div:1
16:14:57.896 -> load:0x408460e0,len:0x11fc
16:14:57.896 -> load:0x4083cad0,len:0xe04
16:14:57.896 -> load:0x4083efd0,len:0x2ec0
16:14:57.896 -> entry 0x4083cad0
16:14:58.151 -> [0;31mE (305) ZB_ESP_NVRAM: Failed to find zb_storage partition[0m
16:14:58.151 -> Zigbee stack assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_port/src/zb_esp_nvram.c:84
16:14:58.536 -> 
16:14:58.536 -> abort() was called at PC 0x42010fed on core 0
16:14:58.536 -> Core  0 register dump:
16:14:58.536 -> MEPC    : 0x408015da  RA      : 0x4080608e  SP      : 0x4081cb70  GP      : 0x4080db04  
16:14:58.536 -> TP      : 0x4081cd30  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
16:14:58.536 -> S0/FP   : 0x4081cb9c  S1      : 0x4081cb9c  A0      : 0x4081cba8  A1      : 0x4081cb8a  
16:14:58.536 -> A2      : 0x00000000  A3      : 0x4081cbd5  A4      : 0x00000001  A5      : 0x40814000  
16:14:58.536 -> A6      : 0x00000000  A7      : 0x76757473  S2      : 0x4080f780  S3      : 0x42075000  
16:14:58.536 -> S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
16:14:58.536 -> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
16:14:58.536 -> T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
16:14:58.536 -> MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
16:14:58.536 -> MHARTID : 0x00000000  
16:14:58.536 -> 
16:14:58.536 -> Stack memory:
16:14:58.536 -> 4081cb70: 0x00000000 0x00000000 0x4081cb88 0x4080c184 0x00000000 0x00000000 0x00000030 0x4080ef08
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

M4rkoV2, 2025-07-20:
Oh you are right I missed the partition scheme settings. 

But I still have a problem, guess registering multiple endpoints in Arduino IDE is not supported? 
Smartthings only recognize first endpoint.

P-R-O-C-H-Y, 2025-07-20:
There is no issue registering multiple endpoints in one device. It works fine using HomeAssistant ZHA or Z2M :). Maybe a SmartThings limitation?? Test with 2 light endpoints, maybe its also related to the contact switch. If yes, I will suggest to use 2 binary inputs instead.

closed 2025-07-22 by P-R-O-C-H-Y

Levanterman, 2026-01-05:
> There is no issue registering multiple endpoints in one device. There are many examples in the Zigbee library that uses multiple endpoints. It works fine when connecting to HomeAssistant ZHA or Z2M :). Maybe a SmartThings limitation?? Test with 2 light endpoints, maybe its also related to the contact switch. If yes, I will suggest to use 2 binary inputs instead.

@P-R-O-C-H-Y  I was also trying to add two ContactSwitches and could not get it working with Z2M , it only registered the last contact sensor and ignored the first one. I tried what you suggested and did 2 digitalInputs instead on one device and that worked first time and perfectly. So maybe its not a SmartThings issue, just something to do with the ContactSensor. I am using the Seeed ESP32-C6 Zigbee. Hope this helps someone else. thanks for your suggestion, it works.