espressif/arduino-esp32 #11615: EAP Authentication not works on v3.2.1(idf-v5.4.2)

truemeg, 2025-07-20:
### Board

ESP32-Dev Module

### Device Description

No board, only esp32 device

Chip Info:
```
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 1.01
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
```

### Hardware Configuration

Only ESP32

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE, VSCode

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I have tested on v2.0.17 and it worked 
But after upgrading as v3.2.1, i confirmed that it didnt work even one time.
Also i found that the problem reconnecting issue which if i reconnect esp32 within 5 minutes, it gets ip but ping not works, but my wifi laptop works fine, it means my FreeRadius Server and AP works fine.

### Sketch

```cpp
I used sample example project. WiFiClientEnterprise
```

### Debug Message

```plain
---- Opened the serial port COM3 ----
[... 172 lines cut ...]
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : esp32:esp32:esp32:UploadSpeed=115200,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=16M,PartitionScheme=app3M_fat9M_16MB,DebugLevel=verbose,PSRAM=enabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
============ Before Setup End ============
[   726][V][esp32-hal-uart.c:676] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   734][V][esp32-hal-uart.c:774] uartBegin(): UART0 not installed. Starting installation
[   744][V][esp32-hal-uart.c:784] uartBegin(): UART0 RX FIFO full threshold set to 120 (value requested: 120 || FIFO Max = 128)
[   758][V][esp32-hal-uart.c:814] uartBegin(): Setting UART0 to use REF_TICK clock
[   768][V][esp32-hal-uart.c:874] uartBegin(): UART0 initialization done.
[   783][V][esp32-hal-uart.c:925] uartSetRxFIFOFull(): UART0 RX FIFO Full value set to 120 from a requested value of 120

Connecting to network: terrance1226
[   803][W][STA.cpp:543] disconnect(): STA already disconnected.
[   824][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 101 - WIFI_READY
[   903][V][STA.cpp:186] _onStaEvent(): STA Started
[   908][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 110 - STA_START
[   915][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 110 - STA_START
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

truemeg, 2025-07-20:
Ah thanks, But how can I confirm if my device supports only old chipers?
and the reason why i upgraded as v3.x is that EAP-TLS feature was not working. even I input right cert files.

Jason2866, 2025-07-21:
Look in the datasheet of the devices.

Jason2866, 2025-07-21:
Which TLS version?

truemeg, 2025-07-22:
max v1.2