espressif/arduino-esp32 #11638: Custom partition is no longer working

enjoyneering, 2025-07-25:
### Board

ESP32 Dev Module

### Device Description

plain module

### Hardware Configuration

plain module

### Version

v3.3.0

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Custom partition is no longer working:

Sketch uses 1329607 bytes (7%) of program storage space. Maximum is 16777216 bytes.
Global variables use 55240 bytes (16%) of dynamic memory, leaving 272440 bytes for local variables. Maximum is 327680 bytes.

### Sketch

```cpp
Contents of the file "partitions.csv" located in the project sketch folder:

# Name,   Type, SubType, Offset,  Size, Flags
nvs,data,nvs,0x9000,0x5000,
otadata,data,ota,0xE000,0x2000,
app0,app,ota_0,0x10000,0x150000,
app1,app,ota_1,0x160000,0x150000,
spiffs,data,spiffs,0x2B0000,0x140000,
coredump,data,coredump,0x3F0000,0x10000,
```

### Debug Message

```plain
empty
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-07-25:
The percentage of "program storage space" there is only calculated from the partition scheme chosen.  Arduino doesn't actually know about the partitions.csv file, so the custom scheme is set to a very large number simply so you don't get an error. It has always been like this.

fcaussa, 2025-08-16:
> The percentage of "program storage space" there is only calculated from the partition scheme chosen. Arduino doesn't actually know about the partitions.csv file, so the custom scheme is set to a very large number simply so you don't get an error. It has always been like this.

I am currently on this project where I need a custom partition table. I was really worried about what I was doing wrong! Specially after using a "custom" scheme, but rather just ctrl+c/ctrl+v the default csv file into the sketch folder, and still getting this "16777216 bytes" of programming space. 

Thank you for your answer!