espressif/arduino-esp32 #11649: Invalid head of packet during uploading code

cniedzi, 2025-07-26:
### Board

ESP32 Dev Module

### Device Description

Plain ESP32 Dev Module with CP2102 USB to UART chip

### Hardware Configuration

Plain module without any connection. Only USB cable connected to PC for uploading code.

### Version

v3.3.0

### IDE Name

Arduino IDE, Platformio

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

any

### Description

During uploading any code based onArduino core 3.3.0 to any ESP32 Dev module with CP2102, I'm getting the following error:

```
Building in release mode
Retrieving maximum program size .pio\build\esp32\firmware.elf
Checking size .pio\build\esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  34.7% (used 113600 bytes from 327680 bytes)
Flash: [==========]  96.8% (used 1395991 bytes from 1441792 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa   
CURRENT: upload_protocol = esptool
Looking for upload port...
Uploading .pio\build\esp32\firmware.bin
esptool v5.0.1
Serial port COM3:
Connecting.......

A fatal error occurred: Invalid head of packet (0x00): Possible serial noise or corruption.
*** [upload] Error 2
```

Upload works seamlessly when used with ESP32 Dev Module based on CH340, or when I revert the core to 3.2.1.

### Sketch

```cpp
n/a
```

### Debug Message

```plain
Building in release mode
Retrieving maximum program size .pio\build\esp32\firmware.elf
Checking size .pio\build\esp32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  34.7% (used 113600 bytes from 327680 bytes)
Flash: [==========]  96.8% (used 1395991 bytes from 1441792 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa   
CURRENT: upload_protocol = esptool
Looking for upload port...
Uploading .pio\build\esp32\firmware.bin
esptool v5.0.1
Serial port COM3:
Connecting.......

A fatal error occurred: Invalid head of packet (0x00): Possible serial noise or corruption.
*** [upload] Error 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-07-28:
You could try to submit an issue in the esptool repository and let them know that `5.0.0-dev1` works, but `5.0.0` doesn't. It could be something specific there, but so far you are the first to report such issue.

cniedzi, 2025-07-28:
reported here: https://github.com/espressif/esptool/issues/1106

closed 2025-07-28 by cniedzi

gluca, 2025-07-30:
+1 Yesterday Arduino ide upgraded esp32 to 3.3.0 and I faced same issue. Reverted to 3.2.1

letrthong, 2025-08-24:
> > +1 Yesterday Arduino ide upgraded esp32 to 3.3.0 and I faced same issue. Reverted to 3.2.1
> 
> the same issue with me

Go tool  -> eps32 -> select v 3.2.1

cniedzi, 2025-08-24:
It has been fixed in Pioarduino, but probably not in Arduino IDE...