espressif/arduino-esp32 #11666: RMT & UART can not share RX pin anymore

devarishi7, 2025-07-29:
### Related area

RMT/UART

### Hardware specification

ESP32 devkit

### Is your feature request related to a problem?

When compiling with the newest released core v3.3.0, RMT seems to be finally working correctly, but the sharing of the RX pin between RMT(RX) & UART(RX) is no longer supported as it was in core v2.0.x . When looking for a break in a DMX data stream using RMT, this saves the use of an extra pin. Starting the UART disables RMT as it is now.

### Describe the solution you'd like

Allow sharing of INPUT pins between peripherals.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

devarishi7, 2025-07-29:
Well this is rather disappointing. Hardware wise they obviously can. Hardware wise there is no conflict in sharing an Input pin on both peripherals. If there would be a better way to measure a break-pulse  using the UART, i would use it, but the peripheral that measures pulse lengths (or the lack of pulses) is the RMT. Unfortunately the RMT isn't really ideal for receiving Serial data, the UART is. I can share my disappointment and my point of view and will be left with using 2 pins, connected to each other and to external hardware, which is less elegant as a solution. I don't see it as a fluke, nor really as a feature, but something that was possible, and that somehow now is not allowed anymore, or actually causes the ESP32 to crash without a clear error report. As such i guess i would consider it a bug, (the error is not handled properly) Although there doesn't have to be an error, at least not when using IDF 4.x , can't say for IDF 5.x. If the IDF allows it, then i think the core should as well. (just an opinion)

me-no-dev, 2025-07-31:
@SuGlider can't he use one of the UART events, like TOUT or something to achieve the same functionality? We can't change the peripheral manager, but as you said, he can use IDF APIs to directly attach the RMT pin also (if IDF will allow that)

devarishi7, 2025-07-31:
> The solution may be change the Peripheral Manager to allow multiple Peripheral attachments or to use pure IDF code to attach a pin to multiple Peripherals.

Well since the former seems to be out of the question, maybe the simplest would be to have a go at the latter. The obvious question that i have now is 'Where is the Peripheral Manager ? " In which core files is it located, so i can have a look and see if i could just re-attach the pin to the RMT after i have initiated the UART, but before i start the RMT readTask. It will save me a lot of time.

I was avoiding pure IDF commands, since, well to tell the truth, it is all a bit beyond me usually and i don't want to do anything where i feel i don't know what i'm doing. But if i can see how the pin is originally attached, and then detached , all i would have to do is 'undo' that detachment in a similar way to how it was attached originally.

me-no-dev, 2025-07-31:
crashing is unexpected yes. If you enable core debug, it will tell you that it detached the pin from one peripheral to another

me-no-dev, 2025-07-31:
Periman lives here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-periman.c

Also it prints out what pin is attached to what at the end of `setup()` if core debug level is `Debug` or `Verbose`

devarishi7, 2025-07-31:
@SuGlider Yes that was my train of thought exactly. I was thinking to do that to the RMT callback, The RMT gets initialized before the UART, but then i considered just to pretend that there is nothing attached to begin with
```
static peripheral_pin_item_t pins[SOC_GPIO_PIN_COUNT];
```
By setting that pin to 
`ESP32_BUS_TYPE_INIT`

anyway, i have to do some more digging and testing, but first dinner.

devarishi7, 2025-08-01:
Well no luck so far. I have tried tricking the peripheral manager into not detaching the RMT from the pin, but since the pins are constantly attached to only 1 bus at a time and are checking on this with the peripheral manager, it would mean quite a lot of modification and complication while the objective was top keep things simple. Seems like the pins can really only have 1 function at a time (similar to how things are on an ESP8266) and that it is possible in core 2 is an anomaly, although it does prove that hardware wise it can be done without issue which suggests that it could be done regardless. Still it is to much of a dive into the code for me and i will simply suggest to connect 2 pins (through a current limiting resistor) together to make use of core 3.3.0 Pity but what to do.

SuGlider, 2025-08-04:
@devarishi7 - Pleae try this modified code using the same RMT and UART RX pin + Arduino Core 3.x
The code ahnges are marker with `// =======> CHANGE:`

``` cpp
// =======> CHANGE: include Peripheral Manager header file
#include "esp32-hal-periman.h"

#define LED_GREEN 13
#define LED_OFF HIGH

#define LED_WHITE 14
#define INVERT_LED 255 -

// =======> CHANGE: use the same pin for RMT RX and UART RX
#define RX_PIN 34      // MAX485 DMX reception
#define RX_RMT RX_PIN  //35      // try with 2 separate pins, works on the same pin in core 2.0.11
//#define RX_RMT 35      // try with 2 separate pins, works on the same pin in core 2.0.11
#define TX_PIN 255     // MAX485 DMX transmission (dummy)
#define RX_DISABLE 33
#define DMX_PORT &Serial1

//---------------------------------------

// =======> CHANGE: Override RMT and Serial dettach routines in order to keep driver working
//and do not detach RMT/UART RX pin whenever this pin is used for within a different perpheral

static bool _rmtRXPinDetachBus(void *busptr) {
  // does nothing and keeps the RMT driver running attached to RX pin
  log_d("_rmtRXPinDetachBus called");
  return true;
}

static bool _uartRXPinDetachBus(void *busptr) {
  // does nothing and keeps the UART driver running attached to RX pin
  log_d("_uartRXPinDetachBus called");
  return true;
}

//---------------------------------------
#define DMX_CHANNELS 513
#define SERIAL_SIZE_RX 530  // the break is also stored as a '0' since it 
// starts with a start bit, that the stop bits aren't present
// does not seem to matter
// most efficiently we read that bit straight away
// we do want to make sure that the FIFO is actually empty already so we waste another 16 bytes

#if ESP_IDF_VERSION_MAJOR >= 5
#define RMT_FREQ_HZ 1000000
#else
/*#include "Arduino.h"
  #include "esp32-hal.h"  // esp32-hal-rmt.h & .c  contain rmtRead() & rmtInit()
  #include "driver/rmt.h"  //  "hal/rmt_types.h" is included in this and contains 
  //  Macros and types
  // in rmt.h most other functions are defined
  #include "driver/periph_ctrl.h"
  #include "soc/rmt_reg.h"*/

// the callBack from the RMT idle implementation once the object has been
// declared
extern "C" IRAM_ATTR void idleCalback(uint32_t *data, size_t len, void * arg);

#endif

// --------------------

class dmxRx {
  private:
    static const uint16_t _maxChannels = DMX_CHANNELS + 1;  // a normal frame ends up being
    // 514 bytes, quickest is to read all of those in one go
#if ESP_IDF_VERSION_MAJOR >= 5
[... 294 lines cut ...]
  if (!dmxReceive.configure()) {  // configures and starts the UART (with default parameters)
    Serial.println("DMX Configure failed.");
  }
  else {
    Serial.println("DMX Configured.");
  }
  delay(10);
  if (dmxReceive.start()) {
    Serial.println("DMX reception Started");
  }
  else {
    Serial.println("DMX aborted");
  }

  pinMode(LED_WHITE, OUTPUT);  // the indicator LED
}

void loop() {

  if (dmxReceive.hasUpdated()) {  // only read new values
    analogWrite(LED_WHITE, INVERT_LED dmxReceive.dmxRead(7));  // the LED i have is active LOW
  }
}

```

devarishi7, 2025-08-05:
Ah as an afterthought, would it be a good idea to return the bus DeInit function callbacks to their originals ? I can think of scenarios where leaving them as empty may be undesirable.

SuGlider, 2025-08-05:
> Ah as an afterthought, would it be a good idea to return the bus DeInit function callbacks to their originals ? I can think of scenarios where leaving them as empty may be undesirable.

it could be done.... but the DeInit funtions from `esp32-hal-uart.c` and `esp32-hal-rmt.c` are declared as static, therefore those are not public, neither visible to the Arduino Sketch code...

devarishi7, 2025-08-05:
Yes, not a simple thing to do. Also the array containing them is declared static. (and in the .c file not in the .h)  So even copying their code is problematic. Still i guess i could just let the user decide. The scenario in which a pin would first serve as an RMT Rx pin or UART Rx pin and then serve as something else would be rare, and the option should be to use 2 separate pins i guess.

devarishi7, 2025-08-12:
@SuGlider I got it all working and it does work pretty well, but then when i implemented it i got some 'artifacts' basically during different processes it seems that the RMT callback gets called before the UART FIFO full ISR is triggered on occasion which results in the channel skipping. Then while i was looking for the priority i saw that actually the uart has an onReceive() where you can set to call the function on the break specifically, and i thought that i would give that a try as well first. (other option would be to use the other core, which i was and probably still am going to implement) 
But i am having some issues with determining which of my instances is calling the callback

This compiles,  but i wanted to keep the idle callback within the class, or at least retrieve the pointer to the object the same way as the RMT callback does, or be able to access the class it's private member variables in that way to be more exact. But i feel i am a snake eating it's own tail.

```
#define LED_GREEN 13
#define LED_OFF HIGH

#define LED_WHITE 14
#define INVERT_LED 255 -

#define RX_PIN 35      // MAX485 DMX reception
#define TX_PIN 255      // MAX485 DMX transmission (dummy)
#define RX_DISABLE 33
#define DMX_PORT &Serial1

// --------------------

#define DMX_CHANNELS 513
#define SERIAL_SIZE_RX 530

//IRAM_ATTR static void idleCallback();

class dmxRx {
  private:
    static const uint16_t _maxChannels = DMX_CHANNELS + 1;  // a normal frame ends up being
    // 514 bytes, quickest is to read all of those in one go
    HardwareSerial* _dmxPort = NULL;
    uint8_t _pinEnable, _pinRx, _pinTx, _pinToggle;
    uint8_t _dmxBuf[_maxChannels];
    bool _configured = false, _firstFrame = true;
    volatile bool _foundBreak = false;
    //volatile uint16_t _readAlready = 0;

    IRAM_ATTR inline void toggleLed() {  // called from an IRAM_ATTR function
      static bool ledOn = false;
      ledOn = !ledOn;
      if (ledOn) digitalWrite(_pinToggle, HIGH);
      else digitalWrite(_pinToggle, LOW);
    };

  public:

    dmxRx(HardwareSerial* port, uint8_t pinRx,
          uint8_t pinEnable = 255, uint8_t pinToggle = 255) {

      _dmxPort = port;
      _pinRx = pinRx;
      _pinTx = 255;
      _pinEnable = pinEnable;
      _pinToggle = pinToggle;

    };

    IRAM_ATTR static void idleCallback();

    bool configure() {  // call this in setup()

      _dmxPort->setRxBufferSize(SERIAL_SIZE_RX);
      _dmxPort->onReceive(idleCallback, true);
      _dmxPort->setRxTimeout(2);
      _dmxPort->begin(250000, SERIAL_8N2, _pinRx, _pinTx);
      //_dmxPort->setRxFIFOFull(1);  // one byte at a time
      _dmxPort->setRxFIFOFull(40);

      pinMode(_pinEnable, OUTPUT);
      digitalWrite(_pinEnable, HIGH); // not receive mode. The library expects the pin to be
      // wired to the !RE pin only, basically leaving it unenabled unless used
      // if the !RE & DE pins are bridged, you should switch the pins manually
[... 109 lines cut ...]
}

void setup() {
  Serial.begin(500000);
  Serial.println("\n");
  Serial.println("DMX using UART timeout looking for break.");

  if (!dmxReceive.configure()) {  // configures and starts the UART (with default parameters)
    Serial.println("DMX Configure failed.");
  }
  if (dmxReceive.start()) {
    Serial.println("DMX reception Started");
  }
  else {
    Serial.println("DMX aborted");
  }
  pinMode(LED_WHITE, OUTPUT);  // the indicator LED
}

void loop() {
  if (dmxReceive.hasUpdated()) {  // only read new values
    analogWrite(LED_WHITE, INVERT_LED dmxReceive.dmxRead(7));  // the LED i have is active LOW
  }
}
```

SuGlider, 2025-08-14:
RMT uses the IDF Driver and it is linked to the RMT ISR.
UART also uses IDF Driver and it is also linked to the UART ISR.

The `onReceive()` callback is called latter after FIFO Full ISR has copied FIFO data to a FreeRTOS Queue.
I think that changing the `onReceive()` priority shall not solve the artifact issue.

Break detection is actually a trick part of the UART IDF driver that Arduino is based on.
There is an example about it, that uses `onReceiveError()`, similar to `onReceive()`usage:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/OnReceiveError_BREAK_Demo/OnReceiveError_BREAK_Demo.ino

devarishi7, 2025-08-15:
Well yeah, just thinking about it, the strange thing about the artifacts is this. Basically i am receiving DMX channels, converting them to WS2812b ledstrip data (or similar) and what i see on the strip is the pixels that should be projected, and quite a few pixels later in the chain also showing up, but not constantly. But somehow in the same color ? that bit does not really make sense to me. That would mean that the sync is lost, or moved to be more correct, but exactly a multiplied by 3 number every time ? Now that is weird.
This happens only when i am receiving in the background. If i wait for a break, and then another break and use what i find in between it's all perfect. But the idea was to receive the break in the background. So i thought that maybe there are still bytes in the FIFO the moment that the RMT callback is called, but the exact skipping of 3 bytes, makes me think something else is going on. 
Anyway, i was then considering to use the 'onReceive' callback, since if it is all happening from the same source that issue could not really exist, but i can't work out how i pass a pointer to the object to that callback, which means i can not use any of the variables, the same way i can with the RMT callback. 

Are these callbacks called straight from the ISR's at all is something i am wondering, or are they put in a Queue somehow ? 

Anyway as i am writing this, and am testing different scenarios, i have come to the conclusion that the artifacts are not really caused by the dmx reception, but rather by my main sketch using it. So guess this part of the discussion is obsolete, though i am still curious about some things, but i now have a much clearer idea of where i should be looking and how to fix what was going :-D It actually should be fairly easy to solve. The skipping of exactly 3 bytes was just so unlikely that i should have been looking in a different direction before. 

So i guess i won't be pursuing the UART only avenue after all (pffff... ) although i am still curious about how to let the onReceive callback of the UART access variables from the object from which it is registered, the same way (or similar) one can do with the RMT idle callback, but that is now a completely different issue i guess.

Thank you so much ! you will of course be credited for the work on using the same pin in core 3 when i complete the testing of the library and find the time to publish it on github (it will be my first) Thank you again.

SuGlider, 2025-08-15:
`OnReceive()` works in a very specific wqy because it uses UART Events that the IDF Driver sends.
By default, it executes the callback after about 120 bytes are received. If the package has more bytes, those will be in the next callback execution.

There is a way to "tune" it, as explained in the https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h:

``` cpp
  // setRxTimeout sets the timeout after which onReceive callback will be called (after receiving data, it waits for this time of UART rx inactivity to call the callback fnc)
  // param symbols_timeout defines a timeout threshold in uart symbol periods. Setting 0 symbol timeout disables the callback call by timeout.
  //                       Maximum timeout setting is calculacted automatically by IDF. If set above the maximum, it is ignored and an error is printed on Serial0 (check console).
  //                       Examples: Maximum for 11 bits symbol is 92 (SERIAL_8N2, SERIAL_8E1, SERIAL_8O1, etc), Maximum for 10 bits symbol is 101 (SERIAL_8N1).
  //                       For example symbols_timeout=1 defines a timeout equal to transmission time of one symbol (~11 bit) on current baudrate.
  //                       For a baudrate of 9600, SERIAL_8N1 (10 bit symbol) and symbols_timeout = 3, the timeout would be 3 / (9600 / 10) = 3.125 ms
  bool setRxTimeout(uint8_t symbols_timeout);

  // setRxFIFOFull(uint8_t fifoBytes) will set the number of bytes that will trigger UART_INTR_RXFIFO_FULL interrupt and fill up RxRingBuffer
  // This affects some functions such as Serial::available() and Serial.read() because, in a UART flow of receiving data, Serial internal
  // RxRingBuffer will be filled only after these number of bytes arrive or a RX Timeout happens.
  // This parameter can be set to 1 in order to receive byte by byte, but it will also consume more CPU time as the ISR will be activates often.
  bool setRxFIFOFull(uint8_t fifoBytes);

  // onReceive will setup a callback that will be called whenever an UART interruption occurs (UART_INTR_RXFIFO_FULL or UART_INTR_RXFIFO_TOUT)
  // UART_INTR_RXFIFO_FULL interrupt triggers at UART_FULL_THRESH_DEFAULT bytes received (defined as 120 bytes by default in IDF)
  // UART_INTR_RXFIFO_TOUT interrupt triggers at UART_TOUT_THRESH_DEFAULT symbols passed without any reception (defined as 10 symbols by default in IDF)
  // onlyOnTimeout parameter will define how onReceive will behave:
  // Default: true -- The callback will only be called when RX Timeout happens.
  //                  Whole stream of bytes will be ready for being read on the callback function at once.
  //                  This option may lead to Rx Overflow depending on the Rx Buffer Size and number of bytes received in the streaming
  //         false -- The callback will be called when FIFO reaches 120 bytes and also on RX Timeout.
  //                  The stream of incommig bytes will be "split" into blocks of 120 bytes on each callback.
  //                  This option avoid any sort of Rx Overflow, but leaves the UART packet reassembling work to the Application.
  void onReceive(OnReceiveCb function, bool onlyOnTimeout = false);
```

You may try to set it changing RX timeout or a smaller FIFO full.
There is a known problem with `onReceive()` that is when the FIFO Full value is exactly the number of bytes received.
In that case, no event is generated by the IDF UART Driver and the callback isn't executed.

You may try to exepriment with the `onReceive()` exampleto check if this is affecting your application:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/OnReceive_Demo/OnReceive_Demo.ino
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/RxFIFOFull_Demo/RxFIFOFull_Demo.ino
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/RxTimeout_Demo/RxTimeout_Demo.ino

There are way to read the RX FIFO directly, configure the UART manually using the IDF functions etc.
For that it may be necessary to go with "LL/HAL - Low Level / Hardware Abstraction Layer" IDF Calls.
This new UART way of working may be mixed with Arduino Code, if `SerialN` is not started with `being(baudrate)`.

Given that your application is very specific, it may be an alternative way to make it work well.

#### This is the IDF code used in Arduino:
- IDF Driver in C:
https://github.com/espressif/esp-idf/tree/master/components/esp_driver_uart
- UART HAL in C (most calling LL functions):
https://github.com/espressif/esp-idf/blob/master/components/hal/uart_hal.c
https://github.com/espressif/esp-idf/blob/master/components/hal/uart_hal_iram.c
- UART LL functions for the ESP32 (change the hal/SoC folder for other):
https://github.com/espressif/esp-idf/blob/master/components/hal/esp32/include/hal/uart_ll.h  
- Arduino UART HAL level (uses IDF / HAL / LL):
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c

------------------------------------------------------

### For instance, it is possible to directly get the number of bytes left in the RX FIFO:

From Arduino it would be something like ` uint32_t available = uart_ll_get_rxfifo_len(UART_LL_GET_HW(uart_num));`
`UART_LL_GET_HW(uart_number)` demands including some header files into the Arduino Sketch: 
`#include "hal/uart_ll.h"`  `#include "soc/uart_struct.h"` `#include "soc/uart_periph.h"`
[... cut ...]
 * @param  rd_len The data length needs to be read.
 *
 * @return None.
 */
FORCE_INLINE_ATTR void uart_ll_read_rxfifo(uart_dev_t *hw, uint8_t *buf, uint32_t rd_len)
{
    //Get the UART APB fifo addr. Read fifo, we use APB address
    uint32_t fifo_addr = (hw == &UART0) ? UART_FIFO_REG(0) : (hw == &UART1) ? UART_FIFO_REG(1) : UART_FIFO_REG(2);
    for(uint32_t i = 0; i < rd_len; i++) {
        buf[i] = READ_PERI_REG(fifo_addr);
#ifdef CONFIG_COMPILER_OPTIMIZATION_PERF
        __asm__ __volatile__("nop");
#endif
    }
}
```

devarishi7, 2025-08-15:
> void onReceive(OnReceiveCb function, bool onlyOnTimeout = false);

Well when i read this, i thought the solution might be to set the 'onlyOnTimeout' to true, and use 2 ticks for the timeout, which would result in looking for an 88us break when using 8N2. Although that is longer than the spec, it would probably work for most cases, but there would be the option of just receiving at 8N1, which will receive 8N2 just fine, and that would result in the perfect 80us break-detection. 

> By default, it executes the callback after about 120 bytes are received. If the package has more bytes, those will be in the next callback execution.

I set the rx-FIFO full to '1' to make sure that the FIFO is transferred to the rx-buffer asap. I know it goes at the expense of processing speed, but i need to know exactly how many bytes have been received at the moment the break is detected.

> it is possible to directly get the number of bytes left in the RX FIFO

Well i was looking at that, in fact, just a way to transfer whatever is in there to the rx-buffer would already improve cpu load, since i could just call that and continue processing from there.

For now i think the question i am still curious about is how one could retrieve or pass the pointer to the object from the member function that is attaching the onReceive callback, from within that callback function. 

But as it is for now i have got the RMT thing working properly (i think) and am just debugging some 'extra' functions at the moment. 

I wanted to keep the code fairly simple and understandable for most people. The differences between the core 2 & 3 in the RMT implementation have gone at the expense of that unfortunately, but with your contribution in being able to share the RX pin for RMT & UART have at least made the core 3 interface consistent with the core 2 interface. That it disables the pin detachment for the RMT & UART pins from then on seems like a small price to pay, but i will let the user decide on that. If your request to add `perimanGetBusDeinit()` gets included in the next patch or version, i will update my code to gracefully re-attach the original callbacks again.

SuGlider, 2025-08-15:
>For now i think the question i am still curious about is how one could retrieve or pass the pointer to the object from the member function that is attaching the onReceive callback, from within that callback function.

onReceive() implements std::functional
Therefore, there many ways using C++ stdlib and lambda functions to make it work.

SuGlider, 2025-08-15:
@devarishi7 - these are the necessary changes to your code:

1. **Callback Function Signature that receives a dmxRx Class pointer**
   ```cpp
   // Old
   IRAM_ATTR static void idleCallback();

   // New
   IRAM_ATTR static void idleCallback(dmxRx* instance);
   ```

2. **Callback Registration using lambda function**
   ```cpp
   // Old
   _dmxPort->onReceive(idleCallback, true);

   // New
   _dmxPort->onReceive([this]() { idleCallback(this); }, true);
   ```

3. **Callback Implementation that uses the "self pointer"**
   ```cpp
   // Old
   IRAM_ATTR void dmxRx::idleCallback() {
     dmxReceive.breakDetected();
   }

   // New
   IRAM_ATTR void dmxRx::idleCallback(dmxRx* instance) {
     if (instance) {
       instance->breakDetected();
     }
   }
   ```

## Benefits
- Eliminates dependency on global instance
- Supports multiple DMX receiver instances
- Maintains type safety using proper C++ instance handling
- Preserves IRAM_ATTR optimization for interrupt handling
- Compatible with the HardwareSerial::onReceive callback signature requirements

devarishi7, 2025-08-15:
Oh wow, oh this is a tad beyond me !

> 2\. Callback Registration using lambda function

I have never done anything like that ! I will now test and see if this is also a reliable method of break detection. It could be. First i will finish the RMT library. There is so many things to do, but that was the question. Thank you.

devarishi7, 2025-08-19:
Ok so i have posted the library on Github [here](https://github.com/devarishi7/Dmx_ESP32) Thank you for the support.