espressif/arduino-esp32 #11672: Stack canary watchpoint triggered (cam_task) with 3.3.0

G-Remo, 2025-07-30:
### Board

ESP32 Cam

### Device Description

ESP32-CAM-MB

### Hardware Configuration

camera OV2640

### Version

v3.3.0

### Type

Task

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When switching to UXGA resolution system crash (more than 10 tests and 100% crash)

### Sketch

```cpp
exemple code "CameraWebServer" and #define CAMERA_MODEL_AI_THINKER
```

### Debug Message

```plain
log:
"WiFi connecting....
WiFi connected
Camera Ready! Use 'http://192.168.1.27' to connect
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (cam_task) 
Core  0 register dump:
PC      : 0x4008e31a  PS      : 0x00060336  A0      : 0x80092658  A1      : 0x3ffb2550  
A2      : 0x3ffb0300  A3      : 0xffffffff  A4      : 0xb33fffff  A5      : 0x3f41d378  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00060323  A9      : 0x3ffb2550  
A10     : 0x00000003  A11     : 0x00060323  A12     : 0x00060323  A13     : 0x00000020  
A14     : 0x3ffb2ab0  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008aaf1  LEND    : 0x4008ab01  LCOUNT  : 0xfffffffe  

Backtrace: 0x4008e317:0x3ffb2550 0x40092655:0x3ffb2580 0x4009268a:0x3ffb25a0 0x40092800:0x3ffb25c0 0x400830df:0x3ffb25e0 0x400830f9:0x3ffb2610 0x40082d75:0x3ffb2630 0x4008e804:0x3ffb2650 0x4008d895:0x3ffb2670 0x4008db0c:0x3ffb2690 0x4008375e:0x3ffb26b0 0x400837a6:0x3ffb26d0 0x40083909:0x3ffb2700 0x400e2889:0x3ffb2720 0x400e6e55:0x3ffb2750 0x4000bd83:0x3ffb2770 0x4000182a:0x3ffb2790 0x400e64fd:0x3ffb27b0 0x400e6e55:0x3ffb27d0 0x4008a7aa:0x3ffb27f0 0x40089c4d:0x3ffb2810 0x40089c9e:0x3ffb2830 0x4008a19d:0x3ffb2850 0x401758a3:0x3ffb2880 0x401752ba:0x3ffb28a0 0x4016bfe1:0x3ffb2bc0 0x400938d5:0x3ffb2bf0 0x40119b8d:0x3ffb2c50 0x4008e1f5:0x3ffb2ca0

ELF file SHA256: 97ddfcc70

Rebooting...
"

exeption decoder:
"
PC: 0x4008e31a
EXCVADDR: 0x00000000

Decoding stack results
0x4008e317: xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h line 552
0x40092655: multi_heap_internal_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h line 581
0x4009268a: multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c line 215
0x40092800: multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 254
0x400830df: heap_caps_aligned_alloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps_base.c line 93
0x400830f9: heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps_base.c line 202
0x40082d75: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 84
0x4008e804: pvPortMalloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/heap_idf.c line 55
0x4008d895: xQueueGenericCreate at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c line 545
0x4008db0c: xQueueCreateMutex at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c line 669
0x4008375e: lock_init_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/src/locks.c line 75
0x400837a6: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/src/locks.c line 133
0x40083909: _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/src/locks.c line 174
0x400e2889: uart_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_uart/src/uart_vfs.c line 236
0x400e6e55: esp_vfs_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c line 982
0x400e64fd: console_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_vfs_console/vfs_console.c line 83
0x400e6e55: esp_vfs_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c line 982
0x4008a7aa: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/stdio.c line 94
0x40089c4d: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c line 224
0x40089c9e: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c line 278
0x4008a19d: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c line 251
0x401758a3: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c line 429
0x401752ba: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c line 1774
0x4016bfe1: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c line 34
0x400938d5: esp_log at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/log/src/log.c line 43
0x40119b8d: cam_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/managed_components/espressif__esp32-camera/driver/cam_hal.c line 221
0x4008e1f5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139
"
```

### Other Steps to Reproduce

ESP32 Dev Module
DIO
4MB
Minimal SPIFFS

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-07-31:
@RubenKelevra PTAL

RubenKelevra, 2025-08-04:
@me-no-dev thanks for the notice, I must have overlooked it - sorry for the delay!

@labichea thanks for the confirmation!

lbernstone, 2025-08-04:
The camera library is compiled into arduino-esp32 archives.  You can't just change the sdkconfig.  The new version with that PR will make it into arduino after it is merged into the esp32-camera repo.

RubenKelevra, 2025-08-05:
Ah, this makes sense why the increase didn't help.

I've created a PR to double the stack size: https://github.com/espressif/esp32-camera/pull/788
And I've modified my fix to use a more lightweight print method:

https://github.com/espressif/esp32-camera/compare/d6fbaf0029c1fe6f5c9dfdb4764f8d9b8ecf9fc7..5bb9bfe5e4230c4045c8a75fc8c9876197851b0e

And I've created a PR to do the same for other error messages:

https://github.com/espressif/esp32-camera/pull/787

G-Remo, 2025-08-06:
After building esp32-camera and replace file "libespressif__esp32-camera.a" (not fun as expected) with:

```
config CAMERA_TASK_STACK_SIZE
    int "CAM task stack size"
    default 4096
    help
        Camera task stack size
```

```
if (cnt >= cam_obj->frame_copy_cnt) {
    ESP_LOGE(TAG, "CAM_TASK_STACK: %u", CAM_TASK_STACK);
    ESP_LOGE(TAG, "cnt: %u >= cam_obj->frame_copy_cnt %u", cnt, cam_obj->frame_copy_cnt);
    ESP_CAMERA_ETS_PRINTF(DRAM_STR("cam_hal: DMA overflowwww\r\n"));
    // ESP_LOGE(TAG, "DMA overflowwwww");
    ll_cam_stop(cam_obj);
    cam_obj->state = CAM_STATE_IDLE;
    continue;
}
```

I miss something
```
10:06:56.979 -> cam_hal: NO-EOI - JPEG end marker missing
10:06:57.442 -> cam_hal: NO-SOI - JPEG start marker missing
```

but no crash

```
10:07:38.127 -> framesize: 15
10:07:38.792 -> E (49261) cam_hal: CAM_TASK_STACK: 4096
10:07:38.792 -> E (49262) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:38.792 -> cam_hal: DMA overflowwww
10:07:38.872 -> E (49343) cam_hal: CAM_TASK_STACK: 4096
10:07:38.872 -> E (49343) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:38.872 -> cam_hal: DMA overflowwww
10:07:38.948 -> E (49422) cam_hal: CAM_TASK_STACK: 4096
10:07:38.948 -> E (49422) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:38.984 -> cam_hal: DMA overflowwww
10:07:39.023 -> E (49502) cam_hal: CAM_TASK_STACK: 4096
10:07:39.023 -> E (49502) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.063 -> cam_hal: DMA overflowwww
10:07:39.103 -> E (49581) cam_hal: CAM_TASK_STACK: 4096
10:07:39.103 -> E (49583) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.143 -> cam_hal: DMA overflowwww
10:07:39.183 -> E (49661) cam_hal: CAM_TASK_STACK: 4096
10:07:39.183 -> E (49662) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.223 -> cam_hal: DMA overflowwww
10:07:39.263 -> E (49741) cam_hal: CAM_TASK_STACK: 4096
10:07:39.263 -> E (49741) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.303 -> cam_hal: DMA overflowwww
10:07:39.343 -> E (49821) cam_hal: CAM_TASK_STACK: 4096
10:07:39.343 -> E (49822) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.383 -> cam_hal: DMA overflowwww
10:07:39.458 -> E (49901) cam_hal: CAM_TASK_STACK: 4096
10:07:39.458 -> E (49902) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.458 -> cam_hal: DMA overflowwww
10:07:39.538 -> E (49981) cam_hal: CAM_TASK_STACK: 4096
10:07:39.538 -> E (49981) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.538 -> cam_hal: DMA overflowwww
10:07:39.578 -> E (50061) cam_hal: CAM_TASK_STACK: 4096
10:07:39.618 -> E (50061) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.618 -> cam_hal: DMA overflowwww
10:07:39.698 -> E (50141) cam_hal: CAM_TASK_STACK: 4096
10:07:39.698 -> E (50142) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.698 -> cam_hal: DMA overflowwww
10:07:39.738 -> Erreur fb
10:07:39.778 -> E (50221) cam_hal: CAM_TASK_STACK: 4096
10:07:39.778 -> E (50222) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.778 -> cam_hal: DMA overflowwww
10:07:39.818 -> E (50301) cam_hal: CAM_TASK_STACK: 4096
10:07:39.858 -> E (50301) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.858 -> cam_hal: DMA overflowwww
10:07:39.898 -> E (50380) cam_hal: CAM_TASK_STACK: 4096
10:07:39.938 -> E (50381) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:39.938 -> cam_hal: DMA overflowwww
10:07:39.979 -> E (50461) cam_hal: CAM_TASK_STACK: 4096
10:07:40.014 -> E (50461) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:40.014 -> cam_hal: DMA overflowwww
10:07:40.094 -> E (50541) cam_hal: CAM_TASK_STACK: 4096
10:07:40.094 -> E (50541) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:40.094 -> cam_hal: DMA overflowwww
10:07:40.134 -> Erreur fb
10:07:40.169 -> E (50621) cam_hal: CAM_TASK_STACK: 4096
10:07:40.169 -> E (50622) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:40.169 -> cam_hal: DMA overflowwww
10:07:40.254 -> E (50701) cam_hal: CAM_TASK_STACK: 4096
10:07:40.254 -> E (50702) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:40.254 -> cam_hal: DMA overflowwww
10:07:40.334 -> E (50782) cam_hal: CAM_TASK_STACK: 4096
10:07:40.334 -> E (50783) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:40.334 -> cam_hal: DMA overflowwww
10:07:40.416 -> E (50862) cam_hal: CAM_TASK_STACK: 4096
10:07:40.416 -> E (50862) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:40.416 -> cam_hal: DMA overflowwww
10:07:40.495 -> E (50942) cam_hal: CAM_TASK_STACK: 4096
10:07:40.495 -> E (50943) cam_hal: cnt: 93 >= cam_obj->frame_copy_cnt 93
10:07:40.495 -> cam_hal: DMA overflowwww
10:07:40.535 -> Erreur fb
10:07:40.535 -> Camera deinit /init
10:07:41.061 -> E (51513) gpio: gpio_install_isr_service(526): GPIO isr service already installed
10:07:52.239 -> quality: 13
10:07:52.413 -> quality: 14
10:07:53.302 -> quality: 15
10:07:53.834 -> quality: 16
10:07:54.127 -> quality: 17
```

Now I can try to reduce quality (increase quality value) before re-init the camera.

RubenKelevra, 2025-08-06:
@G-Remo yesterday was a new version released which includes further reductions in stack usage size and defaults the stack size to 4096.

However on my m5 (ESP32-S3 with a MEGA_CCM) I previously still ran into size issues with the stack with 4096, so I used 8 k for my testing so far.

But the new lighter error messages in 2.1.1 *should* reduce the stack size enough, I think, to fit in 4096, I hope.

TL;DR: Try the 2.1.1 (which includes new fixes) and if you still run into issues, try it with 8 k instead of the default 4.

Hope that's not too much work for you. 😇

Hope this fixes the issue you've encountered :)

G-Remo, 2025-08-06:
@RubenKelevra
"cam_hal: DMA overflowwww"
I don't know much about compiling sources, I did a lot of tests, I added "w" to see if it had an effect on the output;

For rest I not sure to hunderstand, from the begining:
```
Can you share the commit id or tag you've used to compile so I can see which version of the EOI / SOI checks you're using?
I was working on them in the past.
```
with esp32-arduino v3.3.0 when start or switching to UXGA cause  crash (Stack canary watchpoint triggered (cam_task) )
It run with low quality (over 17) or low frame size, but in my code I start at UXGA and use s->set_res_raw to use a window

for the build:
Installer ESP-IDF
esp-idf-tools-setup-online-2.3.5.exe the last esp32 idf. I think

git clone https://github.com/espressif/esp32-camera.git
cd esp32-camera
mkdir camera_build && cd camera_build
cp -r ../esp32-camera components/

Crée un fichier CMakeLists.txt minimal :

```
cmake_minimum_required(VERSION 3.5)
# Inclure le système de build IDF
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
# Définir le composant externe (esp32-camera)
set(EXTRA_COMPONENT_DIRS components/esp32-camera)
# Définir le projet
project(esp32-camera-lib)
```

Crée main/empty.c avec un fichier vide :

`void app_main() {}`

main/CMakeLists.txt

`idf_component_register(SRCS "empty.c" INCLUDE_DIRS "")`

edit cam_hal.c & Konfig 

```
config CAMERA_TASK_STACK_SIZE
    int "CAM task stack size"
    default 4096
    help
        Camera task stack size
```

```
if (cnt >= cam_obj->frame_copy_cnt) {
    ESP_LOGE(TAG, "CAM_TASK_STACK: %u", CAM_TASK_STACK);
    ESP_LOGE(TAG, "cnt: %u >= cam_obj->frame_copy_cnt %u", cnt, cam_obj->frame_copy_cnt);
    ESP_CAMERA_ETS_PRINTF(DRAM_STR("cam_hal: DMA overflow\r\n"));  //don't add "w"
    // ESP_LOGE(TAG, "DMA overflowwwww");
    ll_cam_stop(cam_obj);
    cam_obj->state = CAM_STATE_IDLE;
    continue;
}
```

with ESP-IDF 5.5 PowerShell
```
cd "to the good directory"
idf.py set-target esp32
idf.py build
```
Et voila, it build
Then I rename and copy libespressif__esp32-camera.a and over write in "D:\Arduino\Arduino_ESP32_3\portable\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.5-b66b5448-v1\esp32\lib"

```
But the message basic tells you, that the storage (frame buffer slot) isn't large enough for the images.

But you should never see this message: The check is meant for the psram_mode. In addition it did not account correctly for the reduction in frame buffer slot size it's causing.
```
I don't understand

RubenKelevra, 2025-08-06:
> I don't know much about compiling sources, I did a lot of tests, I added "w" to see if it had an effect on the output;
> 
> For rest I not sure to hunderstand, from the begining:

Oh, sorry. I assumed that you're a developer, if you're willing to go through the pain of replacing compiled modules ;)

> I use Arduino IDE not Espressif IDE, i can't use esp32-camera, it s pre compiled in esp32-arduino
with esp32-arduino v3.3.0 when start or switching to UXGA cause crash

Well, technically you're always using the ESP-IDF Framework (what you're probably mean with "IDE"). The Arduino Framework is just put on top of ESP-IDF.

Anyway: To make things short - there's a new version out there, if you wait it should land shortly also in the esp32-arduino framework.

If you don't want to wait: My PR got merged today in the ESPHome project, so you can test the camera itself out via ESPHome's master. In ESPHome you can swhich can switch back and forth between adding the Arduino Framework layer on top of it or not, but it makes no difference here, as the esp32-camera library is always the most up to date version 2.1.1.

Hope this helps! :)

> I don't understand

In short: I made a mistake and the space for images is reduced. So the image send by the camera is too large for the space reserved for it, so it tells you with the "cam_hal: DMA overflow":

The end of the space was reached.

The main issue here is:

1) The space should be larger
2) This check was meant for a different mode of the library

And both issues were fixed in my next PR.

me-no-dev, 2025-09-10:
Closing this as resolved. Will release a new Arduino version in a few days

closed 2025-09-10 by me-no-dev