espressif/arduino-esp32 #11675: esp_now_register_send_cb(OnDataSent); compile problem

Manik601292, 2025-07-30:
### Board

EPS32 Dev kit module

### Device Description

ESP NOW Master I am facing 
esp_now_register_send_cb(OnDataSent); 
compile problem.

How can I solve it????

### Hardware Configuration

ESP32 V1

### Version

v3.3.0

### Type

Task

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

240MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

ESP NOW Master I am facing 
esp_now_register_send_cb(OnDataSent); 
compile problem.

How can I solve it????

### Sketch

```cpp
// ESP-NOW Demo - Transmit

// Include Libraries
#include <esp_now.h>
#include <WiFi.h>
 
// Variables for test data
int int_value;
float float_value;
bool bool_value = true;
 
// MAC Address of responder - edit as required
uint8_t broadcastAddress[] = {0x14, 0x33, 0x5C, 0x52, 0x4D, 0x00};
 
// Define a data structure
typedef struct struct_message {
  //char a[32];
  int TrackNumber;
  //float c;
  //bool d;
} struct_message;
 
// Create a structured object
struct_message myData;
 
// Peer info
esp_now_peer_info_t peerInfo;
 
// Callback function called when data is sent

void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) 
{
  Serial.print("\r\nLast Packet Send Status:\t");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Delivery Success" : "Delivery Fail");
}

void setup() {
  
  // Set up Serial Monitor
  Serial.begin(115200);
 
  // Set ESP32 as a Wi-Fi Station
  WiFi.mode(WIFI_STA);
 
  // Initilize ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }
 
  // Register the send callback
  esp_now_register_send_cb(OnDataSent);
  
  // Register peer
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;
  
  // Add peer        
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }
}
 
void loop() {
 
  // Create test data
 
  // Generate a random integer
  int_value = random(1,20);
 
  // Use integer to make a new float
  float_value = 1.3 * int_value;
 
  // Invert the boolean value
  bool_value = !bool_value;
  
  // Format structured data
  strcpy(myData.TrackNumber, 15); //Manik senfing number 15
  //myData.TrackNumber = int_value;
  //myData.c = float_value;
  //myData.d = bool_value;
  
  // Send message via ESP-NOW
  esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *) &myData, sizeof(myData));
   
  if (result == ESP_OK) {
    Serial.println("Sending confirmed");
  }
  else {
    Serial.println("Sending error");
  }
  delay(2000);
}
```

### Debug Message

```plain
C:\Users\User\Desktop\HelloWorld\HelloWorld\HelloWorld.ino: In function 'void setup()':
C:\Users\User\Desktop\HelloWorld\HelloWorld\HelloWorld.ino:63:28: error: invalid conversion from 'void (*)(const uint8_t*, esp_now_send_status_t)' {aka 'void (*)(const unsigned char*, esp_now_send_status_t)'} to 'esp_now_send_cb_t' {aka 'void (*)(const wifi_tx_info_t*, esp_now_send_status_t)'} [-fpermissive]
   63 |   esp_now_register_send_cb(OnDataSent);
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-07-30:
Please note that this forum is for issues with the code here, not your code.  Please use https://esp32.com for general programing questions.

Change `void OnDataSent(...` to `esp_now_send_cb_t OnDataSent(...`

Manik601292, 2025-08-02:
> Please note that this forum is for issues with the code in this repository, not your code. Please use https://esp32.com or discord for general programing questions.
> 
> Change `esp_now_register_send_cb(OnDataSent);` to `esp_now_register_send_cb((esp_now_send_cb_t)OnDataSent);`

Thank you Soooooooooooooooo much. It worked :).  Also thanks for your suggestions.

Manik601292, 2025-08-02:
Thank you Soooooooooooooooo much. It worked :). Also thanks for your suggestions.

closed 2025-08-02 by Manik601292