espressif/arduino-esp32 #11686: ESP32-S3 SupperMini clone with batt pins

lkos98, 2025-08-02:
### Board

ESP32-S3 Suppermini

### Device Description

Devkit

### Hardware Configuration

12-13 0.96' OLED

### Version

v3.3.0

### Type

Bug

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

QIO 80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Just reporting a problem with ESP32-S3 Suppermini clone. When a battery is attached to battery pins and code is uploaded a blocking Schottky diode between Batt+ and V+ (also going to IN pin on LDO J2RF) fails. As a result board does not work on battery only (2.4V on 3.3V pin) . All the rest is ok, including charging of the battery. 
Make sure when uploading ALWAYS disconnect the battery first! 

### Sketch

```cpp
I guess any sketch, but probably bigger ones would create more power surge through the diode.
```

### Debug Message

```plain
Replace the diode
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2025-08-02:
This seems more likely to be a manufacturing defect that should be documented by the manufacturer and not a failure of code or upload process by this repository.

However, it is always a good idea to disconnect batteries if the circuit does not include support for preventing the full 5v power from going to the battery which may not support such voltages.

lkos98, 2025-08-02:
It is a manufacturing defect - actually a flaw and bad design. The diode is not powerful enough and also there are no bypassing capacitors to kill spikes. My apologies - I put this post in this repository, just because I didn't know where else to put it, it's a hardware issue. I'm not pulling any request, just wanted to warn more people when using battery powered S3 supperminis. I lost hours of searching and seems there are just few people who had similar issues.

lbernstone, 2025-08-02:
https://esp32.com is a better place to post this

closed 2025-08-02 by Jason2866