espressif/arduino-esp32 #11690: [FR] Add Arduino support for IP_EVENT_TX_RX to count traffic

TD-er, 2025-08-03:
### Related area

NetworkInterface

### Hardware specification

-

### Is your feature request related to a problem?

-

### Describe the solution you'd like

ESP-IDF5.1 and newer appear to have support for events to help count traffic sent and received via that interface.
See `ip_event_t::IP_EVENT_TX_RX` in `esp_netif_types.h` and https://github.com/espressif/esp-idf/blob/4e036983a751e4667ade94c8f6f6bf1e7f78eff0/components/esp_netif/lwip/esp_netif_lwip.c#L1362

This would allow for see how much data is being sent via an interface and its direction.
This would be great for debugging and/or SNMP-like applications.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

ritesh006, 2025-08-31:
### Enable `IP_EVENT_TX_RX` in Arduino‑ESP32 (beginner note)

Hi! I tried this on my **ESP32‑WROOM‑32** with **Arduino core 3.x** (which uses **ESP‑IDF 5.1+**). I’m not very advanced, but this worked for me and might help others.

**What it does:**  
When I enable `IP_EVENT_TX_RX`, I get an event for every packet. From that event I read the direction (RX or TX) and the length, and I keep adding those lengths to two counters. So I can see how many bytes went in (RX) and out (TX).

**My setup:**  
- Board: ESP32‑WROOM‑32 (no PSRAM)  
- Arduino IDE: core 3.x  
- Serial Monitor: 115200 baud  
- (If PSRAM errors appear, I just disable PSRAM in Tools and re‑flash.)

**Small code I used (short version):**
```cpp
extern "C" { 
  #include "esp_event.h"
  #include "esp_netif.h"
  #include "esp_netif_types.h"
}
static volatile uint64_t rx_bytes = 0, tx_bytes = 0;

static void on_tx_rx(void*, esp_event_base_t, int32_t, void* data){
  auto* e = (ip_event_tx_rx_t*) data; 
  if (!e) return;
  if (e->dir == ESP_NETIF_RX) rx_bytes += e->len; 
  else tx_bytes += e->len;
}

void enable_txrx_events(){
  auto* sta = esp_netif_get_handle_from_ifkey("WIFI_STA_DEF");
  if (sta) esp_netif_tx_rx_event_enable(sta);
  esp_event_handler_instance_t inst;
  esp_event_handler_instance_register(IP_EVENT, IP_EVENT_TX_RX, &on_tx_rx, nullptr, &inst);
}
```

I call `enable_txrx_events()` **after** Wi‑Fi connects. Then I make a small HTTP request every few seconds (just to create some traffic), and print the counters.

**My serial output looked like this:**
```
WiFi status=3, IP=192.168.1.9
esp_netif_tx_rx_event_enable -> 0
register(IP_EVENT_TX_RX) -> 0
Traffic — RX: 1970, TX: 624
Traffic — RX: 3920, TX: 1327
...
```

**Why I’m sharing this:**  
I think it would be helpful if Arduino‑ESP32 enabled these per interface and offered a simple API to read/reset the counters (something like `WiFiGetTraffic(rx, tx)` and maybe a reset function). If I misunderstood anything, please correct me.

### Screnshot from my Log
[image]

TD-er, 2025-09-02:
Got it working too...
The key is to set the `esp_netif_tx_rx_event_enable` after the connection is established.

ritesh006, 2025-09-02:
If this working then shall we close this

closed 2026-10-01 by me-no-dev