espressif/arduino-esp32 #11697: SDK Configuration editor confserver error (IDFGH-16190)

txf-, 2025-08-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.2 (release), v5.5.0 (release)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect to be able to configure the project and open the menuconfig menu.

### What is the actual behavior?

It terminates with SDK Configuration editor confserver process exited with code: 1

### Steps to reproduce.

use the idf version, Add esp32-arduino (v3.3.0 as a managed component. Try to open menuconfig.

### Build or installation Logs.

The is the python stack report:
```
Server running, waiting for requests on stdin...

---------------------------ERROR--------------------------

Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/streams.py", line 545, in readline

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    line = await self.readuntil(sep)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/streams.py", line 640, in readuntil
    raise exceptions.LimitOverrunError(
asyncio.exceptions.LimitOverrunError: Separator is found, but chunk is longer than limit

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf.py", line 855, in <module>
    main()
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf.py", line 745, in main
    cli(argv, prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "/home/tiago/.espressif5x/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tiago/.espressif5x/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/tiago/.espressif5x/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages/click/core.py", line 1729, in invoke
    return _process_result(rv)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/tiago/.espressif5x/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages/click/core.py", line 1666, in _process_result

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    value = ctx.invoke(self._result_callback, value, **ctx.params)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tiago/.espressif5x/python_env/idf5.4_py3.11_env/lib/python3.11/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf.py", line 640, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf.py", line 212, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf_py_actions/core_ext.py", line 50, in build_target
    run_target(target_name, args, force_progression=GENERATORS[args.generator].get('force_progression', False))
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf_py_actions/tools.py", line 499, in run_target
    RunTool(generator_cmd[0], generator_cmd + [target_name], args.build_dir, env, custom_error_handler, hints=not args.no_hints,
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf_py_actions/tools.py", line 319, in __call__
    process, stderr_output_file, stdout_output_file = asyncio.run(self.run_command(self.args, env_copy))
                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf_py_actions/tools.py", line 364, in run_command

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    await asyncio.gather(
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf_py_actions/tools.py", line 441, in read_and_write_stream
    output = await read_stream()
             ^^^^^^^^^^^^^^^^^^^
  File "/home/tiago/esp/v5.4.2/esp-idf/tools/idf_py_actions/tools.py", line 399, in read_stream
    output_b = await input_stream.readline()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/streams.py", line 554, in readline
    raise ValueError(e.args[0])
ValueError: Separator is found, but chunk is longer than limit

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 1
```

### More Information.

It fails on windows too, with the versions above provided. It also fails with idf 5.5 and esp32-arduino v3.1.0, I had to downgrade to esp idf v5.3.1 and arduino v3.1.0 , which seems to be the only configuration that works. 

It should be noted that all the other configurations compile, it is just that configuration fails.

txf-, 2025-08-12:
here it is. This is the doctor command using idf v5.4.2. I'll add it as an attachment as it i very long. 

[report.txt](https://github.com/user-attachments/files/21729118/report.txt)

me-no-dev, 2025-08-12:
> Is it ESP-IDF v5.5 or ESP-IDF v5.4.2 supported by esp32-arduino v3.3.0 installed by component manager ?

It supports IDF 5.3.0-5.5

Parsaabasi, 2025-08-26:
Hi @txf- 

Considering the comments above, would you need any further assistance with this issue? 

Kind regards

brianignacio5, 2025-08-26:
Hi @txf- It seems the issue happens in ESP-IDF idf.py confserver itself.

Until we fix it  in upstream please do this hot fix

https://github.com/espressif/esp-idf/blob/master/tools/idf_py_actions/tools.py#L351

Modify the limit in your ESP-IDF `tools/idf_py_actions/tools.py` line 351 (based on current master)  file from 1024*256 to 1024*512 or any value that fit your needs.

Parsaabasi, 2025-09-01:
Hi @txf- 

This issue seems resolved. Therefore, I close the case. In case you need further assistance, please feel free to reopen it or create another issue. 

Kind regards,