espressif/arduino-esp32 #11711: RTC_DATA_ATTR performs differently when used with "struct" depending on initial-values

Rob58329, 2025-08-11:
### Board

ESP32 Dev Module (NodeMCU)

### Device Description

ESP32 Dev Module (NodeMCU) standalone.

### Hardware Configuration

None

### Version

other

### Type

Bug

### IDE Name

Arduino v.1.8.19 with github.com/espressif/arduino-esp32 as at 1June25.

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

RTC_DATA_ATTR performs differently when used with "struct" depending on initial-values

Specifically in some instances the RTC_DATA_ATTR variables are always reset to their default values after deepsleep, but in other instances variables are stored and returned fine after deepsleep.

### Sketch

```cpp
// RTC_DATA_ATTR performs differently when used with "struct" depending on initial-values

RTC_DATA_ATTR int data0=0; // increments

RTC_DATA_ATTR struct {
  int x1;   // increments (start value is nominally zero, but potentially random)
  int x2=5; // fixed
  int x3=6; // fixed
} data1; // it appears that if you leave a single element undefined, it fixes all the other values at their intial values.
//          -------------------------------------------------------------------------------------------------------------

RTC_DATA_ATTR  struct {
  int x1=0; // increments
  int x2=1; // increments
  int x3=2; // increments
} data2; // but if we ensure ALL variables have inital values it works as expected

void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;
  wakeup_reason = esp_sleep_get_wakeup_cause();
  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:     Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1:     Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER:    Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD: Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP:      Serial.println("Wakeup caused by ULP program"); break;
    default:                        Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason); break;
  }
}

void setup() {
  Serial.begin(115200);
  delay(2500);
  Serial.println("---------------");
  print_wakeup_reason();
  Serial.println("---------------");
  Serial.printf("data0   =%i=ok\n",data0++);
  Serial.println();
  Serial.printf("data1.x1=%i=ok\n",data1.x1++);
  Serial.printf("data1.x2=%i=fixed\n",data1.x2++);
  Serial.printf("data1.x3=%i=fixed\n",data1.x3++);
  Serial.println();
  Serial.printf("data2.x1=%i=ok\n",data2.x1++);
  Serial.printf("data2.x2=%i=ok\n",data2.x2++);
  Serial.printf("data2.x3=%i=ok\n",data2.x3++);
  Serial.println("---------------");

  // #define WAKEUP_GPIO GPIO_NUM_0     // The "Boot/Press-to-flash" button on the NodeMCU
  // esp_sleep_enable_ext0_wakeup(WAKEUP_GPIO, 0);  //1 = High, 0 = Low
  esp_sleep_enable_timer_wakeup(10 * 1000000); // 10s
  Serial.println("Going to sleep for 10s");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop() {}
```

### Debug Message

```plain
---------------
Wakeup was not caused by deep sleep: 0
---------------
data0   =0=ok

data1.x1=0=ok
data1.x2=5=fixed
data1.x3=6=fixed

data2.x1=0=ok
data2.x2=1=ok
data2.x3=2=ok
---------------
Going to sleep for 10s

ets Jul 29 2019 12:21:46
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
---------------
Wakeup caused by timer
---------------
data0   =1=ok

data1.x1=1=ok
data1.x2=5=fixed
data1.x3=6=fixed

data2.x1=1=ok
data2.x2=2=ok
data2.x3=3=ok
---------------
Going to sleep for 10s
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-08-25:
Confirmed. Interesting issue, thanks for reporting.
This can't be fixed in the Arduino Core code.

We will report it to the compiler tools team.

SuGlider, 2025-08-26:
Closing given that Arduino Core team can't fix it.

closed 2025-08-26 by SuGlider