espressif/arduino-esp32 #11714: Allow USB Midi descriptor to be changed

aaronaverill, 2025-08-11:
### Related area

USBMIDI

### Hardware specification

Generic

### Is your feature request related to a problem?

The USB device descriptor always shows up as "TinyUSB MIDI", I can't change it

### Describe the solution you'd like

Change the literal in this code to a define: USB_MIDI_DESCRIPTOR

https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/src/USBMIDI.cpp#L21

uint8_t str_index = tinyusb_add_string_descriptor(USB_MIDI_DESCRIPTOR);

At the top of the module, add the following block:
#ifndef USB_MIDI_DESCRIPTOR
#define USB_MIDI_DESCRIPTOR "TinyUSB MIDI"
#endif

Now a build define will allow this to be changed so it shows up correctly under the MIDI devices.

### Describe alternatives you've considered

This is being called by a library I use, so there is really no way to override the name without a whole bunch of copied code.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2025-08-12:
@aaronaverill - The proposed solution inthe PR #11720 would be like this:

``` cpp
// Create the MIDI device with specific descriptor
USBMIDI MIDI("ESP MIDI Device");
```

SuGlider, 2025-08-22:
@aaronaverill - OK, got it.

I'll modify the PR to include other ways to change the MIDI Device Name.
It shall be compatible with MIDI Libraries that run the original ESP32 USBMIDI Code.
I'll also create some very simple examples that will demonstrate how to implement it in the Arduino Sketch.