espressif/arduino-esp32 #11726: Cannot compile the code(Compilation error: exit status 1)

Yorkeryu, 2025-08-14:
### Board

ESP32 Dev Moudel

### Device Description

ESP32 Board

### Hardware Configuration

nothing

### Version

v3.2.0

### Type

Bug

### IDE Name

Arduino IDE

### Operating System

Windows10

### Flash frequency

240MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When I tried to compile a code,it didn't work,and speak

[image]

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
ϵͳ�Ҳ���ָ����·����
exit status 1

Compilation error: exit status 1
```

### Other Steps to Reproduce

I tried to upload the board extension to the latest version,but it didn't work either

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Yorkeryu, 2025-08-14:
I think that's all

[image]

[image]

SuGlider, 2025-08-14:
Please make sure that the ESP32 is selected in the board menu and the COM port is correct.

SuGlider, 2025-08-14:
The problem may be related to the user name `Y.x.Y`.
Arduino + Windows has some limitations related to the path characters.

What is the Arduino Core Version installed?

`C:\Users\Y.x.Y\AppData\Local\Arduino15\packages\esp32\hardware\esp32\111` doesn't match any standard version `111`.
It should be something like `C:\Users\Y.x.Y\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0` for the latest Arduino Version available at this time.

Yorkeryu, 2025-08-14:
so how can I fix it

SuGlider, 2025-08-14:
@Yorkeryu - Please clarify which Arduino Core is being used.

Open the Arduino IDE Menu -> Tools -> "Board: ESP32 Wrover Module" -> Boards Manager ...
Go to the Boards Manager TAB that will be open in the left side of the IDE.
Type "esp32" and press ENTER.

A list of boards for ESP32 will be displayed.
Locate the "esp32 by Espressif Systmes" information and verify which Core Version is installed.

Or send us the Screenshot of the IDE.

SuGlider, 2025-08-14:
I have tested using the Arduino IDE with a userName exactly the same as "Y.x.Y" and it has worked correctly within Arduino IDE 2.3.6 for the ESP32 board, using Arduino Core 3.3.0.

SuGlider, 2025-08-14:
The issue seems to be related to the Arduino IDE environment.
There is no Core version "111" as it seems to be used by the IDE:

**Using core 'esp32' from platform in folder: C:\Users\Y.x.Y\AppData\Local\Arduino15\packages\esp32\hardware\esp32\111**

the `111` folder must be something like `3.3.0` or `3.2.1`or any other valid Arduino Core version. This may be the issue because the IDE can't find `esptool.py` in the correct folder.

SuGlider, 2025-08-14:
> It is very possible that in order to solve it, it is necessary to rename `C:\Users\Y.x.Y\AppData\Local\Arduino15\packages\esp32\hardware\esp32\111` folder to `C:\Users\Y.x.Y\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0` or the version that can seen in the Board Manager.

NOPE... I have tried it and it failed in other step.

Yorkeryu, 2025-08-15:
I have reinstalled it, but it still doesn't work
```
FQBN: esp32:esp32:esp32:DebugLevel=verbose
Using board 'esp32' from platform in folder: C:\Users\Y.x.Y\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0
Using core 'esp32' from platform in folder: C:\Users\Y.x.Y\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0

cmd /c if exist "C:\\Users\\Y.x.Y\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025715-23276-xxx5l1.mkcd\\sketch_aug15a\\partitions.csv" COPY /y "C:\\Users\\Y.x.Y\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025715-23276-xxx5l1.mkcd\\sketch_aug15a\\partitions.csv" "C:\\Users\\Y.x.Y\\AppData\\Local\\arduino\\sketches\\554DBD8EEF6B451BBB0A2F4B43C73656\\partitions.csv"
cmd /c if not exist "C:\\Users\\Y.x.Y\\AppData\\Local\\arduino\\sketches\\554DBD8EEF6B451BBB0A2F4B43C73656\\partitions.csv" if exist "C:\\Users\\Y.x.Y\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\variants\\esp32\\partitions.csv" COPY "C:\\Users\\Y.x.Y\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\variants\\esp32\\partitions.csv" "C:\\Users\\Y.x.Y\\AppData\\Local\\arduino\\sketches\\554DBD8EEF6B451BBB0A2F4B43C73656\\partitions.csv"
cmd /c if not exist "C:\\Users\\Y.x.Y\\AppData\\Local\\arduino\\sketches\\554DBD8EEF6B451BBB0A2F4B43C73656\\partitions.csv" COPY "C:\\Users\\Y.x.Y\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\tools\\partitions\\default.csv" "C:\\Users\\Y.x.Y\\AppData\\Local\\arduino\\sketches\\554DBD8EEF6B451BBB0A2F4B43C73656\\partitions.csv"
ϵͳ�Ҳ���ָ����·����
�Ѹ���         1 ���ļ���
cmd /c IF EXIST "C:\\Users\\Y.x.Y\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025715-23276-xxx5l1.mkcd\\sketch_aug15a\\bootloader.bin" ( COPY /y "C:\\Users\\Y.x.Y\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025715-23276-xxx5l1.mkcd\\sketch_aug15a\\bootloader.bin" "C:\\Users\\Y.x.Y\\AppData\\Local\\arduino\\sketches\\554DBD8EEF6B451BBB0A2F4B43C73656\\sketch_aug15a.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\Y.x.Y\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\variants\\esp32\\bootloader.bin" ( COPY "C:\\Users\\Y.x.Y\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\variants\\esp32\\bootloader.bin" "C:\\Users\\Y.x.Y\\AppData\\Local\\arduino\\sketches\\554DBD8EEF6B451BBB0A2F4B43C73656\\sketch_aug15a.ino.bootloader.bin" ) ELSE ( "C:\\Users\\Y.x.Y\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0/tools/esptool\\esptool.exe" --chip esp32 elf2image --flash-mode dio --flash-freq 80m --flash-size 4MB -o "C:\\Users\\Y.x.Y\\AppData\\Local\\arduino\\sketches\\554DBD8EEF6B451BBB0A2F4B43C73656\\sketch_aug15a.ino.bootloader.bin" "C:\\Users\\Y.x.Y\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\tools\\esp32-arduino-libs\\esp32\\bin\\bootloader_qio_80m.elf" ) )
exit status 1

Compilation error: exit status 1
```

Yorkeryu, 2025-08-16:
Thank you so much，now it can work smoothly！

closed 2025-08-16 by lucasssvaz