espressif/arduino-esp32 #11733: Outgoing Interface Issue

Arolarobola, 2025-08-16:
### Board

ESP32 DEV MOD

### Device Description

ESP32 Dev MOD

### Hardware Configuration

I have three buttons connected.

### Version

latest stable Release (if not listed below)

### Type

Question

### IDE Name

VSCode

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi I am trying to use this code to send out UDP packets to devices on my network from my ESP32. The issue I'm running into is that I essentially have 3 networks on my ESP32... AP+STA+Ethernet and I need the ability to specify which network interface that AsyncUDP uses to send out the UDP packets through. In my testing only the AP network works for sending out packets. If my remote device is connected through either the STA or Ethernet the packets are still only sent through the AP which means it's blocking me from sending packets to devices on the STA or Ethernet networks. Can anybody please shed some light on how I can specify which Network Interface on the ESP32 that the AsyncUDP uses and how i can use this library to work with ALL of the Network Interfaces? This may be more of an ESP32 issue however I haven't been able to find any information concerning this topic. Please point me in the right direction. Thank you

### Sketch

```cpp
#include <AsyncUDP.h>
AsyncUDP aUDP;
if (aUDP.connect(tipAddress, tiport)) {} // i can specify the remote ip and port however i cannot specify which ESP32 Network Interface to use to send out FROM. This matters due to which network the remote device is connected to.
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-08-16:
Maybe @me-no-dev can answer it.

lbernstone, 2025-08-16:
For WiFi connections, this should just use standard IP stack methods of route finding, ie the address should only be routable down one path.  If there is ambiguity, you should be able to use setDefaultInterface:
```
Network.setDefaultInterface(WiFi.STA);
Network.setDefaultInterface(WiFi.AP);
Network.setDefaultInterface(ETH);
```

lbernstone, 2025-08-29:
I tested this with a firmware that sends udp messages out to both a device on the AP (192.168.4.2) and one on my main wifi network (192.168.2.216). It does exactly as expected, and if I move my laptop back and forth between the networks, I get the messages for the respective targets.

Parsaabasi, 2025-10-01:
Hi @Arolarobola 

Seems like this issue is resolved, I close it. In case you would need further assistance, please feel free to reopen it or create a new one. 

Kind regards.

closed 2025-10-01 by Parsaabasi