espressif/arduino-esp32 #11740: ESP NOW Compilation Error

mohdrais, 2025-08-17:
### Board

ESP32C3 Dev Module

### Device Description

plain ESP32C3 Dev Module

### Hardware Configuration

All are gpio
#define IRSensor 3
#define capSensor1 5
#define capSensor2 7
#define Relay 0

### Version

latest stable Release (if not listed below)

### Type

Task

### IDE Name

Aeduino IDE 2.3.5

### Operating System

Window 11

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Hello, last time there were no problems with 
esp_now_register_send_cb(OnDataSent);
but now when I want to compile, I got this problem.
Please help

Compilation error: invalid conversion from 'void (*)(const uint8_t*, esp_now_send_status_t)' {aka 'void (*)(const unsigned char*, esp_now_send_status_t)'} to 'esp_now_send_cb_t' {aka 'void (*)(const wifi_tx_info_t*, esp_now_send_status_t)'} [-fpermissive]

### Sketch

```cpp
#include <ESP32_NOW.h>
#include <WiFi.h>

#define IRSensor 3
#define capSensor1 5
#define capSensor2 7
#define Relay 0

unsigned int uintIRState ;
unsigned int uintCAPState1 ;
unsigned int uintCAPState2 ;
int intResult;
String strIR;
String strCap;
// REPLACE WITH YOUR RECEIVER MAC Address
uint8_t broadcastAddress[] = {0x18, 0x8b, 0x0e, 0x86, 0x08, 0xd0};

typedef struct struct_message {
  int intCAP;
} struct_message;
// Create a struct_message called myData
struct_message myData;
unsigned long duluMillis = 0;   // Stores last time temperature was published
const long selang = 50000;        // Interval at which to publish sensor readings for every 50 sec
esp_now_peer_info_t peerInfo;

// callback when data is sent
void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
  Serial.print("\r\nLast Packet Send Status:\t");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Delivery Success" : "Delivery Fail");
}

void setup() {
  // put your setup code here, to run once:
  
  Serial.begin(115200);
    // Set device as a Wi-Fi Station
  WiFi.mode(WIFI_STA);
    // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }
    // Once ESPNow is successfully Init, we will register for Send CB to
  // get the status of Trasnmitted packet
  esp_now_register_send_cb(OnDataSent);
    // Register peer
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;
    // Add peer        
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }
  pinMode(IRSensor,INPUT);
  pinMode(capSensor1,INPUT);
  pinMode(capSensor2,INPUT);
  pinMode(Relay,OUTPUT);
  uintIRState = 0;
  uintCAPState1 = 0;
  uintCAPState2 = 0;

}

void loop() {
  // put your main code here, to run repeatedly:
  uintIRState = digitalRead(IRSensor);
  if (uintIRState == 1)
    {
      digitalWrite(Relay, HIGH);
      strIR = "Jalan";
    }
    else
    {
      digitalWrite(Relay, LOW);
      strIR = "Henti"; 
    } 
  uintCAPState1 = digitalRead(capSensor1);
  uintCAPState2 = digitalRead(capSensor2);
  if (uintCAPState1 == 0 && uintCAPState2 == 0)
    {
      strCap = "Ada Dedak";
      intResult = 0;
    }
    else
    {
      strCap = "Tiada Dedak";
      intResult = 1;
    } 
  myData.intCAP = intResult;
  Serial.println("-----------------");
  Serial.print("myData.intCAP = ");
  Serial.println(myData.intCAP);
  Serial.println("************");
  Serial.print("strIR = ");
  Serial.println(strIR);
  Serial.print("strCap = ");
  Serial.println(strCap);

  // Send message via ESP-NOW
  esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *) &myData, sizeof(myData));
  if (result == ESP_OK) {
    Serial.println("Sent with success");
  }
  else {
    Serial.println("Error sending the data");
  }
  delay(2500);
}
```

### Debug Message

```plain
Compilation error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-08-17:
@mohdrais - ESP-NOW has changed the function parameters declaration in `esp_now_send_status_t ` fro IDF 5.4 to IDF 5.5

Please compare the declaration of `void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status)` from the sketch shown in this issue from the declaration of `static void example_espnow_send_cb(const esp_now_send_info_t *tx_info, esp_now_send_status_t status)` from the ESP-IDF example from https://github.com/espressif/esp-idf/blob/release/v5.5/examples/wifi/espnow/main/espnow_example_main.c

``` cpp
/* ESPNOW sending or receiving callback function is called in WiFi task.
 * Users should not do lengthy operations from this task. Instead, post
 * necessary data to a queue and handle it from a lower priority task. */
static void example_espnow_send_cb(const esp_now_send_info_t *tx_info, esp_now_send_status_t status)
{
    example_espnow_event_t evt;
    example_espnow_event_send_cb_t *send_cb = &evt.info.send_cb;

    if (tx_info == NULL) {
        ESP_LOGE(TAG, "Send cb arg error");
        return;
    }

    evt.id = EXAMPLE_ESPNOW_SEND_CB;
    memcpy(send_cb->mac_addr, tx_info->des_addr, ESP_NOW_ETH_ALEN);
    send_cb->status = status;
    if (xQueueSend(s_example_espnow_queue, &evt, ESPNOW_MAXDELAY) != pdTRUE) {
        ESP_LOGW(TAG, "Send send queue fail");
    }
}
```

SuGlider, 2025-08-17:
``` cpp
// callback when data is sent
void OnDataSent(const esp_now_send_info_t *tx_info, esp_now_send_status_t status){
//void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
  Serial.print("\r\nLast Packet Send Status:\t");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Delivery Success" : "Delivery Fail");
}

```

closed 2025-08-20 by SuGlider

mohdrais, 2025-08-21:
> Closing this issue. Please feel free to open it again, if necessary.

Thank you very much