espressif/arduino-esp32 #11745: Zigbee.begin(ZIGBEE_ROUTER) creates own network, if a previous network is unavailable

madmagic007, 2025-08-19:
### Board

ESP32C6

### Device Description

Supermini

### Hardware Configuration

-

### Version

latest stable Release (if not listed below)

### Type

Bug

### IDE Name

VSCode with PlatformIO

### Operating System

Linux Nobara 42

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I want to make my ESP32C6 use zigbee for wireless communication with ROUTER mode to create a mesh network.
I am running zigbee2mqtt on a raspberry pi 4 with a skyconnect dongle. 

If the esp has successfully connected to the z2m network before, but then is out of reach / the z2m network is down, `Zigbee.connected()` still returns true, and it appears the esp has created its own zigbee network even tho i dont tell it to. This is 100% reproducible for me across all 5 C6 Supermini boards i currently have.

If instead i begin the zigbee stack with `Zigbee.begin(ZIGBEE_END_DEVICE)`, the stack never gets past the begin() call, even when a prior connection to the z2m network was made.

Adding to this, it appears that with a clean nvs `Zigbee.begin()` passes and returns false on `Zigbee.connected()`. But if a prior connection was established that now is unavailable, the execution halts on `Zigbee.begin()` and only makes it past once the network is available.

I tried to get more info with `Zigbee.setDebugMode(true);`, but i am unsure how to see these logs.

### Sketch

```cpp
#include <Arduino.h>
#include "Zigbee.h"

ZigbeeAnalog zbAnalogDevice = ZigbeeAnalog(1);

void setup() {
    Serial.begin(115200);
    delay(2000);

    rgbLedWrite(RGB_BUILTIN, 1, 0, 0);

    zbAnalogDevice.setManufacturerAndModel("Espressif", "ZigbeeAnalogDevice");
    Zigbee.addEndpoint(&zbAnalogDevice);

    Serial.println("before zigbee begin");
    if (!Zigbee.begin(ZIGBEE_ROUTER)) {
        Serial.println("Zigbee failed to start!");
        Serial.println("Rebooting...");
        ESP.restart();
    }

    rgbLedWrite(RGB_BUILTIN, 0, 0, 1);

    Serial.println("before zigbee connected");
    while (!Zigbee.connected()) delay(100);

    Serial.println("zigbee is connected");
    Serial.println(esp_zb_get_pan_id());
    rgbLedWrite(RGB_BUILTIN, 0, 1, 0);
}

void loop() {
}
```

### Debug Message

```plain
Serial output:

ROUTER/END_DEVICE mode, network available:
before zigbee begin
before zigbee connected
zigbee is connected
22215

ROUTER mode, network NOT available:
before zigbee begin
before zigbee connected
zigbee is connected
22215

END_DEVICE mode, network NOT available:
before zigbee begin
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

madmagic007, 2025-08-21:
Update, i managed to get logs printed of the zigbee stack on the MODE_ROUTER device where the z2m is unavailable but has connected before:

```
[  2689][D][ZigbeeController.cpp:15] setupTask(): [MY_CODE] calling Zigbee.begin
[  2689][D][ZigbeeCore.cpp:149] zigbeeInit(): Initialize Zigbee stack
[  2768][W][ZigbeeCore.cpp:154] zigbeeInit(): No Zigbee EPs to register
[  2779][V][ZigbeeCore.cpp:447] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[  2779][I][ZigbeeCore.cpp:260] esp_zb_app_signal_handler(): Zigbee stack initialized
[  2780][D][ZigbeeCore.cpp:261] esp_zb_app_signal_handler(): Zigbee channel mask: 0x00000800
[  2784][I][ZigbeeCore.cpp:267] esp_zb_app_signal_handler(): Device started up in non factory-reset mode
[  2785][I][ZigbeeCore.cpp:280] esp_zb_app_signal_handler(): Device rebooted
[  2786][V][ZigbeeCore.cpp:750] setNVRAMChannelMask(): Channel mask set to 0x00000800
[  2786][D][ZigbeeCore.cpp:735] searchBindings(): Requesting binding table for address 0xe816
[  2787][D][ZigbeeCore.cpp:543] bindingTableCb(): Binding table callback for address 0xe816 with status 0
[  2788][D][ZigbeeCore.cpp:547] bindingTableCb(): Binding table info: total 0, index 0, count 0
[  2788][D][ZigbeeCore.cpp:550] bindingTableCb(): No binding table entries found
[  2789][D][ZigbeeController.cpp:22] setupTask(): [MY_CODE] Zigbee.begin(ZIGBEE_ROUTER) passed
[  2790][V][esp32-hal-rmt.c:457] rmtInit(): GPIO 8 - TX MODE - MemSize[48] - Freq=10000000Hz
[  2791][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 8 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  2792][V][esp32-hal-rmt.c:307] _rmtWrite(): GPIO: 8 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  2795][D][ZigbeeController.cpp:26] setupTask(): [MY_CODE] Zigbee.connected passed
[  2795][D][ZigbeeController.cpp:28] setupTask(): [MY_CODE] PAN ID: 0x56c7
[  2796][D][ZigbeeController.cpp:33] setupTask(): [MY_CODE] Extended PAN ID: fb:30:9d:4a:7e:d4:17:3c
[  2797][V][esp32-hal-rmt.c:457] rmtInit(): GPIO 8 - TX MODE - MemSize[48] - Freq=10000000Hz
[  2798][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 8 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  2800][V][esp32-hal-rmt.c:307] _rmtWrite(): GPIO: 8 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[ 17782][V][ZigbeeCore.cpp:447] esp_zb_app_signal_handler(): ZDO signal: NWK No Active Links Left (0x18), status: ESP_OK
[ 32783][V][ZigbeeCore.cpp:447] esp_zb_app_signal_handler(): ZDO signal: NWK No Active Links Left (0x18), status: ESP_OK
```

note the `MY_CODE` logs before and after Zigbee.begin(ZIGBEE_ROUTER), and after Zigbee.connected. the bottom 2 lines just repeat indefinetly.

i used a second esp32C6 supermini to scan for networks:
```
Scan done
1 networks found:
Nr | PAN ID | CH | Permit Joining | Router Capacity | End Device Capacity | Extended PAN ID
 1 | 0x56c7 | 11 | No             | Yes             | Yes                 | fb:30:9d:4a:7e:d4:17:3c
```

this matches 100% with PAN ID and Extended PAN ID of the ROUTER device. so this confirms that a network is actually being created instead of the functions falsely passing

Logs for ZIGBEE_END_DEVICE where z2m network unavailable but has connected before:
```
[  2707][D][ZigbeeController.cpp:15] setupTask(): [MY_CODE] calling Zigbee.begin
[  2707][D][ZigbeeCore.cpp:149] zigbeeInit(): Initialize Zigbee stack
[  2790][W][ZigbeeCore.cpp:154] zigbeeInit(): No Zigbee EPs to register
[  2802][V][ZigbeeCore.cpp:447] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[  2802][I][ZigbeeCore.cpp:260] esp_zb_app_signal_handler(): Zigbee stack initialized
[  2803][D][ZigbeeCore.cpp:261] esp_zb_app_signal_handler(): Zigbee channel mask: 0x00000800
[  3044][W][ZigbeeCore.cpp:295] esp_zb_app_signal_handler(): Commissioning failed, trying again...
[  3785][W][ZigbeeCore.cpp:295] esp_zb_app_signal_handler(): Commissioning failed, trying again...
[  4526][W][ZigbeeCore.cpp:295] esp_zb_app_signal_handler(): Commissioning failed, trying again...
```

P-R-O-C-H-Y, 2025-08-21:
Hi @madmagic007, thank you for the report and logs. I will test this issue and reach back to you.

madmagic007, 2025-08-22:
@P-R-O-C-H-Y thanks for the explanation.
Is there a way to disable this behavior?

In my specific setup, the coordinator (dongle connected to z2m) is a necessary aspect of the network. Without it, the other devices cant function properly. as there is a handler running on the mqtt network that takes care of everything. I want the router devices to form a mesh to extend the range.

If the router reboots and it cant reach the z2m coordinator, the device creates its own network thus the entire setup failing to work. This way it is also not easy to visualize (status led) or debug via prints if it is connected to the z2m or not since the connected just returns true.

It would be nice if this was a configurable fallback setting.

P-R-O-C-H-Y, 2025-08-22:
@madmagic007 I can take a look, but this is how the Zigbee works.

The main thing is:

- Only the coordinator can form a Zigbee network. It chooses the PAN ID, channel, and establishes the initial Trust Center (security) if one is used. The network is not destroyed when the coordinator goes offline. Once the network is formed, it "exists" independently of the coordinator. All routers and end devices that already have the network credentials (PAN ID, channel, network key) can keep operating and communicating.

- A router cannot create a Zigbee network. When the coordinator is offline, routers don’t "take over" that role — they simply keep the existing network alive by maintaining routing tables, handling mesh traffic, and allowing new devices to join (if the Trust Center policies allow it).

- In most Zigbee networks, the coordinator is also the Trust Center (handles key distribution and authorization of new devices). If that’s the case and the coordinator is offline:

   - Existing devices will continue working.

   - Routers may still allow devices to attach if they already have the network key.

   - But completely new devices cannot join securely without the Trust Center.

madmagic007, 2025-08-22:
Hm, thats a bit odd, because take this scenario:

- Z2M network is available
- ROUTER device connects to Z2M network.
- both devices are taken offline and setup is moved to a different location.
- ROUTER gets powered on (no Z2M network available)
- ROUTER now creates its own network that other routers/end devices that get powered on after this one, can join.
- if Z2M were to become available, devices wont join because the ROUTER already created the network.

So its not a matter of the router keeping the existing network alive, it actually created it after a fresh boot.

madmagic007, 2025-08-22:
Thanks for the clarification. Some more testing on my end and i can confirm it now does rejoin z2m seamlessly once it becomes available.

I mustve been misusing/misunderstanding what "creating a network" actually means in zigbee terminology.

For my personal use case, it would still be beneficial if the router waited until z2m becomes available, but i can await a simple ping from z2m for that.

Thanks for the help and clarification on this!