espressif/arduino-esp32 #11752: Compatibility problem with arduino-esp32 v3.3.0 and ETH.h (W5500) with LCD Parallel

tresler, 2025-08-21:
### Board

ESP32-s3-wroom

### Device Description

Own board with own pinout. 

### Hardware Configuration

#define ETH_PHY_TYPE     ETH_PHY_W5500
#define ETH_PHY_SPI_FREQ_MHZ 10
#define ETH_PHY_ADDR        1
#define ETH_PHY_CS          10
#define ETH_PHY_IRQ         9
#define ETH_PHY_RST         14
#define ETH_PHY_SPI_HOST   SPI2_HOST
#define ETH_PHY_SPI_SCK     12
#define ETH_PHY_SPI_MISO    13
#define ETH_PHY_SPI_MOSI    11

const int stripPin[] = {21, 38, 39, 40, 1, 2, 3, 4, 5, 6, 7, 8, 17, 18, 41, 42};

### Version

v3.3.0

### Type

Bug

### IDE Name

Arduino IDE 2.3.6

### Operating System

Ubuntu 25.04

### Flash frequency

QIO 80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello,

I am experiencing a spinlock collision on my ESP32-S3 when using Ethernet together with NeoPixelBus and LCD Parallel method.

-     When I use only NeoPixelBus with NeoEsp32LcdX16Ws2813Method, everything works fine.
-     When I use only ETH.h with a W5500 module, everything also runs fine and I have network connection.
-     When I use them together, the code fails on the first strips[i]->Show(); call with the following error:

```
assert failed: spinlock_acquire spinlock.h:142 (lock->count == 0)

Backtrace: 0x40375c75:0x3fcebd70 0x4037bb1d:0x3fcebd90 0x40382656:0x3fcebdb0 
0x4037c9ee:0x3fcebef0 0x4205141e:0x3fcebf20 0x420038c2:0x3fcebf40 
0x42009888:0x3fcebf80 0x4037c7b9:0x3fcebfa0
```

This happens on Arduino-ESP32 v3.3.0 (ESP-IDF v5). Could this be fixed that it works with Ethernet (W5500) on ESP32-S3?

Thank you!

### Sketch

```cpp
#include <NeoPixelBrightnessBus.h>
#include <ETH.h>

const int stripPin[] = {21, 38, 39, 40, 1, 2, 3, 4, 5, 6, 7, 8, 17, 18, 41, 42};
const int stripLength[] = {238, 198, 203, 214, 259, 246, 226, 231, 254, 227, 213, 256, 215, 223, 224, 236};

#define ETH_PHY_TYPE     ETH_PHY_W5500
#define ETH_PHY_SPI_FREQ_MHZ 10
#define ETH_PHY_ADDR        1
#define ETH_PHY_CS          10
#define ETH_PHY_IRQ         9
#define ETH_PHY_RST         14
#define ETH_PHY_SPI_HOST   SPI2_HOST
#define ETH_PHY_SPI_SCK     12
#define ETH_PHY_SPI_MISO    13
#define ETH_PHY_SPI_MOSI    11
#define USE_W5100 false
SPIClass * fspi = NULL;

NeoPixelBrightnessBus<NeoRgbFeature, NeoEsp32LcdX16Ws2813Method>* strips[16]; 

void setup() {
  Serial.begin(115200);

  Serial.println();
  Serial.println("Initializing...");

  //ETH setup
  ETH.setTaskStackSize(8192);
  fspi = new SPIClass(HSPI);
  fspi->begin(ETH_PHY_SPI_SCK, ETH_PHY_SPI_MISO, ETH_PHY_SPI_MOSI, ETH_PHY_CS);
  //Serial.printf("Mac addressa: %u \n", mac);
  //ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, mac, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, *fspi, ETH_PHY_SPI_FREQ_MHZ);
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, *fspi, ETH_PHY_SPI_FREQ_MHZ);
  Serial.println("All network information: ");
  Serial.println(Network);

  for (int i = 0; i < 16; i++) {
    strips[i] = new NeoPixelBrightnessBus<NeoRgbFeature, NeoEsp32LcdX16Ws2813Method>(stripLength[i], stripPin[i]);
    strips[i]->Begin(); // Předpokládám, že metoda Begin() je potřebná pro inicializaci stripu
  }
  
  Serial.println();
  Serial.println("Running...");
}

int actualstrip = 0;
int a = 0;

void loop() {
    strips[actualstrip]->SetBrightness(255);
    strips[actualstrip]->SetPixelColor(a, RgbColor(255, 0, 0));      

    for (int i = 0; i < 16; i++) {
      strips[i]->Show();
    }

    a++;
    if (a > stripLength[actualstrip]) {
       a = 0;
       actualstrip++;
       if (actualstrip > 15) {
           actualstrip = 0;
       }
    }
}
```

### Debug Message

```plain
assert failed: spinlock_acquire spinlock.h:142 (lock->count == 0)

Backtrace: 0x40375c75:0x3fcebd70 0x4037bb1d:0x3fcebd90 0x40382656:0x3fcebdb0 
0x4037c9ee:0x3fcebef0 0x4205141e:0x3fcebf20 0x420038c2:0x3fcebf40 
0x42009888:0x3fcebf80 0x4037c7b9:0x3fcebfa0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-08-21:
I think that @me-no-dev may take a look.

JakubAndrysek, 2025-09-01:
Solved in https://github.com/Makuna/NeoPixelBus/issues/893

closed 2025-09-01 by JakubAndrysek

SuGlider, 2025-09-01:
As pointed by @JakubAndrysek, this issue is related to an external library (https://github.com/Makuna/NeoPixelBus) and not to Arduino Core code. The issue was closed given that it shall be tracked and fixed by https://github.com/Makuna/NeoPixelBus/issues/893