espressif/arduino-esp32 #11754: ESP32-S3 + ETH.h = w5500.mac: received frame was truncated

razvanphp, 2025-08-21:
### Board

esp32-s3-custom-board

### Device Description

Custom built board, but it worked with older Ethernet library, this error happens since we switched to ETH.h (so Wiznet via SPI + iwIP)

### Hardware Configuration

// W5500 Ethernet module settings
#define W5500_CS 11
#define W5500_RST 45
#define SPI_SCK 12
#define SPI_MISO 14
#define SPI_MOSI 13
#define SPI_FREQ 10000000

### Version

v3.3.0

### Type

Bug

### IDE Name

VSCode

### Operating System

Mac Ventura

### Flash frequency

240

### PSRAM enabled

no

### Upload speed

115200

### Description

We are using the latest version v3.3.0, and while the ETH works, we get this from time to time, without an apparent reason.

I know this is scarce information to reproduce, but can you please mention what other steps we could take to debug this further? For us, it happens with the Arduino_SPI Ethernet example, in combination with WiFi enabled in the same time.

```
00:18:55.295 > E (62027) w5500.mac: unexpected error 0x104
00:18:55.312 > E (62031) w5500.mac: emac_w5500_alloc_recv_buf(697): invalid frame length 10
00:18:55.312 > E (62038) w5500.mac: unexpected error 0x104
00:18:55.312 > E (62042) w5500.mac: emac_w5500_alloc_recv_buf(697): invalid frame length 10
00:18:55.327 > E (62048) w5500.mac: unexpected error 0x104
00:18:55.327 > E (62052) w5500.mac: emac_w5500_alloc_recv_buf(697): invalid frame length 10
00:18:55.327 > E (62059) w5500.mac: unexpected error 0x104
00:18:55.344 > E (62063) w5500.mac: emac_w5500_alloc_recv_buf(697): invalid frame length 10
00:18:55.344 > E (62070) w5500.mac: unexpected error 0x104
00:18:55.344 > E (62074) w5500.mac[ 62390][V][ETH.cpp:105] _onEthEvent(): eth0 Disconnected
00:18:55.360 > : received frame was truncated
00:18:55.360 > E (62084) w5500.mac: received frame was truncated
00:18:55.360 > E (62089) w5500.mac: received frame was truncated
00:18:55.360 > E (62093) w5500.mac: received frame was truncated
00:18:55.375 > E (62097) w5500.mac: received frame was truncated
00:18:55.375 > E (62102) w5500.mac: received frame was truncated
00:18:55.375 > E (62106) w5500.mac: received frame was truncated
00:18:55.392 > E (62111) w5500.mac: received frame was truncated
00:18:55.392 > E (62115) w5500.mac: received frame was truncated
00:18:55.392 > E (62120) w5500.mac: received frame was truncated
00:18:55.392 > E (62124) w5500.mac: received frame was truncated
00:18:55.408 > E (62128) w5500.mac: received frame was truncated
00:18:55.408 > E (62133) w5500.mac: received frame was truncated
00:18:55.408 > E (62137) w5500.mac: received frame was truncated
[...]
00:04:25.710 > E (251811) w5500.mac: emac_w5500_aE (251814) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
00:04:25.726 > E (251814) task_wdt:  - IDLE0 (CPU 0)
00:04:25.726 > E (251814) task_wdt: Tasks currently running:
00:04:25.726 > E (251814) task_wdt: CPU 0: wifi
00:04:25.726 > E (251814) task_wdt: CPU 1: w5500_tsk
00:04:25.742 > E (251814) task_wdt: Aborting.
00:04:25.742 > E (251814) task_wdt: Print CPU 0 (current core) backtrace
00:04:25.742 > 
00:04:25.791 > 
00:04:25.791 > 
00:04:25.791 > 
00:04:25.791 > Backtrace: 0x4211e4f5:0x3fcbd2c0 0x4211d939:0x3fcbd2e0 0x4214b2fa:0x3fcbd300 0x42149dfe:0x3fcbd330 0x421499a7:0x3fcbd370 0x42149a75:0x3fcbd390 0x4037ebc9:0x3fcbd3c0
  #0  0x4211e4f5 in mac_tx_set_plcp1 at ??:?
  #1  0x4211d939 in hal_mac_tx_set_ppdu at ??:?
  #2  0x4214b2fa in lmacSetTxFrame at ??:?
  #3  0x42149dfe in lmacTxFrame at ??:?
  #4  0x421499a7 in ppProcessTxQ at ??:?
  #5  0x42149a75 in ppTask at ??:?
  #6  0x4037ebc9 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

00:04:25.791 > 
00:04:25.791 > 
00:04:25.791 > 
00:04:28.870 > 
00:04:28.870 > ELF file SHA256: 83ebd3257
00:04:28.870 > 
00:04:28.870 > Rebooting...
00:04:28.870 > ���ESP-ROM:esp32s3-20210327
```

### Sketch

```cpp
Minimal code:
++
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SPI.setFrequency(SPI_FREQ);

  // Register the ETH event callback
  Network.onEvent(onEthEvent);

  bool began = ETH.begin(ETH_PHY_W5500, 1, W5500_CS, -1, W5500_RST, SPI);
```

### Debug Message

```plain
Stack trace included already.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-08-22:
I have a few suggestions:
- Try with the IDF SPI driver example
- For Arduino SPI, the SPI frequency is actually the last parameter of ETH.begin() (After the SPI bus). As you have defined it it will have no effect
- Problems could come from not optimal hardware design, cross traces, etc.
- Last but not least, have you tried with older Arduino versions?

razvanphp, 2025-08-22:
Maybe related to https://github.com/espressif/esp-idf/issues/14178 ?

razvanphp, 2025-08-25:
We have a custom PCB and the distance is 2 cm, there are no wires. I also tried setting the frequency the "other" way in begin() same result, after some time, the error appears in a loop and ethernet goes down.

Tried 10 MHz, 4, even 2, the same thing....

Can we catch this error somehow so it does not bubble indefinitely? We will try to get the `ESP32-S3-ETH` module from waveshare and test our code, but....

It feels very strange coincidence that we also use FastLED like in this issue: https://github.com/espressif/esp-idf/issues/14178, could this be related?

razvanphp, 2025-09-06:
Thanks for the tip, but what do you mean by 8 MB? Our PSRAM is max 2 MB and we also have boards with no PSRAM at all, so it should work with the 320k DRAM, or?

Today we also received and tested the waveshare ESP32-S3-ETH reference board and we could reproduce the error with our code, so it is not because of our custom PCB, it must be a bug or a limitation, but not sure how to investigate further…

razvanphp, 2025-09-24:
We are still struggling to find this bug, we tried many things, including a complete different hardware (waveshare ESP32-S3-ETH) with our full code and it still triggers the error. Also, we cross checked by flashing on our custom board a simple Arduino sketch + ETH.h demo and it's working great. So we can safely say it's not hardware related (we don't use wires and tried many low FREQ, even 10 or 4).

On the software side, I looked at the source code [here](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/spi/w5500/esp_eth_mac_w5500.c), the RX path reads the 2-byte frame length header only once from the chip’s ring buffer. If that read is torn/corrupted (because the CPU/interrupts were tied up), the driver will treat a bogus length as truth:
- If that (bogus) length is too small, we get
`emac_w5500_alloc_recv_buf(...): invalid frame length 26` → function returns `ESP_ERR_INVALID_SIZE` → task logs unexpected error `0x104`.
- If that length is too big, the task says received frame was truncated and still advances RX_RD by the bogus length, which de-syncs the ring and causes bursts of further errors until reset/relink. 

So anything that briefly starves the W5500 task or interrupts (long I²C transactions with retries, RMT LED updates, flash/PSRAM stalls, Wi-Fi bursts, IPv6 multicast) can tip it into exactly the pattern we see.

Can we maybe make the MAC driver a little bit more resilient here, maybe also helps solve this mystery?

Thank you in advance!
R

me-no-dev, 2025-09-25:
so you share the SPI bus with another device? Is it possible that that device alters the signals on the SPI bus? Would it happen if you only have the ETH connected to the board (on the same SPI pins). Difference between the two SPI approaches is that the IDF one can also use DMA. I was not able to measure any real difference in speeds though.

razvanphp, 2025-09-26:
I did some deep research with ChatGPT and was wondering if those directions make any sense:

1. ETH + SPI via Arduino HAL installs custom SPI callbacks that call back into SPIClass::beginTransaction/transfer/endTransaction for every W5500 access.
2. These calls can happen from multiple contexts (RX task, esp_timer link-check task, etc.). The Arduino SPI HAL is not designed to be IRAM-safe or timer/ISR-friendly, and its locking isn’t strong enough for multi-context, multi-core use with tight timing.
3. During heavy flash/PSRAM activity (when caches/interrupts can be paused), or simply when two ETH contexts overlap (for example irequests to ESPAsyncWebServer), one can end up with a torn read of the W5500 2-byte RX length field, triggering the described behaviour.
4. In the IDF path with Host overload, ETH creates its own ESP-IDF spi_device_handle_t, the W5500 SPI exchanges are atomic and properly serialized by the driver (thread-safe and deterministic); a stall either delays the whole transaction or it cleanly resumes.

So... I must ask, is this a bug/limitation in the Arduino SPIClass integration used by ETH?

Jason2866, 2025-09-26:
Probaby everything from ChatGPT is wrong since it is outdated. The way SPI Ethernet works in Arduino core 3.x has completely changed since core 2.x. The AI has not enough actual training data.

me-no-dev, 2025-09-26:
@razvanphp if the only device on that SPI bus is the ETH chip, then you might be better using the IDF SPI begin. We really added support for Arduino SPI in order to allow multiple devices to exist on the same bus in Arduino.

With that said, I do not agree with everything that GPT says, but I do agree with some/most. And I have not experienced issues with ETH and Arduino SPI myself, but I have not tried many scenarios either. The option to go one way or another are there. You can run ETH on one IDF SPI bus and also run your other devices on another Arduino SPI bus.

razvanphp, 2025-09-27:
I understand, we will stay with IDF host bus anyway, I just continued investigation maybe we can find something we can fix for others and contribute back to the community, so somebody else does not loose days figuring it out.

This is why I'm wondering if it's a limitation or a bug.

I saw this trace while trying to break it, if useful:

```
20:15:18.696 > Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
20:15:18.696 > 
20:15:18.696 > Core  0 register dump:
20:15:18.696 > PC      : 0x4213e2ff  PS      : 0x00060030  A0      : 0x820067f0  A1      : 0x3fca8940  
  #0  0x4213e2ff in spiGetClockDiv at /Users/razvan/.platformio/packages/framework-arduinoespressif32-src-a307a2f5a6b780e67ecb96e113cd5800/cores/esp32/esp32-hal-spi.c:472

20:15:18.696 > A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000000  
20:15:19.091 > A6      : 0x3fc97e48  A7      : 0x3fca8940  A8      : 0x0000023d  A9      : 0x3fca8a50  
  #0  0x3fc97e48 in _lx_intr_stack at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:565

20:15:19.091 > A10     : 0x3fcb00c8  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
20:15:19.091 > A14     : 0x3fcb0134  A15     : 0x3fca8a6f  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
20:15:19.091 > EXCVADDR: 0x0000023d  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
20:15:19.091 > 
20:15:19.091 > 
20:15:19.091 > Backtrace: 0x4213e2fc:0x3fca8940 0x420067ed:0x3fca8960 0x42006c19:0x3fca8990 0x42006cd6:0x3fca89c0 0x4213fd63:0x3fca89e0 0x420c0826:0x3fca8a00 0x4213fce1:0x3fca8a30 0x420c1f41:0x3fca8a50 0x4213fccb:0x3fca8a90 0x420d3da6:0x3fca8ab0 0x4037ebd9:0x3fca8ae0
  #0  0x4213e2fc in spiGetClockDiv at /Users/razvan/.platformio/packages/framework-arduinoespressif32-src-a307a2f5a6b780e67ecb96e113cd5800/cores/esp32/esp32-hal-spi.c:470
  #1  0x420067ed in SPIClass::beginTransaction(SPISettings) at /Users/razvan/.platformio/packages/framework-arduinoespressif32-src-a307a2f5a6b780e67ecb96e113cd5800/libraries/SPI/src/SPI.cpp:190
  #2  0x42006c19 in ETHClass::eth_spi_read(unsigned long, unsigned long, void*, unsigned long) at /Users/razvan/.platformio/packages/framework-arduinoespressif32-src-a307a2f5a6b780e67ecb96e113cd5800/libraries/Ethernet/src/ETH.cpp:456 (discriminator 1)
  #3  0x42006cd6 in ETHClass::_eth_spi_read(void*, unsigned long, unsigned long, void*, unsigned long) at /Users/razvan/.platformio/packages/framework-arduinoespressif32-src-a307a2f5a6b780e67ecb96e113cd5800/libraries/Ethernet/src/ETH.cpp:444
  #4  0x4213fd63 in w5500_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_eth/src/spi/w5500/esp_eth_mac_w5500.c:185
  #5  0x420c0826 in emac_w5500_read_phy_reg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_eth/src/spi/w5500/esp_eth_mac_w5500.c:433
  #6  0x4213fce1 in eth_phy_reg_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_eth/src/esp_eth.c:86
      (inlined by) eth_phy_reg_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_eth/src/esp_eth.c:77
  #7  0x420c1f41 in w5500_update_link_duplex_speed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_eth/src/spi/w5500/esp_eth_phy_w5500.c:70
      (inlined by) w5500_get_link at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_eth/src/spi/w5500/esp_eth_phy_w5500.c:113
  #8  0x4213fccb in eth_check_link_timer_cb at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_eth/src/esp_eth.c:191
  #9  0x420d3da6 in timer_process_alarm at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:435
      (inlined by) timer_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:461
  #10 0x4037ebd9 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

closed 2026-10-03 by github-actions[bot]