espressif/arduino-esp32 #11758: Arduino wrapper support for Matter Electrical Power Measurement Cluster (0x0090)

dsilletti, 2025-08-24:
### Related area

Matter

### Hardware specification

This feature request does not depend on specific hardware.

### Is your feature request related to a problem?

Currently, the Matter.h Arduino wrapper exposes endpoint classes for lights, sensors, thermostat, fans, etc., but does not provide a wrapper for the Matter Electrical Power Measurement (EPM) cluster (0x0090).

Developers working with devices with power monitoring need direct access to this cluster to expose standard attributes (RMSVoltage, RMSCurrent, ActivePower, PowerFactor, etc.) to Matter controllers.

### Describe the solution you'd like

Please add Arduino wrapper support for the Electrical Power Measurement cluster, similar to how MatterOnOffLight, MatterTemperatureSensor, etc. are provided.

Example API proposal:

`
MatterElectricalPowerMeasurement PowerMeter;

// Initialize with default values
PowerMeter.begin();

// Update attributes
PowerMeter.setRMSVoltage(int32_t mV);             // e.g. 230000 mV
PowerMeter.setRMSCurrent(int32_t mA);             // e.g. 500 mA
PowerMeter.setActivePower(int32_t W);               // e.g. 115 W
PowerMeter.setPowerFactor(int16_t pf_x1000);   // e.g: PF = 0.953 → pf_x1000 = 953

`

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

SikuFan, 2026-09-18:
Is there any update @SuGlider ? I just wanted to publish the battery voltage ist his the right use case?

SuGlider, 2026-09-21:
We have just finished a major Arduino Matter refactoring effort.

This endpoint is in the list for implementation.