espressif/arduino-esp32 #11774: Uploading Error

PukkyPukPukt, 2025-08-29:
### Board

ESP32 Dev Module

### Device Description

ESP32 NodeMCU with Silicon Labs CP2102

### Hardware Configuration

Only a Micro-USB cable connected to the Computer.

### Version

v3.3.0

### Type

Bug

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40 and 80 Mhz

### PSRAM enabled

no

### Upload speed

All

### Description

It is not possible to Upload any Sketches, even though the Serail Monitor is working.

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Der Sketch verwendet 281339 Bytes (21%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 20680 Bytes (6%) des dynamischen Speichers, 307000 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
esptool v5.0.0
Serial port COM3:
Connecting.....
A fatal error occurred: Invalid head of packet (0x00): Possible serial noise or corruption.

Fehlgeschlagenes Hochladen: Hochladefehler: exit status 2
```

### Other Steps to Reproduce

I tried exactly the same on three different computers, two with Windows 11 latest version, 1 with Windows 10 latest version. It solely didn't work on one of the Windows 11 machines.
In case it helps, this is my computer:
Acer Aspire A515-45G
NX.A8AEV.00G
You can look up the specs on the Acer support website.
Thank you very much
Felix

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Vinayakahr10, 2025-08-30:
same issue i am facing after updating

SuGlider, 2025-08-30:
@PukkyPukPukt / @Vinayakahr10 - What board are you using? If possible post a picture/website of the board.
If you set the ESP32 board into Download mode (BOOT + RESET) while the sketch is compiled, it is able to upload the firmware?

lbernstone, 2025-08-31:
> [@SuGlider](https://github.com/SuGlider) i am not able to upload

This is very typical behavior of a bad cable. Do you get the boot logging on Serial Monitor?

Vinayakahr10, 2025-08-31:
@lbernstone  yes

Vinayakahr10, 2025-08-31:
@PukkyPukPukt Just uninstall the Driver and install it

SuGlider, 2025-09-01:
> [@PukkyPukPukt](https://github.com/PukkyPukPukt) Just uninstall the Driver and reinstall it from https://www.silabs.com/software-and-tools/usb-to-uart-bridge-vcp-drivers?tab=downloads

Did it solve the issue?

Vinayakahr10, 2025-09-01:
@SuGlider yes

PukkyPukPukt, 2025-09-02:
The cable is intact, I also installed Windows the drivers and the Arduino IDE freshly, but only downgrading solved the issue.

SuGlider, 2025-09-02:
OK. It may be a problem related to preivous ESP32 Arduino instalation.
It may be possible to solve it by deleting some files and folders.

Please try this:
1- Delete all files in  C:\Users\YOUR_USER_NAME\AppData\Local\Arduino15\staging\packages\*.*
2- Delete the whole folder in C:\Users\YOUR_USER_NAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0
3- Reinstall the ESP32 board using the board manager for the Arduino Core 3,3.0

gluca, 2025-09-10:
Same here. Works with 3.2.1

lbernstone, 2025-09-13:
Can you download v5.0.2 of [esptool from github](https://github.com/espressif/esptool/releases/download/v5.0.2/esptool-v5.0.2-windows-amd64.zip), and see if `esptool chip_id` works with that version?

Jason2866, 2025-10-02:
Install latest esptool v5.1.0. There was an issue with Windows and certain USB serial chips. This is fixed in latest esptool v5.1.0