espressif/arduino-esp32 #11775: Fail to compile the empty project, Compilation error: exit status 1

g199209, 2025-08-29:
### Board

ESP32

### Device Description

Not about hardware.

### Hardware Configuration

N/A

### Version

v3.3.0

### Type

Bug

### IDE Name

Arduino IDE 2.3.6

### Operating System

Windows 11

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

Compile error, with verbose output enabled:

```
FQBN: esp32:esp32:esp32_family
Using board 'esp32_family' from platform in folder: C:\Users\g1992\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0
Using core 'arduino' from platform in folder: C:\Users\g1992\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0

cmd /c if exist "C:\\Users\\g1992\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025729-37188-1vn6fn.tepfjh\\sketch_aug29a\\partitions.csv" COPY /y "C:\\Users\\g1992\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025729-37188-1vn6fn.tepfjh\\sketch_aug29a\\partitions.csv" "C:\\Users\\g1992\\AppData\\Local\\arduino\\sketches\\8A475C7650B35E17A4121C88AF4ED35A\\partitions.csv"
cmd /c if not exist "C:\\Users\\g1992\\AppData\\Local\\arduino\\sketches\\8A475C7650B35E17A4121C88AF4ED35A\\partitions.csv" if exist "\\partitions.csv" COPY "\\partitions.csv" "C:\\Users\\g1992\\AppData\\Local\\arduino\\sketches\\8A475C7650B35E17A4121C88AF4ED35A\\partitions.csv"
cmd /c if not exist "C:\\Users\\g1992\\AppData\\Local\\arduino\\sketches\\8A475C7650B35E17A4121C88AF4ED35A\\partitions.csv" COPY "C:\\Users\\g1992\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.3.0\\tools\\partitions\\{build.partitions}.csv" "C:\\Users\\g1992\\AppData\\Local\\arduino\\sketches\\8A475C7650B35E17A4121C88AF4ED35A\\partitions.csv"
ϵͳ�Ҳ���ָ�����ļ���
exit status 1

Compilation error: exit status 1
```

There's no meaningful info in log, only a line of grabled characters: `ϵͳ�Ҳ���ָ�����ļ���`

Similar to #11726, but I have already reinstall Arduino IDE and ESP32 Board a few times, it doesn't solve the problem.

One more thing, I have successed compiled and uploaded to the board, and the Arduino IDE is installed under `D:\xxx` that time. And my D disk is broken, so I reinstall Arduino IDE under `C:\xxx`. It can not compile after that. I have tried to uninstall Arduino IDE, and delete all folder which contains `Arduino` entirly. But it doesn't solve the problem.

I think there're some inconsistent files in my system, but I don't know how to investigate further.

Any suggestions?

### Sketch

```cpp
++
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-08-29:
Check the discussion in a similar issue: https://github.com/espressif/arduino-esp32/issues/11726

The solution that issue was:
Let's try something a bit different:
1- Delete all files in  C:\Users\g1992\AppData\Local\Arduino15\staging\packages\*.*
2- Delete the whole folder in C:\Users\g1992\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0
3- Reinstall the ESP32 board using the board manager for the Arduino Core 3,3.0

ritesh006, 2025-08-30:
The error is happening because the board selected is ESP32 Family (esp32_family), which is only a placeholder/aggregator and not an actual compile target. Since it doesn’t define a partition scheme, compilation fails with the “file not found” message you saw.

***How to fix it:***

In Arduino IDE, go to Tools → Board and select your actual device (e.g. ESP32 Dev Module, ESP32-S3 Dev Module, etc.).

Then under Tools → Partition Scheme, pick any valid option (e.g. Default).

If you still see errors, clear cached files and reinstall the ESP32 core:

Delete C:\Users\<you>\AppData\Local\Arduino15\staging\packages\*

Delete C:\Users\<you>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.0\

Reinstall esp32 by Espressif Systems via Boards Manager.

After that, compilation should work fine.

closed 2025-10-02 by Jason2866

ahmedmon3em1996-netizen, 2025-12-19:
Traceback (most recent call last):
  File "gen_insights_package.py", line 56, in <module>
  File "gen_insights_package.py", line 30, in main
  File "shutil.py", line 435, in copy2
  File "shutil.py", line 264, in copyfile
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Mega\\Store\\AppData\\Local\\arduino\\sketches\\106528E95BCB91554095DA67DD28FB6A.bin'
[12636] Failed to execute script 'gen_insights_package' due to unhandled exception!
exit status 1

Compilation error: exit status 1