espressif/arduino-esp32 #11784: ESP32 Elegoo – Bootloop when enabling Wi-Fi/Bluetooth

nlagardere, 2025-08-31:
### Board

ESP32 Dev Module

### Device Description

DevKitC, USB-C

### Hardware Configuration

Nothing, just the card itself.

### Version

latest stable Release (if not listed below)

### Type

Task

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi,

I’m using an ESP32 Elegoo board. It works fine with simple sketches (e.g., Serial.print, toggling a GPIO LED).
However, whenever I try to enable Wi-Fi or Bluetooth, the board goes into a bootloop (keeps restarting).

✅ What I’ve already tried:

Different USB cables

Different USB ports and even external power supply via adapter

Simple code without Wi-Fi/Bluetooth → works fine

Any code using Wi-Fi or Bluetooth → bootloop

❓ My questions:

Could this be a hardware issue with the Elegoo ESP32?

Or maybe a software/config issue (wrong board selection in Arduino IDE, missing libraries, etc.)?

Could it be a power supply problem when using radio features?

Any help or suggestions would be greatly appreciated!

### Sketch

```cpp
#include <WiFi.h>

const char* ssid = "MonESP32";    
const char* password = "12345678";  

void setup() {
  Serial.begin(115200);
  delay(1000);

  WiFi.softAP(ssid, password);

  Serial.println("Point d'accès Wi-Fi créé !");
  Serial.print("Nom du réseau : ");
  Serial.println(ssid);
  Serial.print("Mot de passe : ");
  Serial.println(password);
  Serial.print("Adresse IP du point d'accès : ");
  Serial.println(WiFi.softAPIP());
}

void loop() {

}
```

### Debug Message

```plain
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4980
load:0x40078000,len:16612
load:0x40080400,len:3480
entry 0x400805b4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4980
load:0x40078000,len:16612
load:0x40080400,len:3480

It displays this in the serial monitor forever until I disconnect my card.
```

### Other Steps to Reproduce

This sketch and every programms work on my other ESP32 (same order, same box).

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-08-31:
The lack of any logging would imply that this is a power issue. If you have a good source of 3.3V power, you could try hooking that up to appropriate pin to bypass the onboard LDO (which may be very low quality). There have also been boards which did not provide enough capacitance on the 3V3 bus. So, if you find that direct powering works, it may be worth trying to solder a 10uF (or more) cap b/w 3v3 and gnd.

lbernstone, 2025-09-04:
If you aren't getting close to enough current, the capacitor is not going to fix it. The instantaneous power needed during WiFi init is about 4x more than the cpu needs.