espressif/arduino-esp32 #11787: I2C.Master logging

wrlaw, 2025-09-02:
### Board

ESP32

### Device Description

ESP32 Do IT V1

### Hardware Configuration

Issue potentially lots of devices but the example below is for an I2C based LCD

### Version

v3.3.0

### Type

Bug

### IDE Name

Arduino IDE 1.8.19

### Operating System

Debian Linux

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Arduino Core release 3.2.0 based on ESP-IDF Rel 5.4.1 introduced increased logging for i2c.master and this change was subsequently included in ESP-IDF 5.5.0 for Arduino Core 3.3.0.

Problem is, that increased logging produces messages which would be best classified as DEBUG but instead are reproduced on the serial port even if the Arduino IDE log level is configured as NONE.   This action fills the serial port with messages from i2c.master which the user may not be interested in.

This increased logging was not generated in ESP-IDF 5.3 used by Arduino Core release 3.1.3 so you either have to keep yourself at Arduino core release 3.1.3 or put up with 100s of messages coming out of i2c.master burying any messages from your application.

### Sketch

```cpp
/*
 * I2C based LCD as an example
 * 
 * On a dev board where for unrelated software development
 * the LCD may not be connected.
 *
 * The serial port is then filled with messages
 * if using Arduino Core releases 3.2.0 and 3.3.0
 * 
 * Release 3.1.3 has no such issue
 *
 * It should be possible to disable these messages
 * by setting the Arduino debug level to NONE
 * 
 */

#include <LiquidCrystal_I2C.h>  

LiquidCrystal_I2C lcd(0x27,16,2);
[... 345 lines cut ...]
13:33:17.701 -> E (3002) i2c.master: I2C hardware NACK detected
13:33:17.701 -> E (3006) i2c.master: I2C transaction unexpected nack detected
13:33:17.701 -> E (3012) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
13:33:17.701 -> E (3019) i2c.master: i2c_master_multi_buffer_transmit(1214): I2C transaction failed
13:33:17.734 -> E (3026) i2c.master: I2C hardware NACK detected
13:33:17.734 -> E (3030) i2c.master: I2C transaction unexpected nack detected
13:33:17.734 -> E (3036) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
13:33:17.734 -> E (3043) i2c.master: i2c_master_multi_buffer_transmit(1214): I2C transaction failed
13:33:17.768 -> E (3050) i2c.master: I2C hardware NACK detected
13:33:17.768 -> E (3054) i2c.master: I2C transaction unexpected nack detected
13:33:17.768 -> E (3060) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
13:33:17.768 -> E (3067) i2c.master: i2c_master_multi_buffer_transmit(1214): I2C transaction failed
13:33:17.768 -> E (3074) i2c.master: I2C hardware NACK detected
13:33:17.768 -> E (3078) i2c.master: I2C transaction unexpected nack detected
13:33:17.768 -> E (3084) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
13:33:17.768 -> E (3091) i2c.master: i2c_master_multi_buffer_transmit(1214): I2C transaction failed
```

### Other Steps to Reproduce

Take the example sketch, load it on a bare bones dev board without any LCD connected and watch the messages come out even if the Arduino debug level is set to NONE for Arduino Core releases 3.3.0 and 3.20 whereas for Release 3.1.3 no such messages come out if the debug level is set to NONE.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

syntax1269, 2025-09-05:
@wrlaw  very true! there is something wrong with I2C on an ESP32. Same code on an ESP8266, and no issues. ESP32, serial port logs look like the end of the world... Hope someone can spend some time and fix the issue.

jasoncouture, 2025-10-17:
Yeah... You should see the angry comments in code bases I work on about this, example (censored):

```
  // Make this shut the **** up by default please.
  // If I need it, I'll ****ing turn it on.
  esp_log_level_set("i2c.master", ESP_LOG_NONE);
```

me-no-dev, 2025-10-18:
PR submitted