espressif/arduino-esp32 #11796: Ble Client connect multi Server

CHNEG8805, 2025-09-04:
### Related area

BLE

### Hardware specification

S3,C6

### Is your feature request related to a problem?

how can S3 or C6 ble client connect multi server

### Describe the solution you'd like

S3 or C6 ble client connect multi server

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

lucasssvaz, 2025-09-04:
Could you explain in more detail what is your project and what are you trying to achieve?

CHNEG8805, 2025-09-08:
i want use a esp32C6 for a ble ap station.this ap connect 10 or more ble server. and c6 transmit message to the ble server.after one transmit ,close the connect,and connect another server