espressif/arduino-esp32 #11849: Upload fails after MacOS Tahoe Update (arduino ide)

ad1k4h, 2025-09-17:
### Board

ESP32-Wroom

### Device Description

ESP32 Wroom relay board

### Hardware Configuration

-

### Version

latest stable Release (if not listed below)

### Type

Task

### IDE Name

Arduino

### Operating System

MacOs 26.0

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

esptool v5.1.0

Serial port /dev/cu.usbserial-110:

Connecting....Traceback (most recent call last):

  File "esptool/__init__.py", line 1173, in _main

  File "esptool/__init__.py", line 1032, in main

  File "esptool/cli_util.py", line 229, in __call__

  File "rich_click/rich_command.py", line 404, in __call__

  File "click/core.py", line 1442, in __call__

  File "rich_click/rich_command.py", line 187, in main

  File "click/core.py", line 1830, in invoke

  File "click/core.py", line 1226, in invoke

  File "click/core.py", line 794, in invoke

  File "click/decorators.py", line 34, in new_func

  File "esptool/__init__.py", line 688, in write_flash_cli

  File "esptool/cmds.py", line 1112, in attach_flash

  File "esptool/loader.py", line 1091, in flash_id

  File "esptool/loader.py", line 1653, in run_spiflash_command

  File "esptool/loader.py", line 901, in read_reg

  File "esptool/loader.py", line 565, in check_command

  File "esptool/loader.py", line 495, in command

  File "esptool/loader.py", line 431, in read

StopIteration 

### Sketch

```cpp
-
```

### Debug Message

```plain
esptool v5.1.0

Serial port /dev/cu.usbserial-110:

Connecting....Traceback (most recent call last):

  File "esptool/__init__.py", line 1173, in _main

  File "esptool/__init__.py", line 1032, in main

  File "esptool/cli_util.py", line 229, in __call__

  File "rich_click/rich_command.py", line 404, in __call__

  File "click/core.py", line 1442, in __call__

  File "rich_click/rich_command.py", line 187, in main

  File "click/core.py", line 1830, in invoke

  File "click/core.py", line 1226, in invoke

  File "click/core.py", line 794, in invoke

  File "click/decorators.py", line 34, in new_func

  File "esptool/__init__.py", line 688, in write_flash_cli

  File "esptool/cmds.py", line 1112, in attach_flash

  File "esptool/loader.py", line 1091, in flash_id

  File "esptool/loader.py", line 1653, in run_spiflash_command

  File "esptool/loader.py", line 901, in read_reg

  File "esptool/loader.py", line 565, in check_command

  File "esptool/loader.py", line 495, in command

  File "esptool/loader.py", line 431, in read

StopIteration
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Loimulohi, 2025-09-18:
I'm not sure if this is the same problem, but uploading fails when using arduino-cli and core 3.3.1

```
arduino-cli  Version: 1.3.1 Commit: 08ff7e2b Date: 2025-08-28T13:51:45Z
esp32:esp32 3.3.1

arduino-cli compile --upload -p 192.168.2.124 --fqbn esp32:esp32:esp32c6 --board-options PartitionScheme=default_8MB --board-options FlashSize=8M

Uploading to specified board using network protocol requires the following info:
Password:`<correct password as it's working with core 3.3.0>
Sending invitation to 192.168.2.124
Authenticating...FAIL
15:19:22 [ERROR]: No Answer to our Authentication
Error during Upload: Failed uploading: uploading error: exit status 1
```

Uploading is working ok when using core 3.3.0

```
arduino-cli  Version: 1.3.1 Commit: 08ff7e2b Date: 2025-08-28T13:51:45Z
esp32:esp32 3.3.0

arduino-cli compile --upload -p 192.168.2.124 --fqbn esp32:esp32:esp32c6 --board-options PartitionScheme=default_8MB --board-options FlashSize=8M
Uploading to specified board using network protocol requires the following info:
Password:
Sending invitation to 192.168.2.124
Authenticating...OK
Uploading: [============================================================] 100% Done...
```

me-no-dev, 2025-09-18:
@lucasssvaz PTAL on the network OTA problem

@Loimulohi this is a different issue. OP can not upload using esptool, not OTA

ad1k4h, 2025-09-18:
If i lower the upload bitrate to 115200 its works again, but slow.

rjp663, 2025-09-22:
I'm using a ESP32 Touchscreen 2.8inch TFT LCD Display ILI9341 Resistive Touch Module 240 * 320 ESP32-2432S028R Bluetooth WiFi Dual Core Development Board for Arduino IDE. The PCB uses a CH340C serial chip.

rjp663, 2025-09-22:
> @rjp663 Not the same error. You are getting a connection, but the upload is failing.  Did this work before your Tahoe update, and now doesn't work?

Ibernstone,Thank you for clarifying. I updated to Tahoe before receiving and testing the ESP32 based display. I did verify that the display works fine on Windows 11.Sent from my iPad

Jason2866, 2025-10-02:
Latest USB serial drivers from manufacturer of the USB serial chips installed?

rjp663, 2025-10-02:
> Jason2866

All,

My bad, operator error.

Rick

Jason2866, 2025-10-02:
@ad1k4h Which driver have you installed. It does not work without installing the according driver from the manufacturer of the USB serial chip. The inbuilt Apple driver is not working for this!

SNNafi, 2025-10-19:
Same issue. Worked after lowering upload baudrate.