espressif/arduino-esp32 #11855: OTA upload failes after updated board from 3.3.0 to 3.3.1

AbedKarmi, 2025-09-18:
### Board

ESP32 Dev Module

### Device Description

DevKitC

### Hardware Configuration

none

### Version

v3.3.1

### Type

Bug

### IDE Name

Arduinoe IDE 2.3.6

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

It was functioning flawlessly before the most recent update. I simply replaced setPasswordHash with setPassword and used the plain text password, and it worked perfectly on one Sketch, while it did not work for both plain text password and password hash after recent update!

### Sketch

```cpp
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "xxx";
const char* password = "(xxx)";

void setup() {      
  Serial.begin(115200);

  // Connect to Wi-Fi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("\nConnected to WiFi");

  // Start OTA
  ArduinoOTA.setHostname("esp32-ota");
  //ArduinoOTA.setPassword("admin");
  ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3"); // admin

  ArduinoOTA.onStart([]() {
    Serial.println("Start updating");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });

  ArduinoOTA.begin();
  Serial.println("OTA Ready! IP: " + WiFi.localIP().toString());
}

void loop() {
  ArduinoOTA.handle();
  delay(100); // small delay to avoid watchdog
}
```

### Debug Message

```plain
Sketch uses 1005027 bytes (51%) of program storage space. Maximum is 1966080 bytes.
Global variables use 49332 bytes (15%) of dynamic memory, leaving 278348 bytes for local variables. Maximum is 327680 bytes.
"C:\Users\karmi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.3.1\tools\espota.exe" -r -i 192.168.0.184 -p 3232 "--auth=admin" -f "C:\Users\karmi\AppData\Local\arduino\sketches\9C2507445C698FDFABD40457B1619791/sketch_sep18a.ino.bin"
Sending invitation to 192.168.0.184 
Authenticating...FAIL
12:23:44 [ERROR]: Authentication Failed
Failed uploading: uploading error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2025-09-18:
Added option to `espota` to allow uploading images with the old authentication method. Thanks for the heads up.

HomeSpan, 2025-10-05:
I am not sure this addresses the full issue and may cause devices to lose OTA accessibility if 3.3.1 is uploaded.  Though the revised espota.py script provided a method to upload a 3.3.1-based sketch via OTA to a device running a pre-3.3.1 sketch, once the 3.3.1-based sketch is loaded, there are potential new problems.

If the sketch sets the ArduinoOTA password in plaintext, once the 3.3.1 sketch is loaded everything will work fine.  But if the sketch used ArduinoOTA.setPasswordHash() by first calling MD5 or using a pre-defined MD5 hash of the password, I think this will be incompatible with the new ArduinoOTA library that not only uses SHA-256, but has additional protocol logic that relies on SHA-256.

I just tested this on a 3.3.0-based sketch that saves a hashed password that used MD-5.   The first OTA upload of the same sketch under 3.3.1 using the new espota.py script worked as described above by falling back to using MD-5.  But after that OTA stopped working.  I think that before you upload a sketch that uses ArduinoOTA.setPasswordHash() (or your sketch uses a library that in turn uses this function), the sketch (or library) must also be updated to use SHA-256, otherwise the stored password on the device (for example, something saved in NVS as an MD-5) will be incompatible with the ArduinoOTA 3.3.1 now running on that device.

For hard-to-access devices that rely on OTA, this could be a real problem.

One possible solution would be to keep both MD-5 and SHA-256 options in the core and add a new ArduinoOTA.setPasswordHash256() function.  The library can then tell whether to use the MD-5 or SHA-256 logic.  You can flag the ArduinoOTA.setPassword() function as deprecated, and then eventually remove it.  Any sketches using that function would fail to compile, which is better than compiling but uploading in a way that breaks the device.

lucasssvaz, 2025-10-06:
@HomeSpan I will take a look.

HomeSpan, 2025-10-06:
Thanks.  Also, I was able to solve the issue with an additional change to esports.py.  

Instead of using SHA256 for the password hash under the new SHA256 method, you can use MD5 but still retain the new challenge method.  The new method uses SHA256 for the challenge keys but anything can be used to hash the OTA password.  For devices that store their password using MD5, this solves the issue.

Also, you may be able to tell if a device is using pre- or post-3.3.1 OTA by the size of the initial AUTH response.  Rather than read either 37 or 69 characters, try reading 69 and if only 37 are available you know to use classic MD5 logic.  No need to first try 256 and then fallback to MD5 in this case. 

But the method you created to try 256 first, then MD5 if 256 fails, could instead be used to address the issue of a 3.3.1 device storing an MD5 password as described above. 

In sum, this allows for 3 combinations:

* updating a pre-3.3.1 (old OTA) device with 3.3.1 (new OTA) - detectable by size of AUTH response

* updating a post 3.3.1 device that uses SHA-256 to store its password (or does not bother to store their password at all) - initial pass

* updating a post 3.3.1 device already has a stored MD5 password - fallback if above fails.

lucasssvaz, 2025-10-07:
@HomeSpan I dont have a device with me right now. Could you please try, if you have time, the fix in https://github.com/espressif/arduino-esp32/pull/11905 ?

lucasssvaz, 2025-10-08:
@HomeSpan Ah yes, I forgot to resend the invitation if it fails. Should be fixed now.

I will test the fix tomorrow fully. Feel free test and/or use it in the mean time. Thank you for the help!

closed 2025-10-08 by me-no-dev